lcdproc/lcdproc #5: Update of Change Log

wilberforce, 2016-07-27:
For each pull request - Should a line be added to refer to the pull request?

https://github.com/lcdproc/lcdproc/blob/master/ChangeLog#L8

v0.5dev (ongoing development)
+ Update hd44780-i2c driver, pins defined via config file, rather than hardcoded https://github.com/lcdproc/lcdproc/pull/1

v0.5.7

wilberforce, 2016-07-27:
Should it be changed to ChangeLog.md so that links link above will work?

willfe, 2016-07-27:
Well, I've renamed and updated the file, but it doesn't seem to be auto-linking to the pull requests for the three new entries I added :(

wilberforce, 2016-07-27:
Thanks - looks good.

For the reference - use full path to the pull request:
`https://github.com/lcdproc/lcdproc/pull/1`

This then gets swapped out on view as #1 and auto links....
;-)

wilberforce, 2016-07-27:
Hmm didn't work ;-(

https://github.com/lcdproc/lcdproc/blob/752abb7a1eb48b7792eb27fd1abc1605580009b7/ChangeLog.md

I'm not sure how - anyone?

willfe, 2016-07-28:
Eh, I just ended up wrapping the URLs in parenthesis. Works well enough (it doesn't shorten the URLs, but it looks reasonably decent while still referencing the appropriate PRs.

willfe, 2016-08-14:
Closing; feel free to reopen if we need to discuss further. Thanks!

closed 2016-08-14 by willfe