lcdproc/lcdproc #13: libusb-0.1 not supported anymore

haraldg, 2016-08-05:
Currently some drivers use libusb-1.0 and some use libusb-0.1 which isn't supported anymore. If drivers aren't ported to the new API, they can't be enabled by distributions like debian and OpenWRT soon. See https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=810428

dod38fr, 2016-11-07:
Note that switch to `libusb1` must be done with switch to  `libftdi1`. See Debian bug [810373](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=810373) for more details

haraldg, 2017-01-06:
While looking at this libusb mess again, I noticed that there probably is no reason that drivers such as `lis` depend on libusb directly. So I figure we can in fact do the libusb and the libftdi transition independently, if we drop spurious dependencies ... 

Not sure if I want to spend time on this before the release. Any volunteers?

dod38fr, 2019-04-18:
hd44780 driver is also impacted.  See [Missing lcd2usb ConnectionType in hd44780 driver](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=927024) bug

haraldg, 2019-04-18:
Well, people hat ample warning that this is going to happen ...

Nobody stepped forward to address this, so I guess the user in question is on his own to port the driver he cares about and submit a PR. Listing affected drivers in this issue is not going to make a difference.

dod38fr, 2019-04-18:
Fair enough. 

Hopefully, people will stumble on this list and won't open new bugs on Debian or elsewhere for a similar issue.

All the best

hogin86, 2019-08-23:
Hello,

I changed from Debian Stretch to Buster and now my display is not working anymore. How can the problem be solved if I understood correctly, is it the libusb of Debian?

Here are some information about my configuration:

**LCDd.conf**  (is not the complete content)

[server]
DriverPath = /usr/lib/x86_64-linux-gnu/lcdproc/
Driver=hd44780

[hd44780]
ConnectionType=lcd2usb
Contrast=850
Brightness=800
OffBrightness=0
Keypad=no
Backlight=yes
Size=20x4

**Output syslog:**

LCDd[3199]: LCDd 0.5.9, LCDproc Protocol 0.3
LCDd[3199]: Part of the LCDproc suite
systemd[1]: LCDd.service: Main process exited, code=exited, status=1/FAILURE
LCDd: hd44780: unknown ConnectionType: lcd2usb
systemd[1]: LCDd.service: Failed with result 'exit-code'.
LCDd: Driver [hd44780] init failed, return code -1
LCDd: Could not load driver hd44780
LCDd: There is no output driver
LCDd: Critical error while initializing, abort.

dod38fr, 2019-08-23:
@hogin86  hd44780 driver works with an obsolete version of libusb. The new one shipped with latest Debian release is not backward compatible, that's why `hd44780` driver is broken.

This can be fixed provided someone bites the bullet and ports `hd44780` driver to new libusb version and proposes a pull request for lcdproc.

All the best

hogin86, 2019-08-23:
Hello @dod38fr this does not sound good. If I understand correctly, this can take a long time or never happen?

Can I use an older version of LCDproc?

hogin86, 2019-08-23:
Hello @dod38fr I got it working again :-)
Have use the old hd44780.so from Debian Stretch and copied into the driver directory.

At the moment it does not look as if the other thing is affecting things. Is for me a quick solution first.

dod38fr, 2019-08-24:
@hogin86 your workaround will break once https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=810470 is resolved. It may also break with future versions of lcdproc.

If you know a little bit about C programming, I'd suggest that you port `hd44780-lcd2usb` driver to new libusb. Since you have the hardware, you're in good position to test the result.

All the best

Martinique, 2020-01-18:
I came across this issue when upgrading from Ubuntu 16.04 to 18.04 and eventually found this thread. 18.04 still has both libusb 0.1 and 1.0, but even the workaround mentioned by @hogin86 doesn't seem to help.

A [compatibility wrapper](https://github.com/libusb/libusb-compat-0.1) exists, so instead of recoding all old drivers can't lcdproc simply depend on that after the old libusb is removed from Debian and derivatives?

Martinique, 2020-01-18:
Apparently Debian maintainers have been hesitating with usblib-compat-0.1 for years, which sounds insane when it could probably solve this lcdproc driver issue and many others: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=731424

The wrapper sounds like a perfect solution, as it allows using both API versions in a single application (quote from libusb developers):

> However, if you have installed libusb-1.0 then we strongly recommend to use libusb-compat-0.1 instead of the ancient libusb-0.1 code, so that programs which use both the 0.1 API and the 1.0 API in different parts of the program, or in different libraries used by the program, will both end up using libusb-1.0 for actual device access. This is important to avoid potential conflicts between libusb-1.0 and libusb-0.1 being used by the same process.

Perhaps lcdproc developers could add some pressure to make Debian folks do something about this. Arch Linux and probably others already support the wrapper.

x12a1f, 2021-04-20:
I have created a pull request (#171) which changes the lcd2usb driver to use libusb 1.0 instead. It was pretty straight forward. If I have a way to test the other drivers I can do them as well.

artizirk, 2021-04-26:
I just tested #171 on debian and it worked for my usecase with lcd2usb

Andrejkab, 2024-02-01:
> I have created a pull request (#171) which changes the lcd2usb driver to use libusb 1.0 instead. It was pretty straight forward. If I have a way to test the other drivers I can do them as well.

Is possible to port driver with libusb 1.0 for glcd2usb  also? Many Thanks!

x12a1f, 2024-02-04:
@Andrejkab based on the changes I made to the lcd2usb, I made the same changes to the glcd2usb driver, see: https://github.com/lcdproc/lcdproc/commit/c62818bf491f06c6bfa68248a3e092f21456824b

It compiles but I have not tested it as I do not have glcd2usb hardware.

Andrejkab, 2024-02-04:
@x12a1f Thanks for Your time and answer! 

Yes, it compiles without errors.  But ...
Starting LCDd I got: _Could not open driver module /usr/local/lib/lcdproc/glcd.so: /usr/local/lib/lcdproc/glcd.so: undefined symbol: libusb_open_
Perhaps something wrong with my libusb? Although libusb-1.0-0-dev is installed on my system (Debian 12)

Thanks!

Andrej

x12a1f, 2024-02-07:
@Andrejkab I checked the source and according to the comments it should work even if the device can not be claimed. I did notice the call to detach the kernel driver was disabled so I enabled it. It might work.

I'll see if I can get a mega16/mega32 so I can test it myself.

Andrejkab, 2024-02-07:
Thanks for Your support! 
Glad to inform, that after last code improvement GLCD2USB started successfully on amd64 Debian. Will see if it compiles and works on my router. 

[image]

All the best!