lcdproc/lcdproc #14: still old and not-so-old issues on sourceforge

haraldg, 2016-08-05:
The issue tracker on sourceforge is still used by some people and has some interessting material. Adding links here to make this easier to find for everybody working on cleaning up and merging patches.

https://sourceforge.net/p/lcdproc/patches/

https://sourceforge.net/p/lcdproc/bugs/

willfe, 2016-08-17:
I've just grabbed an export/snapshot of everything Sourceforge would give me, including the bugs and patches tracked there. I'm a bit swamped tonight but I'll dig through this in the next couple of days and work out what to do with/about the stuff I find.

haraldg, 2017-02-08:
I have addressed most of the old issus one way or the other by now. However new tickets get opened there. Somebody with access to the sf project, should put a pointer to github there.