lcdproc/lcdproc #18: Moving the LCDjava library to LCDproc's organization

antoneliasson, 2016-08-10:
Not sure if the mailing list is still functional so I'm cross-posting the message I sent three days ago here:

I have recently taken over maintainership of [LCDjava][], a Java client library for LCDproc originally created by boncey / Darren Greaves. It is listed under Clients on lcdproc.org. I have fixed some bugs and improved its documentation and packaging. I feel that the project is soon ready for its first stable release and publication to the central Maven repository.

The Maven repository is used by many Java build systems such as Maven itself, Apache Ivy and Scala SBT. It identifies packages globally by a groupId and an artifactId. As you may know, there is a long-standing tradition in Java of using the reverse fully qualified domain name in the groupId [2][]. Typically, the product/technology name is appended to the reverse FQDN and may also be used in the artifactId.

I would like to use LCDproc's name in LCDjava's packaging, to make it the "official" LCDproc Java client library. The groupId would for example be "org.lcdproc.lcdjava" and the artifactId "lcdjava". The Java classes would be filed under org.lcdproc.lcdjava.* . I think this is more suitable for a long-lived OSS project than using a personal domain in the packaging info.

I see that LCDproc is in the process of migrating to Git and has a Github organization [3]. I would also be happy to move LCDjava over to this organization, along with the example projects (see the two repositories linked from [LCDjava][]'s README).

What do the LCDproc developers think about this idea? I have also opened a ticket to see what boncey thinks. [4]

[lcdjava]: https://github.com/boncey/lcdjava
[2]: https://docs.oracle.com/javase/tutorial/java/package/namingpkgs.html
[3]: http://github.com/lcdproc
[4]: https://github.com/boncey/lcdjava/issues/3

antoneliasson, 2016-08-10:
Update since I wrote it: boncey approves of the idea.

willfe, 2016-08-14:
I'd be alright with that. Would you be willing to be the project's maintainer? I'd probably be out of my element trying to take that on myself.

antoneliasson, 2016-08-15:
Yes, that's what I had in mind. I can't promise that I will actively develop it long term, but I don't maintaining it will be a lot of work maintaining it, since lcdproc is a pretty slow-moving project and Java code usually doesn't go stale by itself.

willfe, 2016-08-15:
Sounds good to me! I'll get things set up and an invite sent to you shortly. Is the name "LCDjava" acceptable (and cased correctly)?

willfe, 2016-08-15:
Done, and invite sent. Closing this for now; feel free to reopen if I've made any goofs in setting up the repo at https://github.com/lcdproc/LCDjava or you need other configuration changes made.

Thanks for offering to help out with this!

antoneliasson, 2016-08-16:
Nice! Could you change the repo name to "lcdjava". I find capitalization in URLs confusing and like to use all lower case wherever possible.

antoneliasson, 2016-08-16:
I seem to be unable to reopen this issue. According to https://github.com/isaacs/github/issues/583 it's because I am not an lcdproc collaborator and a collaborator closed it.

willfe, 2016-08-16:
I've renamed the repo (I was torn on the capitalization thing as well, guess my coin toss went the wrong way :)).

antoneliasson, 2016-08-16:
Hmm, could you delete it again? We'll try something else.

Repositories can be transferred from one owner to another, or to an organization. That creates redirects so that links to the old repository still work. But there can't be another repository on the destination before. So after you've deleted it, I'll ask boncey (as the repo owner) to transfer it to organization lcdproc. Then you'll get a notification to accept the transfer. Boncey is then automatically added as a collaborator to lcdproc/lcdjava. Finally you can add me as a collaborator or admin of lcdproc/lcdjava. Background: https://help.github.com/articles/transferring-a-repository-owned-by-your-organization/

willfe, 2016-08-16:
I sent you an invite yesterday for that very purpose. Wonder if it didn't go through somehow. See if you can see it at https://github.com/lcdproc/lcdjava/invitations ... I can either re-send it if not or just delete that repo as you suggest.

antoneliasson, 2016-08-16:
I did receive it, but haven't accepted it. What I would like to do is instead of setting up a new repo and pushing the local Git repo to a new remote, transfer (a Github operation) the existing repo at boncey/lcdjava to the lcdproc organization. But that can't be done yet because there is an existing repo in the way at lcdproc/lcdjava. So just remove the empty repo please.

willfe, 2016-08-16:
Done.

antoneliasson, 2016-08-17:
It turns out that to transfer a repository from a personal account to an organization, you need admin or owner rights in the receiving organization (https://help.github.com/articles/transferring-a-repository-owned-by-your-personal-account/). Could you temporarily give boncey this permission to the lcdproc organization?

willfe, 2016-08-17:
No, that's asking a bit too much :)

The same page indicates that transferring the repo to another _user_ still preserves everything (including permissions, issues, etc.), so it sounds like you can transfer the repo to my account (willfe) and then I can transfer it to the organization.

It's a bit ridiculous that Github requires this much hoop-jumping to get a repo transferred to an organization.

antoneliasson, 2016-08-17:
I agree that it's really asking too much, and that the transfer procedure is silly.

That's a clever workaround that I didn't think of. I've asked @boncey to try that instead.

willfe, 2016-08-17:
Looks good from here! I've updated the issue on that project (at https://github.com/lcdproc/lcdjava/issues/3) with a similar note. I'll leave both issues open (though really now @boncey should be able to close that one too :)) just to make sure everything's running smoothly.

Thanks @antoneliasson for all your help (especially digging through Github's policies/documentation) in facilitating this!

Welcome aboard (to you both)!

closed 2016-08-26 by antoneliasson