lcdproc/lcdproc #28: imonlcd dmesg errors

GZep, 2016-12-12:
Hello. lcdd demon works, but there is some problem with the server driver, I guess:
`...
[13728.771949] imon:send_packet: packet tx failed (-32)
[13728.781381] imon:lcd_write: send packet failed!
[13753.715177] imon:send_packet: packet tx failed (-32)
[13753.724782] imon:lcd_write: send packet failed!
[13817.052136] imon:send_packet: packet tx failed (-32)
[13817.061700] imon:lcd_write: send packet failed!
[13825.242160] imon:send_packet: packet tx failed (-32)
[13825.251699] imon:lcd_write: send packet failed!
[13877.515668] imon:send_packet: packet tx failed (-32)
[13877.525140] imon:lcd_write: send packet failed!
[13883.522322] imon:send_packet: packet tx failed (-32)
[13883.531890] imon:lcd_write: send packet failed!
[13922.119242] imon:send_packet: packet tx failed (-32)
[13922.128599] imon:lcd_write: send packet failed!
[13999.685908] imon:send_packet: packet tx failed (-32)
[13999.695564] imon:lcd_write: send packet failed!
[14018.399307] imon:send_packet: packet tx failed (-32)
[14018.412253] imon:lcd_write: send packet failed!
...`

I'm using imonlcd.so driver with protocol version 1.
Protocol version 0 is not working for me at all.
I tried to add "usleep(100000)" at the end of "send_packet" function, display becomes very "slow", but errors remain.

OS archlinux x64, cpu core i3 6100

haraldg, 2016-12-13:
I can't find your error messages in imonlcd driver of LCDd. Which version of lcdproc/LCDd are you using? Also since you seem to imply that the error messages are in dmesg, the probably come from some kernel driver? Error code 32 seems to be "broken pipe" which I can't make any sense of ATM either. Maybe this information helps you tracking down your issue anyway. I think I can't be of much help without more information.

Harald

GZep, 2016-12-14:
> Which version of lcdproc/LCDd are you using?

I'm using lcdproc 0.5.7-2. I've also tried "ongoing" version.

> the probably come from some kernel driver

Thanks, I will try to fix it. I'll report the progress.

GZep, 2016-12-16:
So, here is the output from LCDd started in foreground mode
**ReportLevel=5**
```
~ sudo LCDd 
LCDd version 0.5.7 starting
Built on Sep  7 2015, protocol version 0.3, API version 0.5
Using Configuration File: /etc/LCDd.conf
Set report level to 5, output to stderr
LCDd 0.5.7, LCDproc Protocol 0.3
Part of the LCDproc suite

(licence text)

Server running in foreground
Listening for queries on 127.0.0.1:13666
screenlist_init()
driver_load(name="imonlcd", filename="/usr/lib/lcdproc/imonlcd.so")
imonlcd: using Device /dev/lcd1
imonlcd: init() done
Key "Escape" is now reserved exclusively by client [-1]
Key "Enter" is now reserved shared by client [-1]
Key "Up" is now reserved shared by client [-1]
Key "Down" is now reserved shared by client [-1]
screenlist_process()
screenlist_switch(s=[_server_screen])
screenlist_switch: switched to screen [_server_screen]
screenlist_process()
screenlist_process()
screenlist_process()
screenlist_process()
screenlist_process()
screenlist_process()
screenlist_process()
screenlist_process()
screenlist_switch(s=[_server_screen])
screenlist_process()
screenlist_switch(s=[_server_screen])
screenlist_process()
screenlist_switch(s=[_server_screen])
```
I think it's ok, just a lot of some debug information

**ReportLevel=4** output:
```
~ sudo LCDd 
LCDd version 0.5.7 starting
Built on Sep  7 2015, protocol version 0.3, API version 0.5
Using Configuration File: /etc/LCDd.conf
Set report level to 4, output to stderr
LCDd 0.5.7, LCDproc Protocol 0.3
Part of the LCDproc suite

(licence text)

Server running in foreground
Listening for queries on 127.0.0.1:13666
imonlcd: using Device /dev/lcd1
Key "Escape" is now reserved exclusively by client [-1]
Key "Enter" is now reserved shared by client [-1]
Key "Up" is now reserved shared by client [-1]
Key "Down" is now reserved shared by client [-1]
screenlist_switch: switched to screen [_server_screen]
imon: error writing to file descriptor: -1imon: error writing to file descriptor: -1imon: error writing to file descriptor: -1imon: error writing to file descriptor: -1
```
Error throws from imonlcd.c:
```
static void
send_packet(PrivateData *p)
{
	int err;
	err = write(p->imon_fd, p->tx_buf, sizeof(p->tx_buf));

	if (err <= 0)
		printf("%s: error writing to file descriptor: %d", "imon", err);

        /* usleep(100000); this doesn't help */
}
```

haraldg, 2016-12-17:
This code, reporting the error is bad in many ways. I'll improve this tomorrow. Maybe this will give us some hint, why the kernel tells us "Operation not permitted". Also you could dig up the source of the kernel module and look where/why it would return this error. (I'm not familiar with this device.)

Are you implying that LCDd works fine with ReportLevel=5 but shows the error with ReportLevel=4? That's very strange. That would point to some kind of race, I guess.

haraldg, 2016-12-17:
Can you test PR #31? It should give better error messages.

GZep, 2016-12-18:
> Are you implying that LCDd works fine with ReportLevel=5 but shows the error with ReportLevel=4?

No, I changed report level because **screenlist_switch/screenlist_process** line output like couple times in each second.

> Can you test PR #31? It should give better error messages.

Yes, thank you. Output:
```
~/D/lcdproc (master ⚡=) sudo LCDd 
LCDd version 0.5dev starting
Built on Dec 18 2016, protocol version 0.3, API version 0.5
Using Configuration File: /etc/LCDd.conf
Set report level to 4, output to stderr
LCDd 0.5dev, LCDproc Protocol 0.3
Part of the LCDproc suite
Copyright (C) 1998-2014 William Ferrell, Selene Scriven
                        and many other contributors

This program is free software; you can redistribute it and/or
modify it under the terms of the GNU General Public License
as published by the Free Software Foundation; either version 2
of the License, or (at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU General Public License for more details.

You should have received a copy of the GNU General Public License
along with this program; if not, write to the Free Software Foundation,
Inc., 51 Franklin Street, Fifth Floor, Boston, MA 02110-1301, USA.

Server running in foreground
Listening for queries on 127.0.0.1:13666
imonlcd: using Device /dev/lcd1
imonlcd: error sending command 1200000000000000: Broken pipe

Key "Escape" is now reserved exclusively by client [-1]
Key "Enter" is now reserved shared by client [-1]
Key "Up" is now reserved shared by client [-1]
Key "Down" is now reserved shared by client [-1]
screenlist_switch: switched to screen [_server_screen]
imonlcd_flush: sending data for msb=20: Broken pipe

imonlcd_flush: sending data for msb=35: Broken pipe

imonlcd_flush: sending data for msb=2b: Broken pipe

imonlcd_flush: sending data for msb=3b: Broken pipe

imonlcd_flush: sending data for msb=2c: Broken pipe

imonlcd_flush: sending data for msb=27: Broken pipe

imonlcd_flush: sending data for msb=2d: Broken pipe
```

haraldg, 2016-12-18:
These errors look pretty random to me. I suspect this is more a problem with the HW and not with the driver of LCDd. If you run LCDd multiple times, do you get the exact same sequence of errors each time?

At this point I think the only advice I can give you is to find the source of the imon kernel module and see under which condition it reports EPIPE.

GZep, 2016-12-18:
> do you get the exact same sequence of errors each time

Yes, there are random **msb** values any time a run it.
Ok, thanks a lot!

closed 2016-12-18 by GZep