lcdproc/lcdproc #34: hd44780 driver linked with system sem_wait instead of lcdproc sem_waut

dod38fr, 2016-12-29:
Hello

With [Debian bug 849659](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=849659), Benoît Allard report a segfault of hd44780 driver:

> Using the hd44780 driver with connectiontype=8bit consistently triggers
> a segmentation fault.
> 
> The drivers of lcdproc define their own sem_get, sem_wait,
> sem_signal, ... (See server/drivers/lcd_sem.h).
> 
> Unfortunately, the linux's version of sem_wait (3) is being used,
> leading to a segmentation fault.

```
Program received signal SIGSEGV, Segmentation fault.
sem_wait () at ../nptl/sysdeps/unix/sysv/linux/x86_64/sem_wait.S:44
44	../nptl/sysdeps/unix/sysv/linux/x86_64/sem_wait.S: No such
file or directory. (gdb) bt
#0  sem_wait () at ../nptl/sysdeps/unix/sysv/linux/x86_64/sem_wait.S:44
#1  0x00007ffff74049d7 in lcdtime_HD44780_senddata (p=p@entry=0x631d00,
displayID=displayID@entry=0 '\000', flags=flags@entry=1 '\001',
ch=ch@entry=48 '0') at hd44780-ext8bit.c:153 #2  0x00007ffff7404c7f in
hd_init_ext8bit (drvthis=0x630810) at hd44780-ext8bit.c:112 #3
0x00007ffff74022a1 in HD44780_init (drvthis=0x630810) at hd44780.c:373
#4  0x00000000004109a0 in driver_load (name=name@entry=0x6220d0
"hd44780", filename=filename@entry=0x6307d0
"/usr/lib/x86_64-linux-gnu/lcdproc/hd44780.so") at driver.c:153 #5
0x000000000040fddf in drivers_load_driver (name=0x6220d0 "hd44780") at
drivers.c:85 #6  0x0000000000407df5 in init_drivers () at main.c:670
#7  0x000000000040635b in main (argc=<optimized out>, argv=<optimized
out>) at main.c:2
```

I'm not sure whether this problem comes from lcdproc ot the way this package is built with Debian.
For reference, here's [Debian build log on i386](https://buildd.debian.org/status/fetch.php?pkg=lcdproc&arch=i386&ver=0.5.7-7&stamp=1458674995).

Do you have an idea on how to solve this problem ? (the frist that come to mind is to rename lcdproc's `sem_wait` function to avoid this collision, but a simpler solution may exist).

All the best

haraldg, 2016-12-29:
That's strange. The linker should prefer local functions over system functions, however the LCDd module implementation is strange. Maybe we are confusing the linker at some point.

Also this semaphore code is pretty ugly and nobody dared to touch it for about 15 years. To top things off it seems entirely pointless: The only driver using it is not multithreaded in any way I can see.

I feel tempted to just cut this out completely. Luckily the issue was reported before I announced the release candidate for the upcoming release.

closed 2016-12-29 by haraldg

benallard, 2016-12-30:
I confirm the segfault no longer occurs with that patch. Unfortunately, the "8bit" connectionType is not the one that is compatible with my LCD, but rather the "lcm162". So I'm not very much interested anymore in that patch. The good news is, I'm working on a documentation update for that lcm162 connectionType :grinning: .

commit 4d5509a4 (2016-12-29): hd44780: removed outdated semaphore code

The code is broken. The driver is not multithreaded so the code was pointless.
This commit closes #34.