lcdproc/lcdproc #36: tarball of 0.5.7 from sourceforge is not identical with tarball from github

benallard, 2016-12-30:
Not sure how critical the issue is, but it is at least pretty confusing that there seems to be two different 0.5.7 releases. On sourceforge, we have the one from 2014 with the following sha1 (provided by sourceforge themselves) eacc65c5f16237beb18b3cdb4835ba0a57f5f811, and by browsing the releases (well, tags) here on that github page, there is another one dated July 2016.

Unfortunately, even the content of the tarballs is slightly different ...
Just renaming the tag would lift-off the ambiguity ...

haraldg, 2016-12-30:
Actually this this kind of expected behaviour: The 0.5.7 release you see on github is just the output of git-archive for the tag. The tarball on sourceforge is the real release tar ball with proper configure script as produced by "make distcheck". At this time I'm not sure what we can do about this.

0.5.8 will be the first release done on github. Maybe once I'm familiar with the process, I will have an idea.

benallard, 2017-01-02:
The main issue is that github is publishing tarballs that looks 'official', so if you don't intend to have those tarballs considered as such, maybe you can switch off the github functionality, or use a different tag naming scheme. At least for the time being.