lcdproc/lcdproc #39: Debian files in script directory are outdated

dod38fr, 2017-01-04:
Hello

The Debian files and information given in `scripts` directory are outdated. I suggest to remove them to avoid confusion.

All the best

haraldg, 2017-01-05:
Sounds reasonable, nobody cared about these files for about 8 years, while the debian package is well maintained. However I note, that the local debian files were added in 2006 when lcdproc had already been in debian for many years. This is an unusual thing to do... Anybody knows what the reason was?

Do you mind if I delay this issue until after the release?

dod38fr, 2017-01-05:
> Do you mind if I delay this issue until after the release?

no problem

haraldg, 2017-01-06:
@marschap You worked on the debian files. Maybe you want to comment on this issue?

marschap, 2017-01-09:
Hi,

this was 10 years ago. I don't remember the curcumstances - maybe I wanted to make it easier for me to create a .deb from upstream LCDproc.

In the mean time i learned that newer Debian packaging tools ignore the debian/ directory in upstream tarballs / repos. With this knowledge I would not repeat the effort.

I see the followig options:
* copy debian/ from Debian's lcdproc git repo (and repeat whenever necessary)
  This helps to creating debian-cimpliant packages directly from upstream
* replace contents of debian/ with a document how to create a .deb by using both
   lcdproc git repos: upstream + Debian.
   Open: find someone to do it (not me)
* simply drop debian/

Best
Peter

willfe, 2017-01-09:
If we just drop debian/ entirely, I take it that just leaves it up to the Debian maintainer to package up LCDproc for each build/release then?

dod38fr, 2017-01-10:
Yes. That's what I've been doing for years for lcdproc and with other packages.

commit 719c73e6 (2017-01-22): remove scripts/debian

as per request of the debian maintainer. Nobody took care for it for a long
time and the official debian package is well maintained.

This closes #39.
closed 2017-01-22 by haraldg