lcdproc/lcdproc #40: new driver futuba fails to compile from the release tarball

haraldg, 2017-01-04:
@dod38fr writes on the mailing list:
> I'm using rc1 to build a new Debian package for lcdproc.
>
> Unfortunately, compilation fails:
> 
> gcc -DHAVE_CONFIG_H -I. -I../..  -I../.. -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Wall -g -O2 -fdebug-prefix-map=/home/domi/debian-dev/build-area/lcdproc-0.5.8~rc1=. -fstack-protector-strong -Wformat -Werror=format-security -Wno-unused-function -c -o futaba-futaba.o `test -f 'futaba.c' || echo './'`futaba.c
> futaba.c: In function ‘futaba_start_driver’:
> futaba.c:435:26: error: ‘my_driver’ undeclared (first use in this function)
>  if (usb_claim_interface(my_driver->my_handle, 0) < 0) {
>                          ^~~~~~~~~
> futaba.c:435:26: note: each undeclared identifier is reported only once for each function it appears in

@ajw107 Can you look into this?

closed 2017-01-16 by haraldg

ajw107, 2017-01-17:
Apologies for the delay, getting over the dreaded christmas flu.  I'm not entirely sure how that error crept in there as I compiled it both with libusb 0.1 and 1.0, but thank you harold for supplying the fix.  On anohter note, now I'm back to almost normal health, I'll have alook at the rest if of ther code I promised to look at.  Also the fix to get everything working in the linux kernel is now in (as of kernel version 4.10).

haraldg, 2017-01-17:
Alex Wood writes:
> On anohter note, now I'm back to almost normal health, I'll have
> alook at the rest if of ther code I promised to look at.

Great, no hurry though. This will only be merged after the release
anyway.

> Also the fix to get everything working in the linux kernel is now
> in (as of kernel version 4.10).

I'm probably cutting rc2 later today, but I guess documentation updates
between the (hopefully) final rc and the actual release should be fine.
So if you are fast you can still make it ... :)

Thanks for your help!
Harald

commit 13379aee (2017-01-16): futaba: Fix compilation of libusb0.1 code

This fixes #40.