lcdproc/lcdproc #41: Some drivers are not compiled with libusb1

dod38fr, 2017-01-06:
lcdproc compiles fine when on my laptop where both old and new libusb are installed,

However, the packaging fails in a chroot where **only** libusb-1.0-0-dev and libftdi1-dev are installed:
```
$ grep -E 'lib(usb|ftdi)' /tmp/lcdproc.log |tail -25
Selecting previously unselected package libusb-1.0-0:amd64.
Preparing to unpack .../079-libusb-1.0-0_2%3a1.0.21-1_amd64.deb ...
Unpacking libusb-1.0-0:amd64 (2:1.0.21-1) ...
Selecting previously unselected package libftdi1-2:amd64.
Preparing to unpack .../080-libftdi1-2_1.3-2_amd64.deb ...
Unpacking libftdi1-2:amd64 (1.3-2) ...
Selecting previously unselected package libusb-1.0-0-dev:amd64.
Preparing to unpack .../081-libusb-1.0-0-dev_2%3a1.0.21-1_amd64.deb ...
Unpacking libusb-1.0-0-dev:amd64 (2:1.0.21-1) ...
Selecting previously unselected package libftdi1-dev.
Preparing to unpack .../082-libftdi1-dev_1.3-2_amd64.deb ...
Unpacking libftdi1-dev (1.3-2) ...
Setting up libusb-1.0-0:amd64 (2:1.0.21-1) ...
Setting up libusb-1.0-0-dev:amd64 (2:1.0.21-1) ...
Setting up libftdi1-2:amd64 (1.3-2) ...
Setting up libftdi1-dev (1.3-2) ...
        --enable-libusb \
        ./configure --build=x86_64-linux-gnu --prefix=/usr --includedir=\${prefix}/include --mandir=\${prefix}/share/man --infodir=\${prefix}/share/info --sysconfdir=/etc --localstatedir=/var --disable-silent-rules --libdir=\${prefix}/lib/x86_64-linux-gnu --libexecdir=\${prefix}/lib/x86_64-linux-gnu --disable-maintainer-mode --disable-dependency-tracking --enable-stat-nfs --enable-stat-smbfs --enable-drivers=all,\!irman,\!svga --enable-libusb --enable-seamless-hbars --enable-testmenus --enable-permissive-menu-goto --enable-lcdproc-menus
checking if libusb support has been enabled... yes
checking if libusb-1-0 support has been enabled... yes
checking if libftdi support has been enabled... yes
configure: WARNING: The futaba driver needs the libusb library.
configure: WARNING: The IOWarrior driver needs the libusb library.
configure: WARNING: The picolcd driver needs the libusb library.
configure: WARNING: The shuttleVFD driver needs the libusb library.
```
packaging fails with:
```
$ grep missing /tmp/lcdproc.log 
dh_install: lcdproc-extra-drivers missing files: usr/lib/*/lcdproc/IOWarrior.so
dh_install: lcdproc-extra-drivers missing files: usr/lib/*/lcdproc/i2500vfd.so
dh_install: lcdproc-extra-drivers missing files: usr/lib/*/lcdproc/lis.so
dh_install: lcdproc-extra-drivers missing files: usr/lib/*/lcdproc/picolcd.so
dh_install: lcdproc-extra-drivers missing files: usr/lib/*/lcdproc/shuttleVFD.so
dh_install: lcdproc-extra-drivers missing files: usr/lib/*/lcdproc/ula200.so
dh_install: missing files, aborting
```

All the best

haraldg, 2017-01-06:
Some drivers not being available, when the old libusb is not available, is no surprise as nobody cared to work on #13. However lot's of the affeced drivers support both versions, so something is clearly wrong at the configure level.

Looking at the log you provided it seems one of the problems is, that you called configure with `--enable-libusb` which enables the old libusb. The new libusb is enabled with `--enable-libusb-1-0`.

haraldg, 2017-01-06:
Ok, so looking at this in more detail, it seems that you can't enable libusb1.0 without also enabling legacy libusb. This is bad.

Oddly, I did an openwrt package of lcdproc with only new libusb, and everything looks fine there. Not sure what the difference is at the moment. Also I can't really figure out your log file. It says:
```
checking if libusb support has been enabled... yes
checking if libusb-1-0 support has been enabled... yes
```
and later
```
configure: WARNING: The futaba driver needs the libusb library.
configure: WARNING: The IOWarrior driver needs the libusb library.
configure: WARNING: The picolcd driver needs the libusb library.
configure: WARNING: The shuttleVFD driver needs the libusb library.
```

I guess that means that the configure script is broken.

haraldg, 2017-01-06:
I think I figured it out. The lines, where it disables libusb again, are in upper case - thus missing from the log. I will push a fix, after a few tests.

dod38fr, 2017-01-06:
Indeed. Here's a new excerpt from the log that shows the uppercase LIBUSB:
```
$ grep -i -E 'lib(usb|ftdi)' /tmp/lcdproc.log |tail -25
Selecting previously unselected package libftdi1-2:amd64.
Preparing to unpack .../080-libftdi1-2_1.3-2_amd64.deb ...
Unpacking libftdi1-2:amd64 (1.3-2) ...
Selecting previously unselected package libusb-1.0-0-dev:amd64.
Preparing to unpack .../081-libusb-1.0-0-dev_2%3a1.0.21-1_amd64.deb ...
Unpacking libusb-1.0-0-dev:amd64 (2:1.0.21-1) ...
Selecting previously unselected package libftdi1-dev.
Preparing to unpack .../082-libftdi1-dev_1.3-2_amd64.deb ...
Unpacking libftdi1-dev (1.3-2) ...
Setting up libusb-1.0-0:amd64 (2:1.0.21-1) ...
Setting up libusb-1.0-0-dev:amd64 (2:1.0.21-1) ...
Setting up libftdi1-2:amd64 (1.3-2) ...
Setting up libftdi1-dev (1.3-2) ...
        --enable-libusb-1-0 \
        ./configure --build=x86_64-linux-gnu --prefix=/usr --includedir=\${prefix}/include --mandir=\${prefix}/share/man --infodir=\${prefix}/share/info --sysconfdir=/etc --localstatedir=/var --disable-silent-rules --libdir=\${prefix}/lib/x86_64-linux-gnu --libexecdir=\${prefix}/lib/x86_64-linux-gnu --disable-maintainer-mode --disable-dependency-tracking --enable-stat-nfs --enable-stat-smbfs --enable-drivers=all,\!irman,\!svga --enable-libusb-1-0 --enable-seamless-hbars --enable-testmenus --enable-permissive-menu-goto --enable-lcdproc-menus
checking if libusb support has been enabled... yes
checking for LIBUSB... no
checking if libusb-1-0 support has been enabled... yes
checking for LIBUSB_1_0... yes
checking if libftdi support has been enabled... yes
checking for LIBFTDI... no
configure: WARNING: The futaba driver needs the libusb library.
configure: WARNING: The IOWarrior driver needs the libusb library.
configure: WARNING: The picolcd driver needs the libusb library.
configure: WARNING: The shuttleVFD driver needs the libusb library.
```

haraldg, 2017-01-06:
The only drivers using the new libusb are picolcd and futaba (both have code for the old libusb as well). If you switch the debian package to the new libusb exclusively, all other libusb based drivers and sub-drivers will get disabled. @dod38fr I wonder if there is a strong reason against build-depending on both libusb APIs at the same time? (Of course the autoconf issue will be fixed anyways.)

closed 2017-01-06 by haraldg

dod38fr, 2017-01-07:
Debian project is trying to [move away from old libusb](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=810428) and [libftdi](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=810373) (Wow, it's already been a year since these bugs were entered).

Since Debian is entering freeze, I cannot remove support for driver that use exclusively old libusb. Once next Debian (code name Stretch) is released, I'll remove support for these drivers and see if people complain (or provide patches).

So, for the time being, I'll try enabling both libusb1 and old libusb (likewise for libftdi).

Thanks for your help

haraldg, 2017-01-07:
Great. Just to point out: I think lcdproc doesn't have any drivers using new libftdi. (Unless you plan to submit the patches you tried last year. I'd merge them, if they are ready in time for rc2.)

dod38fr, 2017-01-07:
Thanks for the hint. I'll remove the dependency on libftdi1. 

I think the patches I got in March are not enough: If I understand correctly lis driver must also be moved to libusb1 (because libfdti1 one uses libusb1)

Otherwise, the rc1 package is working fine on Debian unstable, although I can test only with imonlcd. I'm going to upload this package to Debian/experimental

Otherwise, I think the mailing list is down. 2 of my mails have not showned up yet.