lcdproc/lcdproc #43: GPIO list in hd44780-rpi.c for rev 2 boards incorrect

mchack23, 2017-01-07:
Hi,
first time to use github (no coder myself yet) so I'm sorry I don't know how to fork and bugfix and submit a pull request for it, so I'm just telling you via an issue that there's a small error in the hd44780-rpi.c:
When I installed lcdproc it told me via syslog that:

"LCDd: check_pin: Use of GPIO pin 2 not allowed" 

which is simply not true for rev 2 boards (see http://elinux.org/Rpi_Low-level_peripherals#GPIO_hardware_hacking")

so a small change in hd44780-rpi.c line 91 is necessary:
from: -1, -1, -1, 3, 4, -1, -1, 7,
to:     -1, -1, 2, 3, 4, -1, -1, 7,

after which it worked perfectly fine.

haraldg, 2017-01-07:
@marschap You did the last work on rpi. Can you look into this?

marschap, 2017-01-08:
According to what I was able to look up on the internet, mchack23 is right.
I consider this issue more an enhancement request than a bug.
With GPIO2 allowed on Rev2 boards, the users have more degrees of freedom
whenhey have additional HW (beyond and LCD) connected to the GPIO pins.
Please find the (trivial) patch attached:
[rpi-GPIO2.txt](https://github.com/lcdproc/lcdproc/files/692419/rpi-GPIO2.txt)

commit cb627bb6 (2017-01-16): hd44780-rpi.c: Allow pin 2 on raspberry pi rev. 2 baords

This closes #43.
closed 2017-01-16 by haraldg