lcdproc/lcdproc #45: LCDd floods syslog with critical messages

dod38fr, 2017-01-08:
In [bug 806800](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=806800), a Debian user reports:
```
LCDd seems to send critical syslog messages on some weird
situations. I don't know why. Here's an example of a tail on a
logfile:

Message from syslogd@marcos at Dec  1 10:16:06 ...
 sten L

Message from syslogd@marcos at Dec  1 10:16:06 ...
 nore L
déc 01 10:16:05 marcos LCDd[11778]: ignore I
déc 01 10:16:05 marcos LCDd[11778]: listen NT
déc 01 10:16:05 marcos LCDd[11778]: ignore NT
déc 01 10:16:05 marcos LCDd[11778]: listen M
déc 01 10:16:05 marcos LCDd[11778]: ignore M
déc 01 10:16:05 marcos LCDd[11778]: listen L
déc 01 10:16:05 marcos LCDd[11778]: ignore L

Notice how you got `Message` lines above? Those flood *all* terminals
I am connected to - not just the one where i was tailing the
logfile. oddly enough, part of that message is stripped as well
("listen" becomes "sten" and "ignore" becomes "nore", not sure why).
```

The main issue is that critical messages are issued by LCDd. Anarcat mentions also:
```
I wonder if this isn't connected to the vsyslog call:

  vsyslog(LOG_USER | (level + 2), format, ap);

This seems to bump up the level, which could distort the levels...
```

He may be on to something.

All the best

haraldg, 2017-01-08:
On 08.01.2017 15:53, Dominique Dumont wrote:
> In [bug
> 806800](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=806800), a
> Debian user reports:

Looking at the bug report, it seems that LCDd logs to syslog because
it was started with `-s 1`

> ```
> LCDd seems to send critical syslog messages on some weird
> situations. I don't know why. Here's an example of a tail on a
> logfile:
>
> Message from syslogd@marcos at Dec  1 10:16:06 ...
>  sten L [...]

Here I don't fully understand what he says: He is getting a number of
valid listen/ignore messages to syslog with the proper level, but
occasionally there are critical messages, that are mangled?

In this case there seems to be some serious corruption of data
structures, i.e. something not obvious from reading the source code.

I think the only way forward is to run LCDd from within a debugger.

> ```
> I wonder if this isn't connected to the vsyslog call:
>
>   vsyslog(LOG_USER | (level + 2), format, ap);
>
> This seems to bump up the level, which could distort the levels...
> ```

While the code is clearly fishy (non-portable) it is actually correct
at least on debian systems.

HTH,
Harald

-- 
If you want to support my work:
see http://friends.ccbib.org/harald/supporting/
or donate via peercoin to P98LRdhit3gZbHDBe7ta5jtXrMJUms4p7w
or CLAM xASPBtezLNqj4cUe8MT5nZjthRSEjrRQXN

haraldg, 2017-01-14:
On Sunday, 8 January 2017 21:40:56 CET Harald Geyer wrote:
> While the code is clearly fishy (non-portable) it is actually correct
> at least on debian systems.

Do you know why anarcat got logs at "critical" level ? (which is indeed a bit 
overboard)

All the best

-- 
 https://github.com/dod38fr/   -o- http://search.cpan.org/~ddumont/
http://ddumont.wordpress.com/  -o-   irc: dod at irc.debian.org

haraldg, 2017-01-14:
On 14.01.2017 11:49, Dominique Dumont wrote:
> Do you know why anarcat got logs at "critical" level ? (which is
> indeed a bit overboard)

Too me his problems look like corruption of some data structures or
the stack or whatever. This could be caused by a buffer overflow
somewhere or tempering with the binary or a number of other things.
I don't think there is any way to narrow this down other then
running LCDd inside a debugger...

HTH,
Harald

-- 
If you want to support my work:
see http://friends.ccbib.org/harald/supporting/
or donate via peercoin to P98LRdhit3gZbHDBe7ta5jtXrMJUms4p7w
or CLAM xASPBtezLNqj4cUe8MT5nZjthRSEjrRQXN

haraldg, 2017-02-16:
Closing this issue as it relates to an old, unsupported version of lcdproc and apparently isn't reproducible with the old version either. If anybody sees this with 0.5.8 or newer, please reopen and provide additional information.

closed 2017-02-16 by haraldg