lcdproc/lcdproc #49: Feature Request - Parallel port - RPi GPIO compatibility wrapper?

JamesGKent, 2017-01-18:
I'm looking to use a graphical LCD (with a T6963 driver) on a raspberry pi, however it seems that the only driver for this is written to use the parallel port. i know that the HD44780 type displays have had the driver adapted to use the GPIO pins of the pi with the correct connection type set in the driver config.
Given that a lot of the drivers are written to use either USB or a parallel port i'm wondering if it would be possible to write a compatibility wrapper to translate parallel port calls to GPIO in a common way to prevent having to rewrite each driver to use the GPIO?

I would attempt to adapt the driver i need but i'm a novice with C/C++ and having had a look at the driver files i'm not even sure where to begin...

haraldg, 2017-01-18:
Having a more modulare architecture of lcdproc and a hardware abstraction layer for the different kinds of ports would be possible and actually a very good thing. However, I don't think anybody is working on that.

If you want to work on a new connection type for the T6963 driver have a look at t6963_low.c which is just above 200 lines of code. Also use hd44780-gpio.c as a reference how to use gpios.

Since this is a graphical lcd it would also be reasonable to write a driver for a linux framebuffer or fbtft device. Of course submitting a kernel driver might be a daunting task for somebody new to C and just writing a replacement or improvement for t6963_low.c will be a lot less frustrating.

JamesGKent, 2017-02-10:
further to your comment about the unmaintained patches, and your comment about using ppdev as a base for a hardware abstraction wrapper.
I was thinking that it should be possible to write a generic wrapper that emulates a parallel port (for unmodified drivers) and maps the pins present in a parallel port to the linux sysfs gpio interface, that way the wrapper would work for more than just the RPi GPIO and be more useful.
however I can't see a clean way to do this without defining a new section in the config such as:
```
[output]
# valid values: parallel, gpio
Type=gpio

# valid only when Type=parallel
Address=0x3F

# valid only when Type=gpio
# used to map parallel port pins to GPIO pins
pin_P_Strobe=1
pin_P_D0=2
pin_P_D1=3
pin_P_D2=4
pin_P_D3=5
pin_P_D4=6
pin_P_D5=7
pin_P_D6=8
pin_P_D7=9
pin_P_Ack=10
pin_P_Busy=11
pin_P_PaperOut=12
pin_P_Select=13
pin_P_Linefeed=14
pin_P_Error=15
pin_P_Reset=16
pin_P_SelectPrinter=17
```

and then using a different constructor for drivers that are rewritten to use it specifically so that the pins can be directly assigned by the driver rather than remapping them from the parallel port.

My thinking was it would provide a big benefit to some drivers as they were originally written for the parallel port and as such the version that targets parallel has more/better developed features, E.G. the hd44780 driver in 4bit mode on the parallel port can run up to 7 displays, whereas the RPi version can do 2, and the GPIO version can currently only run 1.

any thoughts on this?
specifically if you can think of a cleaner way to implement this?
I was thinking it would be nice to keep all pin assignments in the output section, but not sure how feasible that is, or how it would play nicely with having multiple drivers enabled (assuming you can?)

haraldg, 2017-02-10:
I think your toughts go in the right direction.

Yes, a new config section will be necessary. No problem there. LCDd already can load multiple drivers at the same time (think input and output) - probably one can introduce a new kind of HAL driver, that provides a generic ABI to access pins. (E.g. hal-gpio and hd44780-hal would be used toghether). If loading this HAL stuff dynamically is too complicated, we could also just link it statically into LCDd - this would still be a tremendous improvement to the status quo.

However all parport based drivers would need either porting to the new HAL API or a new HAL connection type. The nice thing about the ppdev stuff is, that somebody tried to support a new interface without changeing every driver (if I understood that part correctly). I'm not sure if this is a sane approach. Maybe if somebody worked out the HAL layer, people would port the (popular) drivers to it anyway.

Yes, our current state with zillions of drivers/CTs all supporting slightly different sets of features, is insane. If somebody started working on cleaning this up, this would be just awesome!

I haven't thought about all these things in detail, but my gut feeling about the way forward is:
1. Figure out a LCDd internal API that is generic enough to allow manipulations of pins on every HW we care about and at the same time abstract enough that the driver doesn't need to know anything about the HW. Maybe this is easy. Maybe it is not.
2. Figure out how to represent the necessary data in the configuration file without driving everybody crazy.
3. Write all the common HAL code and at least one HW interface (probably the GPIO one as this seems where the world is moving to anyway).
4. Ask people to start porting drivers.

Does this make sense to you? others?

JamesGKent, 2017-02-10:
I would have to agree, trying to wrap old drivers without having to rewrite/port them would be nice, but could be severely limiting in other ways.

I suppose a good starting point would be to try and work out the pin requirements of each driver and the notations used to see how many are common (in name if nothing else)
I suspect a lot of drivers will at the very least use the same notation for the data pins (D0-D7) and many will have RW, and EN pins (not sure how many will have a RS)

but as far as I can see there are several basic connections types that are analogous to parallel port/gpio:
- parallel
- gpio (sysfs gpio interface)
- gpio (RPi type - as raspbian now includes the sysfs system probably not worth implementing unless requested)
- usb FTDI FT2232 or FT245

Of these I would suggest the sysfs gpio is the most capable/flexible due to having a greater availability of pins on most platforms where it is available. the parallel port will have fewer pins available but still more than enough to be useful. i think trying to write a HAL without targeting more than one output would be futile, so I would suggest starting with the GPIO and parallel port.

I suppose one advantage of baking the HAL into LCDd instead of each driver would be a reduced total size for systems that have multiple drivers installed, but it may need to be written in such a way that more than one driver can instantiate/initialize it if LCDd is to retain the ability to run more than one driver (not that i can find any guides on how to do this...?)

haraldg, 2017-02-10:
> i think trying to write a HAL without targeting more than one output would be futile, so I would suggest starting with the GPIO and parallel port.

Sure, the list above was meant as whan I think one person needs to commit to doing, to get this actually started. In theory more outputs could be implemented by different people.

> I suppose one advantage of baking the HAL into LCDd instead of each driver would be a reduced total size for systems that have multiple drivers installed

Yes, and it is also the much clearer approach. In theory, if we have the same symbol in multiple shared objects (drivers) ELF would resolve to just use the local versions. But everybody loading this mess into a debugger will be very confused. And static linking several drivers into LCDd would probably be dangerous.

> but it may need to be written in such a way that more than one driver can instantiate/initialize it if LCDd is to retain the ability to run more than one driver

I expect this to be reasonably easy. If I'm wrong, we can still revisit this issue later.

However, if you want to work on this HAL thing: I don't want to micromanage your work. If you think you see a way forward, feel free to pursue it. I shall be happy with (almost) any approach that kills this huge amount of duplicate code, we have. ;)

> (not that i can find any guides on how to do this...?)

Just enumerate them:
```
Driver=Olimex_MOD_LCD1x9
Driver=linux_input
```

haraldg, 2017-02-17:
@JamesGKent are you working on this?

Anyway, I had a few ideas for this. First I realized that having the pinout configurable, we not only can share the low-level HW access between drivers, but also get rid of the many subtypes that just exist for the different wiring variants out there. In the end LCDd.conf would be some kind of HW database. I'm not sure if configuration and HW description should be in one file, but it seems reasonable at them moment as there is no real difference. With that in mind it becomes clear, that we will have lots of sections describing different wirings in the configuration file, thus the section names themselves need to be arbitrary. Secondly I realized there always will be drivers, that don't use the HAL - think of displays natively connected to some bus like i2c, the curses driver, etc. - thus the HAL stuff probably doesn't belong into the server core.

With that in mind, I came up with the following configuration example - assuming the 8bit CT has been ported to HAL and ommitting not relevant settings.
```
[server]
Driver=hd44780

[hd44780]
ConnectionType=8bit
HAL=hd44780-lcdtime

[hd44780-lcdtime]
HALDriver=parport
Port=0x378
Pin_D0=D0
Pin_D1=D1
Pin_D2=D2
Pin_D3=D3
Pin_D4=D4
Pin_D5=D5
Pin_D6=D6
Pin_D7=D7
Pin_EN=INIT
Pin_RS=nSTRB

[hd44780-winamp]
HALDriver=parport
Port=0x378
Pin_D0=D0
Pin_D1=D1
Pin_D2=D2
Pin_D3=D3
Pin_D4=D4
Pin_D5=D5
Pin_D6=D6
Pin_D7=D7
Pin_EN=nSTRB
Pin_RS=INIT
```
where the `Pin_??` strings would make sense to the hd44780-8bit driver and the values on the right side make sense to the hal-parport driver.

At a later stage we could add a knew configuration key `Setup` as syntactic sugar, to allow something like:
```
[server]
Setup=fooappliance

[fooappliance]
Driver=hd44780
ConnectionType=8bit
HAL=fooappliance
HALDriver=parport
Port=0x378
Pin_D0=D0
Pin_D1=D1
Pin_D2=D2
Pin_D3=D3
Pin_D4=D4
Pin_D5=D5
Pin_D6=D6
Pin_D7=D7
Pin_EN=INIT
Pin_RS=nSTRB
```

The hd44780-8bit driver would do something like:
```
#define PIN_D0 0x1
#define PIN_D1 0x2
...
#define PIN_EN 0x100
#define PIN_RS 0x200

// load the hal driver
privateData->hal = hal_open(config_get_string("HAL"));
// tell the hal driver our mapping from bit mask to pins
hal_request_pin(privateData->hal, "Pin_D0", PIN_D0);
...
hal_request_pin(privateData->hal, "Pin_RS", PIN_RS);

hal_set(privateData->hal, PIN_EN | PIN_RS); // setting selected pins
hal_clear(privateData->hal, PIN_BL); // clearing selected pins
hal_output(privateData->hal, PIN_D0 | PIN_D3 | PIN_D4); // updating all pins
```
Since the driver has full freedom to choose bit masks for pins, it can easily arange for things like `hal_output(privateData->hal, 's') to just work. There could also be additional functions to just affect a certain number of lower bits.

What do people think about this proposal?

I haven't thought about handling input pins yet.

JamesGKent, 2017-02-17:
I haven't been working on it as such, like you I was taking some time to think about the best way to achieve this, although my line of though had gone to something a bit different, I know that some devices will always be on their own drivers directly controlling their own hardware, such as those that connect via USB that aren't something like the FTDI devices. And as such trying to wrap everything 

what you are proposing for the HAL configuration does seem a good approach, I was just wondering if there was a way to have the driver itself set up the pin configuration first (similar to the different connection types set in the driver now) and if needed, for gpio etc... able to override those settings in the HAL configuration?

haraldg, 2017-02-17:
James writes:
> what you are proposing for the HAL configuration does seem a good approach, I was just wondering if there was a way to have the driver itself set up the pin configuration first (similar to the [...]

On one hand I'm wondering if we really want the HW database in the
configuration file. Having it in a dedicated database and compiling it
somehow into the drivers looks nice. But having the same thing in two
places also makes things more complicated and it seems the current
practice of lcdproc is to set the defaults in the configuration file
anyway.

OTOH my examples have a lot of trivial lines like `Pin_D0=D0`. I guess
this could be shortened by a fall back rule in hal_request_pin() like
"if config_get_string() turns up empty, just try if the key or the
key minus the leading `Pin_` are valid pins of the HW interface.
But that's syntactic sugar, not the driver actually setting up anything.

Finally there is of course the question whether we want to have some
backwards compatibility. Ie. if a driver is ported to use the HAL and
encounters a configuration without a `HAL=` setting, should it be able
to make sense of the old style configuration and somehow get the HAL to
do the right thing. I think this is low priority and at the same time
it would make things a lot more complicated, because it's more than just
some default setting: It's two independent ways to parse configuration.
My gut feeling at the moment is, that we shouldn't do that.

-- 
If you want to support my work:
see http://friends.ccbib.org/harald/supporting/
or donate via CLAM to xASPBtezLNqj4cUe8MT5nZjthRSEjrRQXN
or via peercoin to P98LRdhit3gZbHDBe7ta5jtXrMJUms4p7w

JamesGKent, 2017-02-20:
this is party why i wanted to discuss this here before even strating to try and write any code, i think it would be very easy to start down a path writing something thats clunky/complicated/confusing to configure etc and difficult to clean up after the fact.
something like a HAL should be architected from the get go, rather than trying to clean it up afterward, and at the moment the best way i can think of to do that (and keep it relatively simple to set up) would be to keep the pin configuration flexible to some degree, I.E. for something like the hd44780 driver you would retain the `connectiontype` parameter, which would call into the HAL and set up the default pin numbers (eg for the parallel port) which the HAL could then alter if there are any override options set.

this way the existing configuration need not be update when/if this change rolls through, there wouldn't be a huge database of different configurations residing in the core of LCDd (which i don't think there should be) while still offering a relatively simple interface for configuration if using a different port (eg gpio)
 so for each pin the driver needs it would call something like:
`hal_register_output_pin(privateData->hal, "Pin_D0", PIN_D0);`

which would map to D0 on the parallel port normally, and if in the HAL config we had something like:
```
[HAL]
outputtype=gpio
pin_D0=17
...
```

it would instead register gpio pin 17 as an output and map any signals to that pin...
but if no redirection is needed then the defaults (which would reside in the driver itself) would seamlessly apply.

I know it may seem like wrapping everything twice, but I think that it offers a good compromise between flexibility, capability and ease of configuration.

any thoughts?

haraldg, 2017-02-20:
Can you elaborate on this a bit more, maybe provide a detailed example of how this would work, like I tried above?

In particular: What would be the difference between your `hal_register_output_pin()` and my `hal_request_pin()`? Where is `PIN_D0` defined? How would PIN_BL, PIN_EN, PIN_RW be handled? Would there be any semanics attached to strings like `"Pin_D0"` - if yes, who would interpret them? Which code would be handling which parts of the configuration?

Of course the hd44780 driver will retain connection types (I have them in my example above too), after all the 4bit protocol and the 8bit protocol actually are different connection types. But at the moment we have three different 8bit connection types, all for the parallel port and only differing in wiring AFAICT. (And at least one completely independent implementation of the hd44780, also for the parallel port.) So at the moment we already have a hardware database, implemented by duplicating a huge amount of code... :(

How would we handle this three 8bit connection types in your scenario?

JamesGKent, 2017-02-20:
sorry, was written in a hurry, should probably have left it till i had a bit more time to elaborate, in essence the driver would call in all the default pin allocations it needs with an ident, this ident would then be used to look at the HAL section of the config file for a matching allocation so the pin number could be overridden if needed, as this would prevent having to write the HAL to know all the possible pin names that may be assigned by drivers.
then as long as the functionality of the code remains the same we would only need 8 bit code and 4 bit code for the hd44780 which simply calls into the HAL with the relevant preset pins, thus condensing the 3 8 bit drivers into one, keeping the 4 bit separate where needed.

I'm not saying that this would do away with the database of configurations, but they would now be able to live in a smaller number of files and if planned right could even keep all the possible preset configurations of a driver in a single file, with the relevant setup function called on driver initialization.

I called it output pin partly to allow for expansion so that input pins could be added later, though I suppose this could instead be done by using an additional argument instead of a whole separate function (though I would argue to keep them separate)

As for the defines PIN_D0 etc. I would suggest these would be defined inside the driver and then used as the pin ident when calling into the HAL to set pin states, allowing the driver for a display type to use the naming convention that matches the display datasheet without having to duplicate anything in the HAL (eg look up D0 and try DB0 if it fails)

as for configurations without a HAL setting, this is where the driver calling the defaults first would come in (with the settings predefined in the driver) so e.g. the config for a hd44780 is:
```
[hd44780]
connectionType=4bit
```
in which case the new HAL enabled hd44780 driver would call into the HAL to use the parport with the normal pin assignments, effectively wrapping the old configuration and should allow people to do a simple update on existing systems without breaking anything.

the more times I reread what you put the more it seems like we are thinking along similar lines with regard to pin setup, however i note that in your example you have the pin defines in the hd44780 driver, but use the values when calling into the HAL, how would the HAL know what to do with these values?
am i right in thinking you intended the same as me in that they are just a simple ident for that pin and the HAL uses that number only to keep track of which pin to set/clear?

haraldg, 2017-02-20:
I believe we are thinking along very similar lines, with the main difference being, that you try to stay backwards compatible by means that I haven't fully understood yet. Can you provide a simple (pseudo)code example of a hd44780-8bit driver, showing configuration access and HAL-calls necessary to implement what you have in mind?

> am i right in thinking you intended the same as me in that they are just a simple ident for that pin and the HAL uses that number only to keep track of which pin to set/clear?

Yes, exactly. Maybe I have been over-complicating my example by discussing how some rough edges could be smoothed by syntactic sugar. Just ignore the fooappliance-stuff, it's not important for discussing the HAL API.

Maybe also `hal_request_pin()` was a bad name. Does `hal_setup_pin_mapping()` make more sense?

In my example 
* `hal_open(config_get_string("HAL"))` would and load the configured HALDriver (ie. parport, ugpio) and initialize things
* `hal_request_pin()` would setup a mapping between local bits (that make sense to the driver) and pins that make sense to the HAL. This probably is just a wrapper calling a function of the HALDriver, which in turn would access configuration.
* the driver uses bit set/clear functions to tell the HAL what it should do

Is it clearer now?

JamesGKent, 2017-02-20:
yes thats a lot clearer, i'm sure i didn't help by reading some of your posts while not fully awake yet...

so for clarity all the drivers would have to be rewritten to make use of the HAL but the new driver would do something like (with the following config):
```
[hd44780]
connectionType=8bit 
```

so hd44780 driver would see the connection type and call a function something like:
```
void map_pins_8bit(...){
    hal_request_pin(... "pin_D0", PIN_D0);
    ...
    hal_request_pin(... "pin_D7", PIN_D7);
    hal_request_pin(... "pin_BL", PIN_BL); // you get the idea for each pin
    ...
}
```

and inside the function `hal_request_pin` would be a lookup to the config file for an entry with the same keyname.
e.g. if the HAL section looked like:
```
[HAL]
output=gpio
pin_BL=17
```

Then the pin number in the config file would take precedence over the value passed in by the driver.
this would have to be a little more complicated than this example initially appears for several reasons:

1. if output type does not match that set by driver then pin assignments will likely not match either, eg using gpio where pin mappings are for parallel port
2. some pins are optional (eg there are multiple enables to support multiple displays, backlight etc)

so to get around that the HAL would have to keep track of whether the output type matches what the driver expects naturally and ensure all the required pins are actually set if it doesn't. and must have a way of letting the driver know a pin isn't mapped when it's not required (could be the same mechanism, but handled differently by the driver? eg return true if found/mapped and false if not?)

and like the above example there could be a different `map_pins` function for each connection type, then all the rest of the code could be commonised (as far as possible, obviously the 8bit and 4bit probably wouldn't)

does this make sense?

so it retains configuration backwards compatibility rather than binary or any other type

haraldg, 2017-02-20:
> does this make sense?

I think I'm still missing some crucial information. In your example the
calls like `hal_request_pin(... "pin_BL", PIN_BL)` are the same as in
mine, yet in your example you imply that this somehow states default
pins, while in mine they do not. Obviously in your example either
`"pin_BL"` or `PIN_BL` have different semantics wrt `hal_request_pin()`
then what I'm thinking about. So, if there is no pin_BL entry in the
configuration section, how does the parport driver know which pin to
use?
 
> so it retains configuration backwards compatibility rather than binary or any other type

If configuration backwards compatibility is the only goal, then I'm not
convinced that machine code is the best way to encode that information.
Maybe a text file is the better way to store the information and if you
don't want to change old configuration files on upgrade, then there could
also be a new file provided, that is read together with the configuration
and has sections like

```
[hd44780-4bit-legacy]
HALDriver=parport
pin_D0=...
...
pin_BL=...
```

If the driver doesn't find a `HAL=` entry in its configuration section,
it could just fall back to `HAL=drivername-connectiontype-legacy`.
However since `Port=` would be in `[hd44780]` both sections would need
to be looked at.

I'm still not convinced that configuration backwards compatibility is
worth the trouble, though.

-- 
If you want to support my work:
see http://friends.ccbib.org/harald/supporting/
or donate via CLAM to xASPBtezLNqj4cUe8MT5nZjthRSEjrRQXN
or via peercoin to P98LRdhit3gZbHDBe7ta5jtXrMJUms4p7w

JamesGKent, 2017-02-21:
sorry, you are right, there would need to be an extra argument in that function to call into the HAL with a default pin number, and also a previous call into the HAL to set the default output type (parallel port/gpio etc)
i can see what you are saying R.E. configuration backwards compatibility, but i think this approach would offer more than just that, every driver I've looked at in some way sets defaults for the wiring/connection type that can be overridden in the config file, providing a safe fallback as it were. this would allow the driver to set up sensible defaults for itself (without needing an overly verbose config file) and add the ability to redirect those pins if required, rather than having to specify every pin used from the get go.

and again regarding the port argument (if parport used) the driver can check it's own config section, and call into the HAL with the value it finds as the default, then if nothing is found in the HAL config to override it, the drivers port is the one used.

so in essence:

1. LCDd starts
1.1. LCDd loads driver
1.2. driver checks config for connection type
1.3. driver calls into HAL with the output type, address (if needed) and all pins needed
1.3.1. if matching key names found in HAL config then values passed in by driver are ignore in favor of HAL values
1.4. HAL configures the outputs
2. driver uses the constants that were passed in during pin setup to set/clear the pins

is this any clearer?

haraldg, 2017-02-21:
Yes, it is clear now, how you plan it to be.

However you still owe an example showing that this is workable.
Ie. what this third argument (and the other call) would be. What semantics
are attached. As you wrote yourself earlier, a HAL should be designed from
the get go, not retrofitted. I think your proposal is dangerously close
to retrofitting, so I'd like something less vague to be confident, that
this ends up as a clear API. Of course in the end we need to write
`hal.h` to declare the API in detail, but ATM a pseudo-code example would
do.

Do you have a reason to prefer writing the defaults in code instead of
a text file? (This text file need not be `LCDd.conf`, if that's your
concern.)

I just realized that having the defaults in the binary driver adds a new
issue when combined with a HAL: In the old days (ie now) a driver using
the parallel port just isn't compiled on systems without such a port,
while a driver using GPIOs isn't compiled on systems without GPIOs.
If a driver is using HAL, obviously it will be compiled on all systems.
If it hard-codes some parport configuration, what will happen on systems,
where there is no parallel port at all?

Also I guess it is quite likely for OS vendors like debian to change
defaults. (Maybe some security policy is going to deny raw io access in
the future, but a HAL driver for a kernel API exists.) If the defaults are
in the driver, the vendor would need to patch all drivers. If the defaults
are in some text file, all it takes is a script that e.g. translates all
parport configurations into ppdev configurations.

-- 
If you want to support my work:
see http://friends.ccbib.org/harald/supporting/
or donate via CLAM to xASPBtezLNqj4cUe8MT5nZjthRSEjrRQXN
or via peercoin to P98LRdhit3gZbHDBe7ta5jtXrMJUms4p7w

JamesGKent, 2017-02-21:
I must admit, I hadn't thought through about compiling on systems that don't have the requisite output device for a drivers default configuration. I'll have a think on this and get back to you.

JamesGKent, 2017-02-21:
I still like the idea of having some kind of default, and if the defaults are to reside in a file (i still think they should be kept out of the main config file) then I can see that it would make it much easier for package maintainers (can only be a good thing)

but I am struggling to picture a neat way that would still support connection types for drivers like hd44780.
the only thing I can picture would mean dropping all connection types except 8bit and 4bit (and those that use USB, serial etc) because through the HAL gpio and raspberrypi would be exactly the same as 4bit, and the 3 8bit types are the same driver with different pin assignments

unless you can see a simply way to have defaults fetched by connection type?
if so then you make a lot of sense, having the defaults defined in a config file instead of hard coded/compiled would allow all drivers on all platforms as long as at least 1 of the output types is available.

but something would need to be worked out regarding updating/upgrading old systems, and I would suggest that such a radical change would warrant a major version bump, to signify a break in backwards compatibility? not sure how that would work for package maintainers though, last thing we would want is an update that breaks things...

haraldg, 2017-02-21:
> but I am struggling to picture a neat way that would still support connection types for drivers like hd44780.
> 
> unless you can see a simply way to have defaults fetched by connection type?

For an idea how to support something similar in the future, see the
fooappliance example above.

For backwards compatibility we could in theory hack the hd44780 driver so
that a connection type of lcm162 is an alias for 8bit and hack the 8bit
connection type, so that it looks for an HAL section of
hd44780-connectiontype-legacy. However I'm not saying that I think this
is worth it.

> but something would need to be worked out regarding updating/upgrading old systems,

IIRC currently `make install` just overwrites any `LCDd.conf` that might
already be installed on the system. I guess any users compiling lcdproc
themselves, are already used to lcdproc upgrade pain ... :(

> and I would suggest that such a radical change would warrant a major version bump, to signify a break in backwards compatibility?

I'm sure whoever will cut that release, will consider this in due time.

> not sure how that would work for package maintainers though, last thing we would want is an update that breaks things...

The debian maintainer has already learned his lesson from lcdproc's poor
handling of configuration updates and is automatically generating the
`LCDd.conf` instead of using the version we ship. I guess he could adapt
his scripts easily.

I never had any interaction with other maintainers, so not sure there
are any around.

-- 
If you want to support my work:
see http://friends.ccbib.org/harald/supporting/
or donate via CLAM to xASPBtezLNqj4cUe8MT5nZjthRSEjrRQXN
or via peercoin to P98LRdhit3gZbHDBe7ta5jtXrMJUms4p7w

JamesGKent, 2017-02-22:
in which case i would say you are right, design the HAL to be capable/flexible and forget about backwards compatibility, put defaults for connection types into a config file so that they can be changed on a per platform basis (depending on presence of output types).

and if that is the case then none of the wiring details would reside in the HAL itself, so would you agree that the HAL is best of being part of the LCDd core? as opposed to an extra .so file?

haraldg, 2017-02-22:
> and if that is the case then none of the wiring details would reside in the HAL itself, so would you agree that the HAL is best of being part of the LCDd core? as opposed to an extra .so file?

I have been thinking about that too and am still uncertain. Of course the
individual HAL drivers should be separte file `hal-ugpio.so`, `hal-parport.so`,
etc. For the core of the HAL which provides a call like `hal_open()` it
seems to make sense to be in LCDd. OTOH many drivers probably won't need
it, so it also makes sense to be separate.

Linking in lcdproc is a big mess at the moment. We already have `libbignum.a`
statically linked into 21 drivers and `libLCD.a` statically linked into 26
drivers. Both should either be part of LCDd or a shared library. Also there
is shared/libLCDstuff.a statically linked into LCDd and all clients. I hope
somebody will work on it, but it is independent of the HAL issue of course.

Maybe it's easiest to just start with a `hal.c` compiled into all drivers
that use it. That way we would keep binary compatibility for the moment
and clean up the whole mess in one go - changing the API and ABI with 0.6
whenever that happens. Ie whenever enough drivers are ported to HAL and
we are confident that the API holds.

I think this is nothing we can get wrong, so let's not get distracted by
it.

-- 
If you want to support my work:
see http://friends.ccbib.org/harald/supporting/
or donate via CLAM to xASPBtezLNqj4cUe8MT5nZjthRSEjrRQXN
or via peercoin to P98LRdhit3gZbHDBe7ta5jtXrMJUms4p7w

markus2330, 2017-03-12:
A very elegant solution is always to auto-detect hardware. Unfortunately, I do not know enough about LCDs if this is feasible. E.g. [gpsd](http://www.aosabook.org/en/gpsd.html) did this very successfully. So maybe you do not need to store the pin mappings if you can somehow detect it?

In any way, having drivers which are identical except of pin mappings is definitely an area for improvement. HAL solutions are ubiquitous, e.g. [LinuxCNC](http://linuxcnc.org/docs/html/hal/intro.html) and [Machinekit](http://www.machinekit.io/docs/index-HAL/) have something like that.

If there is really no way around storing the pin mappings, I agree with @JamesGKent to store such "hardware profiles" externally to the current configuration file. I do not really see a problem to store it as (separate) configuration file, tough. From your discussion it looks like as if these mappings should be customizable for users.

JamesGKent, 2017-03-13:
I think the current consensus is to have a config file with 'default' pin mappings that could be customized per platform (so no parport setups for things like RPi etc) and a config section in the main LCDd.conf file that could be used to overwrite those defaults if desired.

@haraldg out of curiosity what would be your thoughts on trying to make use of another project such as [u8g2](https://github.com/olikraus/u8g2)?
the reason I ask is because this library has been designed from the ground up with a HAL (so might be easier to port) and has a large list of displays and connection types that it supports (more than lcdproc I believe) just a thought.

haraldg, 2017-03-14:
u8g2 seems to be an arduino library, so probably not interessting unless somebody wants to port LCDd to arduino. Also it seems to be a library for graphics displays, while LCDd is designed for text displays (but does support printing text on some graphics displays too).

That being said, LCDd has a number of drivers for external libraries (even X11), adding one more wouldn't be a problem.