lcdproc/lcdproc #50: code in contrib/patches unmaintained

haraldg, 2017-01-20:
There are some patches from around 2006 below `contrib/patches` that for whatever reason didn't get merged. Of course nobody keeps them up to date with the code. We should figure out what to do with them.

JamesGKent, 2017-02-09:
I've tried to have a look at these, but they appear to be so old that most of the code these patches affect have been rewritten so much there's no way easy I can see to merge them, and the features would have to be rewritten to add them.
i do think it would be a nice feature to be able to synchronize the scrolled text elements though. even if that's an optional setting.

haraldg, 2017-02-09:
Yeah, I guess these patches would mostly serve as guidance for writing the same feature again, nowadays - if the features are actually desired, of course.

Maybe the ppdev stuff could be a start for writing an abstraction layer like was suggested in #49.