lcdproc/lcdproc #53: yard2LCD: Warnings building on OpenWRT

haraldg, 2017-02-06:
@YARD2 can you look into this:
```
yard2LCD.c:226:1: warning: return type defaults to 'int' [-Wimplicit-int]
 yard_hwSetBrightness(Driver *drvthis, unsigned char brightVal)
 ^
yard2LCD.c: In function 'yard_init':
yard2LCD.c:308:20: warning: embedded '\0' in format [-Wformat-contains-nul]
  sprintf(Recbuffer,"LCDPROC\0");
                    ^
yard2LCD.c:349:14: warning: pointer targets in assignment differ in signedness [-Wpointer-sign]
  p->framebuf = (unsigned char *) malloc((p->width * p->height)*2);
              ^
yard2LCD.c: At top level:
yard2LCD.c:79:12: warning: 'yard_hwEnterGmode' declared 'static' but never defined [-Wunused-function]
 static int yard_hwEnterGmode(Driver *drvthis);
            ^
yard2LCD.c:187:1: warning: 'yard_hwPrintChar' defined but not used [-Wunused-function]
 yard_hwPrintChar(Driver *drvthis, char c)
 ^
```

I guess nobody ever compiled this driver with `-Wall` ...

dod38fr, 2017-02-07:
Most drivers are compiled with `-Wall` on Debian. See the last [build log for rc2](https://buildd.debian.org/status/fetch.php?pkg=lcdproc&arch=amd64&ver=0.5.8~rc2-1&stamp=1484767480&raw=0).

All the best

YARD2, 2017-02-07:
Yes I never compiled it will -wall
I fixed all isssue in my branch except
  sprintf(Recbuffer,"LCDPROC\0");
The yard daemon need the \0

haraldg, 2017-02-08:
@dod38fr wrote:
> Most drivers are compiled with -Wall on Debian. See the last build log for rc2.
Sure, what I wanted to say was that we should have caught this before merging the driver - as it was only added in the recent release.

I guess reviewing code on github has its limits, nobody bothers to actually compile test patches anymore, when the buildbot runs successfully.

Nice to read from you though: I started to get concerned since you didn't upload 0.5.8 to debian and there where no other signs of activity either.

dod38fr, 2017-02-08:
>Nice to read from you though: I started to get concerned since you didn't upload 0.5.8 to debian and 
> there where no other signs of activity either.

Sorry about that, I got sidetracked by other non-foss issues.

Anyway, lcdproc release came in too close to Debian freeze date: people tend to update lcdproc quite slowly, so there would be very few actual usage between release of 0.5.8 to Debian unstable and stable release. I prefer to release 0.5.8 after Debian stable is out.

All the best

YARD2, 2017-02-08:
I check the \0 on weekend

haraldg, 2017-02-20:
@YARD2 What is the status of this?

YARD2, 2017-02-21:
Sorry was ill, I changed the code again.
Now it should work, but I do not know how to merge it now :-(

commit 156983af (2017-02-21): Yard2lcd: Fix compiler warnings

This fixes #53.
closed 2017-02-21 by haraldg