lcdproc/lcdproc #54: hd44780 Driver only supports single driver displays

JamesGKent, 2017-02-07:
the new 0.5.8 release lists some connection types as being obsolete due to the new addition of the `gpio` connection type, however the `raspberrypi` connection type supports dual controller displays such as needed to 40x4 format displays by including a second enable pin for the second driver IC, this is missing from the `gpio` connection type.

as such when i connect my 40x4 driver and use the `gpio` connection type only the top half of the display is driven.
switching to the `raspberrypi` connection type fixes this and drives the whole display, i would say that it's wrong to mark it as obsolete when its replacement does not do everything that its predecessor can do.

JamesGKent, 2017-02-07:
further to that, if the gpio driver is updated to add the missing functionality, a future release could use the `raspberrypi` connection type as an alias and use the `gpio` connection type behind the scenes.  
this might be possible for some of the other obsolete connection types, such as `piplate` and `pifacecad` but having not used them i can't know for sure.

haraldg, 2017-02-08:
Thanks for pointing this out. Ideally we should notice such issues during testing of the release candidates. Unfortunately it seems nobody who did tests had a dual controler display. We will look into adding this feature to the `gpio` connection type eventually. In case you are considering to look into this yourself and submit a PR, this would a appreciated a lot, however.

JamesGKent, 2017-02-08:
apologies, I wasn't paying much attention to the project and I didn't even see there had been a release candidate until after the final release. I will look at adding the feature to the gpio connection type, but it might take me a while.

ethandicks, 2017-02-09:
I have a 40x4 display in a drawer I can see about hooking up to something.  I have plenty of Raspberry Pi units for testing.  Just need to make up a cable.

JamesGKent, 2017-03-29:
this issue should be fixed by [https://github.com/lcdproc/lcdproc/pull/76](https://github.com/lcdproc/lcdproc/pull/76)

I will try to test sometime in the next week and report back.

commit 034954a5 (2017-04-08): add documentation for 412fba5f73cc346f3dc75d8cc91380659c502152

This closes #54.
closed 2017-04-08 by haraldg