lcdproc/lcdproc #56: timing issues related to nanosleep()

pprindeville, 2017-02-09:
I was trying to port lcdproc 0.5.8 to a new distro and had some issues getting timing.h to do the right thing in the hd44780 driver (which as luck would have it, was the only display hardware I had available to test on).

Distilling the issue:

1. `configure.ac` expects `nanosleep` to be in an ancillary library, not in `libc` (not the case on linux for instance);

this is fixed with:

```
--- a/lcdproc-0.5.8/configure.ac     2017-02-07 16:28:00.894415517 -0700
+++ b/lcdproc-0.5.8/configure.ac  2017-02-07 16:40:08.906990034 -0700
@@ -86,7 +86,11 @@
 AC_CHECK_FUNC(connect,,[AC_CHECK_LIB(socket,connect)])
 AC_CHECK_FUNC(inet_aton,,[AC_CHECK_LIB(resolv,inet_aton)])
 AC_CHECK_LIB(kstat, kstat_open)
-AC_CHECK_LIB(posix4, nanosleep)
+dnl nanosleep should be in libc if not, try in a POSIX compatibility library
+AC_CHECK_FUNCS(nanosleep)
+AC_CHECK_LIB(posix4, nanosleep,
+       AC_DEFINE([HAVE_NANOSLEEP], [1],
+                       [Define if you have the nanosleep function.]))
 AC_CHECK_FUNCS(getloadavg swapctl)
 AC_CHECK_HEADERS(procfs.h sys/procfs.h sys/loadavg.h utmpx.h)
 
```
i.e. checking against the default libraries first before checking against `libposix4`.

2. testing for `HAVE_NANOSLEEP` and nothing else, such as:

```
--- a/lcdproc-0.5.8/server/drivers/timing.h  2015-03-27 10:10:45.000000000 -0600
+++ b/lcdproc-0.5.8/server/drivers/timing.h       2017-02-07 16:44:00.944497415 -0700
@@ -46,7 +46,7 @@
 
 /* Autoselect...  Does this always work well ? */
 #ifdef DELAY_AUTOSELECT
-# if defined HAVE_SCHED_H && defined HAVE_SCHED_SETSCHEDULER
+# if defined HAVE_NANOSLEEP
 #  define DELAY_NANOSLEEP
 # else
 #  define DELAY_GETTIMEOFDAY
@@ -76,7 +76,6 @@
 #else /* assume DELAY_NANOSLEEP */
 # undef DELAY_GETTIMEOFDAY
 # undef DELAY_IOCALLS
-# include <sched.h>
 #endif
 
 /*
@@ -130,17 +129,7 @@
 static inline int
 timing_init()
 {
-#if defined DELAY_NANOSLEEP
-       /* Change to Round-Robin scheduling for nanosleep */
-       {
-               /* Set priority to 1 */
-               struct sched_param param;
-               param.sched_priority=1;
-               if (( sched_setscheduler(0, SCHED_RR, &param)) == -1) {
-                       return -1;
-               }
-       }
-#elif defined DELAY_IOCALLS
+#if defined DELAY_IOCALLS
        if (port_access(0x3BD) == -1) {
                return -1;
        }
```

most OSes today don't require changing out the scheduler for `nanosleep()` to work so we can drop that check.

3. The comment says it all.

```
--- a/lcdproc-0.5.8/server/drivers/hd44780.c 2015-03-27 10:10:45.000000000 -0600
+++ b/lcdproc-0.5.8/server/drivers/hd44780.c      2017-02-07 16:00:21.435216682 -0700
@@ -51,16 +51,6 @@
 */ 

 
-/*
- * Uncomment one of the lines below to select your desired delay generation
- * mechanism. Using DELAY_NANOSLEEP  seems to provide the best performance.
- *
- * Setting this here, overrides the set or selected algorithm in timing.h.
- * FIXME: Is this on purpose?
- */
-//#define DELAY_GETTIMEOFDAY
-#define DELAY_NANOSLEEP
-//#define DELAY_IOCALLS
 
 /* Default parallel port address */
 #define LPTPORT         0x378
```

All that said, I've not managed to bring up the hd44780 display I have (it's actually an ST7063) so perhaps someone else can confirm my changes?

Thanks

pprindeville, 2017-02-09:
@haraldg so this doesn't slip through the cracks...

haraldg, 2017-02-09:
Hi, I did some forensic in the git history, and it appears to me, that at least the code in hd44780.c was accidentally introduced by a merge issue in 2001.

I will review the other changes, split them into logical changes and open a PR so that people can easily test them.

haraldg, 2017-02-10:
@pprindeville after thinking about this a bit more there seems very little wrong with the code you are proposing to change:

* Maybe checking for nanosleep() before using it might be cleaner, but all platforms we care about have it, so just linking in the right library seems ok.
* Trying to get realtime priority also doesn't look wrong to me, though maybe this shouldn't be a strict requirement for using nanosleep() delays.
* As per our private communication it seems like the real problem you are trying to solve here, is that setting the priority fails in ways that `timing.h` can't detect at compile time. So maybe `timinig_init()) should just emit a big warning but return successfully in that case?

Of course having an obscure header file decide on the delay algorithm is a bit strange by itself. Maybe I should try to put this logic into configure...

pprindeville, 2017-02-10:
> @pprindeville after thinking about this a bit more there seems very little wrong with the code you are proposing to change:
> 
>   *  Maybe checking for nanosleep() before using it might be cleaner, but all platforms we care about have it, so just linking in the right library seems ok.

I could see porting lcdproc to Green Hills Integrity or VxWorks in the future... where the libraries are much more compartmentalized.

>   *  Trying to get realtime priority also doesn't look wrong to me, though maybe this shouldn't be a strict requirement for using nanosleep() delays.

Yeah, failing to do this even if `HAVE_NANOSLEEP` is set shouldn't be a fatal error.  Maybe just `report()` a warning or notice.

>   *  As per our private communication it seems like the real problem you are trying to solve here, is that setting the priority fails in ways that timing.h can't detect at compile time. So maybe `timinig_init()) should just emit a big warning but return successfully in that case?

Sure, works for me.

Also let's not have `hd44780.c` try to override what `timing.h` decides is the best strategy to use.  That should be a modular and internal decision to the timing code...

> Of course having an obscure header file decide on the delay algorithm is a bit strange by itself. Maybe I should try to put this logic into configure...

That would be nice but not vital enough to be blocking.

commit 300ccce7 (2017-02-11): timing.h: Always use nanosleep() by default

The old code prefered busy loops if we don't have the syscalls to obtain
realtime priority available. This doesn't look like a sensible tradeoff
nowadays.

This closes #56.
closed 2017-02-11 by haraldg