lcdproc/lcdproc #57: Multiple errors in hd44780-bwct-usb when compiling with ./configure --enable-driver=hd44780

dagelf, 2017-02-09:
Any idea what I'm missing? This error isn't present in previous version.

hd44780-bwct-usb.c: In function 'hd_init_bwct_usb':
hd44780-bwct-usb.c:68:3: warning: implicit declaration of function 'usb_init' [-Wimplicit-function-declaration]
   usb_init();
   ^
hd44780-bwct-usb.c:69:3: warning: implicit declaration of function 'usb_find_busses' [-Wimplicit-function-declaration]
   usb_find_busses();
   ^
hd44780-bwct-usb.c:70:3: warning: implicit declaration of function 'usb_find_devices' [-Wimplicit-function-declaration]
   usb_find_devices();
   ^
hd44780-bwct-usb.c:72:4: error: 'PrivateData' has no member named 'usbHandle'
   p->usbHandle = NULL;
    ^
hd44780-bwct-usb.c:73:4: error: 'PrivateData' has no member named 'usbIndex'
   p->usbIndex = 0;
    ^
hd44780-bwct-usb.c:74:3: warning: implicit declaration of function 'usb_get_busses' [-Wimplicit-function-declaration]
   for (bus = usb_get_busses(); bus != NULL; bus = bus->next) {
   ^
hd44780-bwct-usb.c:74:12: warning: assignment makes pointer from integer without a cast
   for (bus = usb_get_busses(); bus != NULL; bus = bus->next) {
            ^
hd44780-bwct-usb.c:74:54: error: dereferencing pointer to incomplete type
   for (bus = usb_get_busses(); bus != NULL; bus = bus->next) {
                                                      ^
hd44780-bwct-usb.c:77:19: error: dereferencing pointer to incomplete type
     for (dev = bus->devices; dev != NULL; dev = dev->next) {
                   ^
hd44780-bwct-usb.c:77:52: error: dereferencing pointer to incomplete type
     for (dev = bus->devices; dev != NULL; dev = dev->next) {
                                                    ^
hd44780-bwct-usb.c:81:14: error: dereferencing pointer to incomplete type
       if (dev->descriptor.idVendor != BWCT_USB_VENDORID) {
              ^
hd44780-bwct-usb.c:85:26: error: dereferencing pointer to incomplete type
       for (c = 0; c < dev->descriptor.bNumConfigurations; c++) {
                          ^
hd44780-bwct-usb.c:87:15: error: 'PrivateData' has no member named 'usbIndex'
         for (p->usbIndex = 0; p->usbIndex < dev->config[c].bNumInterfaces; p->usbIndex++) {
               ^
hd44780-bwct-usb.c:87:32: error: 'PrivateData' has no member named 'usbIndex'
         for (p->usbIndex = 0; p->usbIndex < dev->config[c].bNumInterfaces; p->usbIndex++) {
                                ^
hd44780-bwct-usb.c:87:48: error: dereferencing pointer to incomplete type
         for (p->usbIndex = 0; p->usbIndex < dev->config[c].bNumInterfaces; p->usbIndex++) {
                                                ^
hd44780-bwct-usb.c:87:77: error: 'PrivateData' has no member named 'usbIndex'
         for (p->usbIndex = 0; p->usbIndex < dev->config[c].bNumInterfaces; p->usbIndex++) {
                                                                             ^
hd44780-bwct-usb.c:91:30: error: dereferencing pointer to incomplete type
           for (a = 0; a < dev->config[c].interface[p->usbIndex].num_altsetting; a++) {
                              ^
hd44780-bwct-usb.c:91:53: error: 'PrivateData' has no member named 'usbIndex'
           for (a = 0; a < dev->config[c].interface[p->usbIndex].num_altsetting; a++) {
                                                     ^
hd44780-bwct-usb.c:93:22: error: dereferencing pointer to incomplete type
             if (((dev->config[c].interface[p->usbIndex].altsetting[a].bInterfaceClass == 0xFF) &&
                      ^
hd44780-bwct-usb.c:93:45: error: 'PrivateData' has no member named 'usbIndex'
             if (((dev->config[c].interface[p->usbIndex].altsetting[a].bInterfaceClass == 0xFF) &&
                                             ^
hd44780-bwct-usb.c:94:22: error: dereferencing pointer to incomplete type
                  (dev->config[c].interface[p->usbIndex].altsetting[a].bInterfaceSubClass == 0x01)) ||
                      ^
hd44780-bwct-usb.c:94:45: error: 'PrivateData' has no member named 'usbIndex'
                  (dev->config[c].interface[p->usbIndex].altsetting[a].bInterfaceSubClass == 0x01)) ||
                                             ^
hd44780-bwct-usb.c:95:21: error: dereferencing pointer to incomplete type
[... 76 lines cut ...]
                ^
hd44780-bwct-usb.c:203:6: error: 'PrivateData' has no member named 'usbHandle'
     p->usbHandle = NULL;
      ^
hd44780-bwct-usb.c: In function 'bwct_usb_HD44780_set_contrast':
hd44780-bwct-usb.c:216:24: error: 'PrivateData' has no member named 'usbHandle'
   if (usb_control_msg(p->usbHandle, USB_TYPE_VENDOR, BWCT_LCD_SET_CONTRAST,
                        ^
hd44780-bwct-usb.c:216:37: error: 'USB_TYPE_VENDOR' undeclared (first use in this function)
   if (usb_control_msg(p->usbHandle, USB_TYPE_VENDOR, BWCT_LCD_SET_CONTRAST,
                                     ^
hd44780-bwct-usb.c:217:31: error: 'PrivateData' has no member named 'usbIndex'
                       value, p->usbIndex, NULL, 0, 1000) < 0)
                               ^
Makefile:1334: recipe for target 'hd44780-hd44780-bwct-usb.o' failed
make[3]: *** [hd44780-hd44780-bwct-usb.o] Error 1
make[3]: Leaving directory '/usr/src/lcdproc/server/drivers'
Makefile:472: recipe for target 'all-recursive' failed
make[2]: *** [all-recursive] Error 1
make[2]: Leaving directory '/usr/src/lcdproc/server'
Makefile:451: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/usr/src/lcdproc'
Makefile:370: recipe for target 'all' failed
make: *** [all] Error 2

haraldg, 2017-02-09:
Looks like files from libusb0.1 are missing.
However configure should detect this and refuse to build the connection type. I'll have a look if anything is obviously wrong with the logic there.

haraldg, 2017-02-09:
@dagelf I think the error might be triggered by pkg-config not being present. Can you see if you have in `config.log` a line like:
```
pkg-config not (fully) installed; drivers requiring libusb may not be built
```

In this case HAVE_LIBUSB would be undefined, but I think $enable_libusb would still be yes. Thus causing the error.

TIA,
Harald

dagelf, 2017-02-10:
Compiling on Debian jessie on CHIP. Pkg-config is present.

Works fine if I dont specify drivers for configure, I presume it just
builds all then?

Also works fine if I remove the drivers with "usb" in them from the
Makefile, and also works fine on previous version.  Maybe some dependency
is not properly defined in new version?

haraldg, 2017-02-10:
If you don't specify drivers, it builds a predefined set but not all.

The errors you pasted only make sense if $enable_libusb = yes but HAVE_LIBUSB is undefined. I don't see (from reading the soure) how this could happen by other reasons then pkg-config issues. If you really don't have anything like the line I quoted above in the config.log, then please attach the config.log itself. I don't have divination skills.

fmertz, 2017-05-09:
So, in this scenario where pkg-config is not installed, what should the build do? Here are a few options I can think of:

1.  Should the build stop altogether? Should pkg-config be a hard dependency? Should we update autogen.sh to detect pkg-config and warn or stop if it is missing?

2. Should the build continue, but leave the dependent code behind (same as configure --without-lib...)

3. Should we make additions to configure.ac to test for the presence and usability of each header file (like usb.h) and add the best guess library name? In essence, should we use pkg-config if available, but have fallback code if it is not.

I was compiling the package on a vanilla FreeBSD (after I added autoconf, etc, but not pkg-config) with the HD44780 driver, and I ran into similar errors where it is trying to compile the USB and FTDI interface code. 

I am leaning towards a small change in the build system to accomplish option 2 in the short term, and maybe try and get to option 3 later...

haraldg, 2017-05-12:
@fmertz If you can reproduce the problem and test a fix then I'm happy with any of the options you propose. I agree that (2) seems the best approach.

(I see little point in spending effort on (3). If anybody prefers (1) then I think the check for pkg-config should bi in configure, not in autogen.sh.)

clach04, 2021-01-27:
My solution to get past compilation failures:

    ./configure  --disable-libftdi --disable-libusb       --disable-libusb-1-0  --enable-drivers=curses,hd44780,text ; make

I'm seeing the same problem with usb as reported by @dagelf , I'm also seeing issues with ftdi as well (using latest commit from 2020 https://github.com/lcdproc/lcdproc/commit/9ec9ba4e5dda653288bc55d2898723aa2c2ad9c1)

I do not have pkg-config installed and the above was fine for my quick experiment so I've not dug into this further.