lcdproc/lcdproc #71: no IPv6 support

haraldg, 2017-03-19:
This is probably low priority for everybody, but I guess we shouldn't forget to clean this up as well, when we rework the socket stuff.

fmertz, 2017-04-24:
I have not done research on the specifics, but the lack of support for IPv6 comes up on the pfSense forum once in a while. I suspect it has to do with whatever is in the hosts file for "localhost" and/or the pfSense PHP LCDproc client.

This older article might be a good start: [https://www.akkadia.org/drepper/userapi-ipv6.html](url)

haraldg, 2017-04-24:
The problem is in the C code of both the official clients and the server. I have a draft of liblcdproc, that also supports IPv6. This will fix the problem for the clients, once they are ported to the library. The entire network code of the server is quite messy. I'm not sure yet, but I hope we can use some parts of liblcdproc for it as well.

If anybody cares about this enough to actually work on it: Please speak up, so that we can coordinate efforts.