lcdproc/lcdproc #72: Configuration reloading doesn't work

haraldg, 2017-03-24:
* it is not documented
* it may read the wrong configuration file (ignores `-c`)
* it doesn't notify the clients that screen size might have changed
* it didn't actually work when I tried it

I'm mentioning this mostly to warn @markus2330 and @tom-wa who consider using that feature.

markus2330, 2017-03-24:
Thanks for reporting!

The code in question has to be rewritten anyway (`server/main.c:724 do_reload`).

>   it doesn't notify the clients that screen size might have changed

I assume it is not safe to call `clients_init` again and the complete implementation is missing?

haraldg, 2017-03-24:
The code was added in 2002 with a comment suggesting that it's only the start of an implementation. The comment was removed later, but the implementation never completed. I didn't look into the details of how much is missing, but as nobody has bothered for 15 years there is little reason to work on it.