lcdproc/lcdproc #74: suggested changes to the protocol (client ABI)

haraldg, 2017-03-26:
This ticket is to collect all the ideas for changes to the network protocol (backwards compatible or not). Also maybe to collect current and future real world use cases.

# Character encoding (character set?)
Currently LCDd just shuffles bytes between clients and drivers and it is the responsibility of the user to set up everything, so that things fit together. This means we are mostly using 7bit ascii, but exceptions exist. There should be a way to at least negotiate what's supported. Even better if LCDd could be told to do some conversions.

# Frames
are specified in the current version of the protocol, but most parts never got implemented (for more then a decade). Maybe it is time to acknoledge that we target small displays where these are not very useful and nobody really needs them. We should figure out which parts are actually implemented (and work!) and drop the rest from the ABI and also remove any code that never actually worked as intended.

# Way to edit times (or strings)
This could be a new menu widget or a general input widget (lcdproc currently doesn't have general input widgets though). Either it would be special for editing time strings like we already have for IPv4 addresses or it would take a specification of allowed characters in the string to be edited. The latter would allow to not only edit times, but also cron lines, etc.

Example for simple specification syntax: `[012][0-9]:[0-5][0-9]`  
This example has the problem that "29:00" is not a valid time. The purpose of the specification would be more to make the editing process easier with only Up/Down keys easier, than to ensure that only legal values can be entered. A more complex specification language would be necessary for the latter. Not sure if it is worth the effort, though.

haraldg, 2017-04-19:
An other issue with the current protocol, that makes me slightly uncomfortable, is the way how key reservation works:
* `client_add_key` takes an arbitrary number of keys (strings) as argument, returns a "huh?" message for each key that it can't grab and finally an unconditional "success". "huh?" messages aren't meant for machine parsing, so the client doesn't actually know which keys it now owns.
* If the client requested just one key, we return a "huh?" message and "success", which is unambiguous but bad taste IMO.
* keys are requested on the level of clients not screens. Maybe instead of the -shared/-exclusively distinction there should be a screen/client level distinction: Shared keys would be requested per screen and only returned when this very screen is visible.

emteejay, 2018-01-14:
Graphic displays also need consideration. It should be possible for positions and sizes to be specified in pixels rather than characters and lines, also scrolling should be in pixel steps.

haraldg, 2018-01-14:
Martin Jones writes:
> Graphic displays also need consideration. It should be possible for positions and sizes to be specified in pixels rather than characters and lines, also scrolling should be in pixel steps.

Hm, this suggestion has come up before. So far this never went anywhere.
A few reasons:

* It's not clear why somebody, who wants to display graphics, would use
  lcdproc instead of some more graphics oriented display system.
* It's unlikely that there would be much common code between text based
  lcdproc and graphics based lcdproc. So maybe it is better to make this
  two projects anyway.
* AFAIK nobody ever followed up with an actual proposal.

I notice, that you aren't actually proposing to display graphics, but rather
to place text on the screen based on pixels rather then character cells. A
few comments about that:

There is some room for improvement of scrolling. Not only smooth scrolling on
graphics displays, but also proper scrolling of multibyte characters.
However for this to become possible, some support from the drivers will be
necessary. Ie some part of the rendering would need to move from the
widget code to the driver code. For this a change to the driver API will
be necessary. However I don't think the client language should be affected
by that. (The details are still very unclear to me.)

The other thing, that would benefit from giving up character cell based
resolution, is variable length fonts on displays that support them. Allowing
pixel based widget positions might be one way to go about that. Other
options are using character positions, but allowing float values, and
using a percentage of the display width and hight. However at the moment
the LCDd widget rendering code only (correctly) implements a fraction of
what could be specified with the widget language, so I'm uncertain
how much fine control over the widget positions actually gets you.
(I assume this fine control would mostly show off with fancy stuff, that
doesn't actually work ATM.)

haraldg, 2018-02-21:
Next idea for an extension on the client protocol: Some way to synchronize the server->client and client->server streams on demand.

Rationale: Right now all commands give some feedback to the user. Usually "success" or "huh?", but no client bothers to keep them in sync. Instead they just feed a bunch of commands to the server and deal with the result once it arrives. Anything else probably would also be quite slow (at least over physical networks), so that's reasonable.

However maybe sometimes you really want to know if you command was successful or not. Now that's practically impossible, I think. But if there was some command to send a token to the server and have it answer with the same token, it would be easy to match the steams when necessary.

The command could be called "echo" or "sync" and would be quite easy to implement.

haraldg, 2018-02-22:
One thing that bug's me about the current protocol is, that `menu_set_item` and `menu_del_item` require a parent id. That doesn't make much sense as menu items need unique ids anyway.

I suggest we formally declare that item ids need to be unique (at least per client) and allow the parent id in these commands to be garbage. (Ie using "" would be fine, even if the menu is deep down in the hieararchy.)

haraldg, 2019-04-25:
Since it looks like we are going to bump the protocol version with the next release, let's review the discussion here and add some more ideas:

One thing I always disliked about the protocol is, how very verbose it is. In particular the many  underscores I have to type. I propose the addition of two new commands:

wi (widget_init) which basically is widget_add, but also takes all the arguments of widget_set, to allow setting up static stuff like labels in one go.

wu (widget_update) which is a stripped down version of widget_set, taking exactly one widget specific parameter - for most widgets the last parameter of widget_set.

emteejay, 2019-07-07:
Some time ago someone asked if it was possible to set the brightness. This should be at client level and as a screen attribute in a similar fashion to backlight.

There is also a case for the output command to be a screen attribute as well as the current client level setting.

When tidying the code there are sections of code implementing 'string', 'title' & 'scroller' which should be shared.

bdshields, 2019-10-21:
> wi (widget_init) which basically is widget_add, but also takes all the arguments of widget_set, to allow setting up static stuff like labels in one go.
>
> wu (widget_update) which is a stripped down [...]

These are good ideas, but I suggest retaining widget_set because sometimes you want to move a widget without a delete/add cycle.

I feel frames should be retained because they allow the display of a table of information that is larger than the screen, and then scroll that table either horizontally, or vertically. I noticed horizontal scrolling is not implemented, but I'm happy to take on that work if you wish.

bdshields, 2019-10-23:
I'd like to see frames (or whatever form they take) further developed, but happy to help in other areas also.
I see frames have two benefits: screens larger than the display; but also allowing other widgets to have a "private" origin or coordinates reference (eg 1,1).

haraldg, 2019-10-28:
Sorry for the delay in communication on my side. Feel free to jump in, if something catches your fancy.

I'd give up on the "private origin" idea. That is usefull for full widget systems like GTK. But in my experience any LCDproc client neds to be quite clever placing widgets on the tiny screen, to make the most out of it. This is especially true when the screen needs scrolling. Therefore this kind of elegant abstraction isn't useful. Clients tend to micromanage their widgets.

In particular I believe we shouldn't allow nesting of frames or scrolling widgets. Maybe scrollers and frames should even be unified somehow.

bdshields, 2019-10-28:
> Sorry for the delay in communication
We all have day jobs :-)

To keep this thread clean, I'm going to create a new issue to track the discussion of the work I'm doing. #141 The main goal will be to allow screens larger than the display.

ethandicks, 2023-11-24:
I work with various sized panels, mostly 16x up to 40x4 but I also have a set of them that does 24x8.  Given the real estate of the larger panels, I also write clients to use bignums, not just for a simple clock display.  One of the gaps in the present protocol is that bignums have to start on the top line of any display.  For the 24x8, that means that you can't easily craft a display with normal-sized text on the top 4 lines and some bignums on the bottom 4 lines.

The proposal is to extend the bignum request to allow for Y pos as well as X pos, and to also consider permitting requesting _specific_ bignum rendering with less than the maximum allowable redefinable characters (currently, each driver knows how many redefinable chars there are and selects the "best" rendering method - I would like to make that an option to override how many redefinable characters there are to any number at or below what the hardware supports)