lcdproc/lcdproc #75: rendering inefficient

haraldg, 2017-03-26:
Currently all widgets on the active screen are redrawn 8 times per second, even if the screen is entirely static and widget data hasn't changed. LCDd doesn't contribute to the system load significantly, so this is low priority to clean up. (Actually on my system it seems to consume more memory the CPU.)

Still I was surprised to find this, given that lot's of driver try hard to implement delta-updates to the frame buffer.

shenghaoyang, 2019-06-08:
Maybe the core should manage a notion of a frame buffer and implement basic delta-update functionality  to avoid code duplication with the drivers providing cost functions for some of the common operations (moving the cursor, updating a single character, updating the whole screen). Probably a simple algorithm that looks for differences in the framebuffer, coalesces adjacent characters that have been modified, and chooses the most optimal method of updating the display by comparing the cost of updating the modified character blocks via cursor movements versus the cost of simply updating the whole screen after resetting the cursor to the origin.

Still, it's important that this delta-update functionality also be configurable, as my POS displays seem to slow down processing and drop commands when sent too many cursor-move commands to implement partial update functionality (that was the reason I made the POS driver update line-by-line, as it seemed to be the most robust). 

If the driver has its own delta-update functionality it can simply set the cost function pointers to NULL and handle the functionality internally.

haraldg, 2019-06-08:
No, the core can't manage a frame buffer, because there is no
univeral notion of a frame buffer among the drivers. For example some
drivers don't draw hbars with characters at all.

However what the core could do is: not render the same constant string
at the same location 8 times per second. Ie only render widgets when
they are updated or when they are scrolling.