lcdproc/lcdproc #79: Driver Code Consolidation and a "Universal" Driver

fmertz, 2017-04-19:
In the spirit of moving this project forward and keeping changes incremental, how about combining all (or most) existing drivers into just one "universal" driver (yes, one more driver, for now).

Let's get this out of the way first: [https://xkcd.com/927/](url)

This new driver would implement a matrix, by de-coupling the "language" from the "hardware interface", meaning this driver could be told what command set to use on the one hand (the few drivers I have seen seem to have similar command sets), and then separately what particular hardware interface to issue those commands over. This new driver could have all the interface code compiled-in or use some new ABI to dynamically load one interface at a time to minimize memory footprint on embedded platforms. 

Remember, there is no computer problem that can't be simplified by adding a level of indirection!

If we were to implement something like this, I believe the majority of the needed code would be already there in the existing implementation of each driver. There would be a need for a (massive) refactoring. The HD driver(s) seem like a great place to start.

Also, it would not be disruptive as it would not require changes to LCDd, or ABI (unless we wanted to). It could also be a phased implementation, with more and more LCDs supported over time. This would allow us to retire the corresponding discrete drivers over time, too.

The main idea is that it could fill in the matrix of combinations "for free" and avoid silly situations where we have a driver with the right commands here, and another driver for the hardware interface there, but no implementation combining the two.

Also, we could add "high level" configurations entries that would describe popular combinations of LCDs and interface for pre-made hardware, like commercial firewalls, e.g. an entry for "Watchguard/Lanner" would know to engage the SDEC "language" and the parallel port "interface", making it EASY for end users to get their hardware going, and be the next best thing to an automatic detection. The idea here is to make it easy for individual distributions to create glue code to integrate LCDproc. An example is pfSense, which needs PHP code to integrate packages into the GUI of the distribution. A complicated set of drivers and parameters makes it hard for users to get things to work. If there were fewer options, and pre-configured known-good combinations, this glue-code would be easier to write and maintain, and eventually serve the users better.

Thanks for any thought.

haraldg, 2017-04-19:
I think one universal driver probably is too much -- modularity is a good thing after all, but our current amount of code duplication hints that modularity is done in a poor way.

Your proposal of a language/hardware interface matrix is pretty close to what we have been discussing in #49. I think we have done there about 50% of what's necessary to propose an API suitable for implementing such a decoupling. Maybe you can start from there?

Personally I believe addressing this issue is important for the future development of lcdproc, but since I don't use any HW for which that change is relevant, I stay a bit in the background. However I might help with the hardware independent parts, if somebody actually starts working on this seriously.

fmertz, 2017-04-19:
As I understand:

Today, drivers are called from the top (using a known API) and handle BOTH the command set (the "language") of the LCD as well as its hardware interface. This coupling is done on a per-driver basis, as each author saw fit.

With #49, the hardware interface is proposed to be sort of clawed back from the driver space, and put back to the universal server space. In other words, drivers would now be understood to be responsible for the specific commands. The hardware part would be handled through a (new, universal?) API. Individual drivers would then have to be changed to accommodate the new scheme?

With #79, the proposal is to let the server be, as is, but to write a new super driver that, based on configurations, would load up the right command set (through a new, universal API), and pair it up with the right hardware-specific code (through a new, universal API, quite possibly the same as #49). This is not disruptive, and can be deployed when it is ready. It can be enhanced over time to support more and more LCDs. It is just another driver.

I suppose, by symmetry, one could also do another proposal where the "language" is now part of the server, based on configurations. So the server would manage a configuration file where each individual LCD would have the commands for Clear Screen, Place Cursor, Read Key, etc. From that perspective, drivers would only be in charge of the hardware interface, as a "true" driver is supposed to be.

Fun to think about this.

dglaude, 2017-04-19:
The rule for making new driver always was to take a driver similar to the
new hardware you wanted to drive, duplicate, rename and modify.
Based on that, some driver are very similar and other very different.
Some driver started to implement smart things such as delta update where
other lacked behind.
And when a bug is fixed somewhere, other driver do not directly benefit
from it.
And the more driver we got, the more messy it became.

As far as I remember, the language parsing is done on the server side, the
driver API is function call with parameter.

In the driver code there are two part, one is the logic (like buffering
what the screen should look like) and communicating with the physical
display.
That communication can be of various type such as Serial, Serial over USB,
I2C, GPIO, ... and the same "protocol" can be implemented over various
communication.

One of the attempt (maybe in place for some driver such as HD44780) was to
abstract that communication so that the same logic can be used in various
usecase.
I believe this is what #49 is trying to generalise.

I believe that super driver has already been attempted in the past for the
multi-screen aggregation.
That is, if you have two LCD, put them one above another (or side by side)
you want to address them as a single big screen.
It is another kind of super driver, that was calling real driver.

New driver can co-exist with old driver, so as long as the API stay the
same, this is a way to move the new developpement without breaking.
The major issue is for each supported hardware, to find an owner that can
test the code and change.
This is true for every release, but even more true if the driver is
modified.

Regards.

haraldg, 2017-04-19:
> With #49, the hardware interface is proposed to be sort of clawed back from the driver space, and put back to the universal server space.

I think there is no reason why the hardware interface would be part of the server. Maybe its convenient for early prototyping, but in the end I don't think we want the server binary to depend on libusb, libftdi, libugpio, etc.

> In other words, drivers would now be understood to be responsible for the specific commands. The hardware part would be handled through a (new, universal?) API.

Yes, a new universal API (which is why it is somewhat non-trvial to come up with), but implemented in external modules as the drivers are.

> Individual drivers would then have to be changed to accommodate the new scheme?

Can be changed, but don't have to. I don't see a reason why we would drop the old scheme. After all there are lots of drivers (curses, linux_input, etc) that don't need a external HW interface.

> With #79, the proposal is to let the server be, as is, but to write a new super driver that, based on configurations, would load up the right command set (through a new, universal API), and pair it [...]

If solving #49 is a precondition to implementing your plan, then I wonder: What do we gain by this super driver? 

> I suppose, by symmetry, one could also do another proposal where the "language" is now part of the server, based on configurations. So the server would manage a configuration file where each [...]

Well, if after the decoupling is implemented we still have too much code duplication left, then we can move the common parts to the server or some header file or library. Actually we kind of do this already with libLCD.a ...

fmertz, 2017-04-19:
To take a step back, here is where I am coming from:

I was hoping we, in the lcdproc project, would be able to create basic (software) building blocks, the hard parts, written in C, and package them as modules/libraries. These modules would comply to this new hardware layer API we are talking about. These modules would be written by experienced programmers, i.e. us, as it has been done in the past. These would be true, real drivers, in the traditional sense of handling the hardware. Examples would be Serial, Parallel, GPIO, I2C, etc. These would be a transport layer, in a sense. Each one would stand alone.

This super driver, sitting between LCDd and these hardware drivers, would have the responsibility of handling the "language" part of the LCD, meaning it would take care of properly identifying the applicable codes for the usual functions required to update an LCD (clear screen, position cursor, etc.). I was hoping this could be done in a generic manner, based on some configuration file. The overall idea would be that this file could be "indexed" by LCD type (SDEC, mtb, etc.), and have entries for clear screen, position cursor, read key, etc. In addition, this super driver would know how to load (dynamic load) and invoke the driver layer, again based on a configuration. This would be the application layer, in a sense, and could invoke a transport layer.

The net payout would be great, I think, in the sense that it could potentially limit the need for new driver code going forward. As long as a transport would be available, new drivers could be created by power users (instead of C programmers) by editing this config files. In theory, no new code would be required for some combination of "language" and interface someone could have out there, but that none of us have come across, yet. It would be like filling that matrix, so to speak.

I hope this clarifies my proposal a bit. Thanks for talking this through.

JamesGKent, 2017-04-20:
@fmertz as much as i can see what you are getting at, I think you are trying to solve a non existent problem, and instead of increasing simplicity, i think this would make things more complicated, making a HAL as in https://github.com/lcdproc/lcdproc/issues/49 would make drivers simpler to interface with the hardware, as the low level bits would be taken care of in a simple place, leaving only the protocol/command set to take care of in the driver. Obviously this would make some bits of writing a driver much simpler.

however trying to abstract these commands into some kind of configuration file is probably going to lead to a lot of duplication within said config file, along with making it harder to read the driver source to work out what commands are being sent and when. therefore i don't think adding some kind of "universal driver" would solve anything.

we all agree that the source for lcdproc is less than perfect, the linking and duplication of code in many places leads to bigger modules that could be possible, along with some of the other issues @haraldg has found. but adding yet another driver which tries to do the job of every other driver would just add to the mess instead of improving it.

I think the best approach will be to create a HAL as a module as in https://github.com/lcdproc/lcdproc/issues/49 which other drivers can then depend on, meaning that LCDd avoids these dependencies directly, and one by one drivers can be ported as required/tested to use the HAL, thereby reducing the duplication of hardware interfacing code in every driver.

haraldg, 2017-04-20:
@fmertz I think we are all in the boat about your "transport layer". While #49 is only about pin control, there is already shared code for i2c present. For other HW interfaces either there is a proper external library or we will add our own abstraction code when such devices become popular.

About your proposal to describe LCD controllers via a configuration language: I doubt this is possible without the "configuration language" being turing complete itself. I'm afraid in the end your configuration language would look a lot like python.

However, if you think you see a way how to make your proposal happen, I don't want to discourage you: All progress depends on insane people trying what's supposed to be impossible. If it works, it would have impact far beyond lcdproc - think of machine readable datasheets! That would change the world! :)

So yes, I agree the net payout would be great. But as a mere mortal I don't think I can make it happen. BTW I encourage all power users to learn enough C to help writing code for lcdproc. Espeically writing drivers is already almost trivial, which is one of the reasons why we have so many similar drivers.

fmertz, 2017-04-20:
> About your proposal to describe LCD controllers via a configuration language: I doubt this is possible without the "configuration language" being turing complete itself. I'm afraid in the end your configuration language would look a lot like python.

Well, I certainly thought of (and then dismissed) something like that. Either include an existing scripting engine (like Javascript, or even GNU Guile), but it would clearly be overkill, and that kind of bloat feels wrong for a driver. Alternatively, it would be possible in theory to built some form of LCD command language from scratch, and use bison and flex to generate plain C out of it. Although involving a ton of fun and learning, this is hardly justified, that we agree on.

What I had in mind was something way simpler. Something based on simple text config files, and sort of relational (in the sense of a relational database/parent-child type of stuff).

- We could have base tables of command sets (clear screen, position cursor, etc), however many we need, which is probably not that many.

- Then we could have higher level configurations for each type of LCD and their particulars (hasBacklight, hasKeypad, screenSize, line1Address, line2Address, etc.), of course what driver (serial, parallel, etc,), and the parameters for the driver (speed, port, etc). These LCD level configurations would of course refer to an existing command set defined above.

- Then we could possibly have even higher configurations for appliances with built-in LCDs (Lanner, Watchguard, etc) that would immediately help users with known-good configurations.

At this particular time, I am looking at writing a conventional driver for the Portwell EZIO devices. They are interfaced on the serial port. There are at least 2 existing LCDproc drivers that have the basics for this EZIO device (the mtc and lb216), but there are differences, too, like initialization sequence, no backlight and a keypad. Taking a look back at my old SDEC driver, even if it is on the parallel port, the commands themselves are really pretty similar. To me, it begs for some form of factorization/consolidation.

I suppose at some point, someone like me would have to propose a skeleton of an implementation for this in some detail for further evaluation...

haraldg, 2017-04-20:
About your grand plan: As I wrote, I don't want to discourage you, if you think you have a practical idea. The most ambitious hardware description system I have worked with so far is devicetree - see https://www.devicetree.org/ - it does a good job in reducing code in OS kernels. Still it is mundane compared to what you pursue.

About the device you want to work on: backlight and keypad should be no problem. We have already many drivers that read the information about such features from the existing configuration. About initialization sequence I'm less sure - we probably don't want this in `LCDd.conf`. OTOH I see nothing wrong with having a `Variant=foo` setting to decide which initialization sequence to use. If both drivers are similar to the device you want to work on, then maybe you can just merge them together and add your device as third variant?

However I do agree that all drivers including `termios.h` have some shared code, that should get factored out. Again I believe this could be done by writing a simple `serial.h` file that provides a few (inline) functions. I guess C++ programmers would solve the many-variants problem by inheritance between classes. Maybe we could even simulate that in C by clever design of our data structures ...

I never understood why we have the sdec driver besides the hd44780-8bit driver?

fmertz, 2017-04-20:
> I never understood why we have the sdec driver besides the hd44780-8bit driver?

Quite frankly, that HD44780 driver is worth a harder read on my part. It seems to already contain a lot of what we are talking about...