lcdproc/lcdproc #84: "unable to initialize GPIO pins" on Raspberry Pi 2 B+

kddlb, 2017-05-09:
I built from the latest source (d6119a9) with `./configure --enable-drivers=hd44780`, and I get this error:

```
pi@frambuesa:~/lcdproc $ sudo /etc/init.d/LCDd start
Starting LCDproc display server daemon: hd_init_gpio: unable to initialize GPIO pins
Driver [hd44780] init failed, return code -1
Could not load driver hd44780
There is no output driver
Critical error while initializing, abort.

```

kddlb, 2017-05-09:
I changed the connection type from `gpio` to `raspberrypi`, and it "works" (I have no way to test it until I get an actual display 😅.) Will reopen when I'm able to test this properly...

closed 2017-05-09 by kddlb

haraldg, 2017-05-09:
Looks to me like you just forgot to setup you pin mapping in the config file.

(The raspberrypi connection type has defaults, but the gpio connection type doesn't.)

JamesGKent, 2017-05-09:
to use the gpio connection type you need to download and build/install the ugpio library as a dependency [https://github.com/mhei/libugpio](https://github.com/mhei/libugpio)

kddlb, 2017-05-09:
@JamesGKent I took care of that.

@haraldp the documentation didn't  mention that, as far as I remember. I'll let you know in a few hours, when I get my LCD.

kddlb, 2017-05-09:
I couldn't wait, so I propped up a tester thing using some LED's I had around, and it works using the `raspberrypi` driver, and using the default pinout assignments: https://www.youtube.com/watch?v=ipAzmZZczhI&feature=youtu.be