lcdproc/lcdproc #86: hd44780 not working on pi zero W

ElFishi, 2017-06-28:
I can't get LCDproc to work with a HD44780 display on a Pi zero W. 
Hardware is OK as I can drive the display by a python script. 

LCDd reports 
`Server running in foreground`
`Listening for queries on 127.0.0.1:13666`
`HD44780: using ConnectionType: raspberrypi`
`hd44780: Using hd44780_default charmap`
`check_board_rev: Raspberry Pi 2 or higher detected`

I suspect the hardware detection and corresponding determination of GPIO base adress may be a problem here. Pi zero runs on the same SoC as a Pi 1, but is apprently treated as a "Pi2 or higher"

terba, 2017-06-28:
This commit may be relevant: [https://github.com/WiringPi/WiringPi/commit/96344ff7125182989f98d3be8d111952a8f74e15](url)

JamesGKent, 2017-06-28:
i agree this ought to be fixed, but the raspberrypi connection type is being depreciated in favor of the gpio connection type, assuming you have define all the pins in the config file can you try changing the connection type?

if you are using the default pins in the raspberrypi connection type but haven't declared them, add them into the config file first or this will not work.

haraldg, 2017-06-28:
Which vesion of lcdproc?

As noted the raspberrypi cennection type is deprecated in favor or the gpio connection type. So this is not a valid issue IMO.

terba, 2017-06-28:
It was compiled from github master branch at 2017-05-29

terba, 2017-06-28:
I have to recompile then because gpio support was not compiled in. https://github.com/mhei/libugpio is the required library?

haraldg, 2017-06-28:
Yes, the link is correct. Since you are recompiling, maybe you can give the release candidate a try?

terba, 2017-06-28:
Of course, I will compile from master head. Thanks for the quick support!

haraldg, 2017-06-28:
If possible, please compile from the release (candidate) tar ball. This tests the build system in a different way...

terba, 2017-06-28:
OK

terba, 2017-06-28:
I could compile and package it for piCore(Player) without any trouble. It works well with the gpio connection type on a rPi 1. Let's see if @ElFishi can use it on a Zero W. Thanks for the support.

haraldg, 2017-06-28:
Thanks for your input.

ElFishi, 2017-06-29:
I can confirm that LCDproc works with a Rapsberry Pi Zero W and ConnectionType=gpio.

closed 2017-06-29 by ElFishi