lcdproc/lcdproc #87: Option to disable the parameter display on startup

terba, 2017-06-28:
I'm always removing the code in the hd44780 driver which displays the basic parameters and waits 2 secs at startup. It would be more easier for me if it would be a configuration setting. I do this:

```C
diff --git a/server/drivers/hd44780.c b/server/drivers/hd44780.c
index a53639e..920a47a 100644
--- a/server/drivers/hd44780.c
+++ b/server/drivers/hd44780.c
@@ -391,55 +391,6 @@ HD44780_init(Driver *drvthis)
 
 	/* Display startup parameters on the LCD */
 	HD44780_clear(drvthis);
-	sprintf(buf, "HD44780 %dx%d", p->width, p->height);
-	HD44780_string(drvthis, 1, 1, buf);
- 	switch(if_type) {
- 	  case IF_TYPE_USB:
-  		sprintf(buf, "USB %s%s%s",
- 			 (p->have_backlight?" bl":""),
- 			 (p->have_keypad?" key":""),
- 			 (p->have_output?" out":"")
- 			);
- 		break;
- 	  case IF_TYPE_SERIAL:
- 		sprintf(buf, "SERIAL %s%s%s",
- 			 (p->have_backlight?" bl":""),
- 			 (p->have_keypad?" key":""),
- 			 (p->have_output?" out":"")
- 			);
- 		break;
- 	  case IF_TYPE_I2C:
- 		sprintf(buf, "I2C %s%s%s",
- 			 (p->have_backlight?" bl":""),
- 			 (p->have_keypad?" key":""),
- 			 (p->have_output?" out":"")
- 			);
- 		break;
- 	  case IF_TYPE_SPI:
- 		sprintf(buf, "SPI %s%s%s",
- 			 (p->have_backlight?" bl":""),
- 			 (p->have_keypad?" key":""),
- 			 (p->have_output?" out":"")
- 			);
- 		break;
- 	  case IF_TYPE_TCP:
- 		sprintf(buf, "TCP %s%s%s",
- 			 (p->have_backlight?" bl":""),
- 			 (p->have_keypad?" key":""),
- 			 (p->have_output?" out":"")
- 			);
- 		break;
-	  case IF_TYPE_PARPORT:
- 	  default:
- 		sprintf(buf, "LPT 0x%x%s%s%s", p->port,
- 			 (p->have_backlight?" bl":""),
- 			 (p->have_keypad?" key":""),
- 			 (p->have_output?" out":"")
-  			);
-  	}
-	HD44780_string(drvthis, 1, 2, buf);
-	HD44780_flush(drvthis);
-	sleep(2);
 
 	return 0;
 }
```

terba, 2017-06-28:
Or could be shown only if LCDd is running with a higher ReportLevel (or -r command line parameter).

haraldg, 2017-06-28:
I think this code is misplaced where it is (and also is a dirty hack). If people really want this, then it should be somewhere in the server, not in a driver. I suggest let's drop it.

Unless somebody speaks up in favor of this code soon, PR welcome.

terba, 2017-06-29:
My vote is also dropping those lines. It is useless even at configuration time as if screen is working, one knows how it is connected. Later on it is even more useless/annoying.

commit 419c7b9d (2019-06-08): hd44780: Don't display parameters at startup

This is not really a reasonable thing for a driver to do. No other
drivers do it AFAIK and people have been complaining about it. (Closes: #87)
closed 2019-06-08 by haraldg