lcdproc/lcdproc #91: hd44780: unknown ConnectionType: gpio

DareDevil73, 2017-09-09:
Hi, I get the "**hd44780: unknown ConnectionType: gpio**" when I try to start LCDproc by this config file:

> ...
>  [hd44780]
> ConnectionType=gpio
> Backlight=no
> Size=16x2
> ...

I configured my source folder by this command:

> ./configure \
>   --enable-debug \
>   --enable-drivers=hd44780,curses \
>   --disable-libusb \
>   --disable-libusb-1-0 \
>   --disable-libftdi \
>   --disable-libX11 \
>   --disable-libhid \
>   --disable-libpng \
>   --disable-freetype \
>   --disable-ethlcd

Why I get the above error!? :(

Thank you so much!
Antonio

haraldg, 2017-09-10:
Maybe libugpio is not installed on your system?

-- 
If you want to support my work:
see http://friends.ccbib.org/harald/supporting/
or donate via peercoin to P98LRdhit3gZbHDBe7ta5jtXrMJUms4p7w
or CLAM xASPBtezLNqj4cUe8MT5nZjthRSEjrRQXN

DareDevil73, 2017-09-10:
Sure, I built **libugpio** from sources, and now I get:

> Could not open driver module /usr/local/lib/lcdproc/hd44780.so: libugpio.so.1: cannot open shared object file: No such file or directory
> Driver [hd44780] binding failed
> Module /usr/local/lib/lcdproc/hd44780.so could not be loaded
> Could not load driver hd44780

haraldg, 2017-09-10:
So you either need to install libugio where the dynamic
linker of your system can find it or set
LD_LIBRARY_PATH to include the path of your libugio binary.

DareDevil73, 2017-09-10:
The symbolic link it is enough :wink:

haraldg, 2017-09-11:
So I guess this issue is solved then? Feel free to reopen, if you think otherwise.

closed 2017-09-11 by haraldg