lcdproc/lcdproc #98: Expose set_char and get_free_char over socket

SleepProgger, 2018-01-01:
It would be awesome if the Drivers `set_char` and maybe `get_free_char` would be exposed over the socket connection so clients could use custom characters.

I realize that this defeats the abstraction layer purpose of LCDproc a bit but IMHO it is better to have this feature only work for supporting devices instead of not having it at all.
Alternative to exposing `get_free_char` the server could add this information to the hello response.

Is there any chance a PR containing this change will be accepted ?
Otherwise i'll do it quick an dirty only for myself.

haraldg, 2018-01-02:
Both the driver API and the socket protocol would benefit from an update. See #73 and #74 for a few ideas collected so far. Handling custom characters, multibyte characters and extra symbols/icons is one of the areas where LCDproc currently is lacking. However I don't have time beyond reviewing the occastional PR and nobody else has been interessted so far. So there is not progress on this front ATM.

It is okay that not every feature works on every device LCDd supports. However making the driver API available on the socket seems like a bad idea to me for a lot of reasons. Such a quick and dirty PR would probably be rejected. OTOH if we can figure out a proper protocol change, then the implementation would be very welcome.

However before we start discussion what "proper" means in this case, I'd be really curious what problem requires you to control custom characters from the clients instead of say from `LCDd.conf`?

SleepProgger, 2018-01-02:
In my case mostly stuff like pixel"art", animation, logos and so on.
But it would also be handy for custom elements (nicer progress bars/slider and so on).
And since each screen can use different custom character that can't be really done via the config i think.

I already implemented it quick and dirty for me and one of the problems is that if exposed it needs to reset custom chars each time a screen is switched and custom characters changed. I believe this is already done for bignum.

One of the things i am not sure about implementation wise is the question if the client should be able to decide which character should be replaced or if the client should just tell the server "i want a custom character looking like this (8 bytes, at least for the hd44780)" and the server responds with the character number to use for it or an error if that screen already has all available custom char slots in use or doesn't support custom chars at all.

haraldg, 2018-01-02:
> In my case mostly stuff like pixel"art", animation, logos and so on.

I'm not sure how animation and logos are not pixelart, so in my mind I have folded this in one for now.

> But it would also be handy for custom elements (nicer progress bars/slider and so on).

There is already an implementation for that in LCDd: seamless hbars.

> And since each screen can use different custom character that can't be really done via the config i think.

Hm, actually that's one of the reasons why I think directly exposing the driver API would be a bad ideas: Different screens might use custom characters and the server needs some way to ensure that everything runs smoothly.

However, I see that an application trying to display generic pixelart might want to control custom characters as much as possible. So I kind of see your usecase. OTOH the bignum set of custom characters is already a quite good approximation of a generic pixelart set. And at lower resolution pixelart probably doesn't work all that well anyway. So how much is there really to gain for you? Ie how often do you actually end up with signigicantly different sets of custom characters when rendering pixelart?

A good guideline for extending the client protocol is always, that it should be possible to implement a reasonalbe fallback on/for devices that don't support some fancy feature. In this case my first thought is: Maybe the code handling the pixelart can be included into the server and the change to the protocol is a simple pixelart widget? (Then also all custom character handling would remain in the server as a nice sideeffect.) If not enough custom characters are available, it could fall back to some kind of ascii art.

An further good guideline is: It should be possible to use the protocol from a simple script or even interactively in a telnet/netcat session. Thus anything too dynamic is frowned upon. As is sending binary data over the link. The latter most likely also just doesn't work ATM.