lcdproc/lcdproc #100: LCDd basic menu edit

AndradeFelip, 2018-04-03:
I am having trouble making my own alterations to the LCDd menu that has "options" "screens"  on the menu even when the lcdproc server is off but LCDd is on. I only want to add menus there and use "menu_goto" on this level. Also "backlight" method does not work on menu or lcdproc screens. i want to use the lcdproc server and be able to modify it.

Thanks

haraldg, 2018-04-04:
Hi,

I'm not sure I understand what problem you have. What did you try in how did
it fail?

To add custom menus you'd usually either use lcdexec or write your own
client (might be some simple script).

AndradeFelip, 2018-04-04:
I have made a custom client and it fully works with itself. But I want to make it work with LCDd and LCDproc as well.

The LCDd when is running has a built in menu. I want to edit and access that.
Like this: My menu will point to LCDd options menu (already done) but the options menu will be set to go not back to LCDd menu, but to mine like this: ( menu_set_item {?} {options} -prev {my menu} )
for me this being the simplest way. I would need the name tree of the menu, all the IDs from the LCDd menu. I couldn’t find it anywhere. And if there is another option to modify it, I could do that too. 

Also I want to know if i can edit settings from the LCDproc screens like Backlight {on|off}.
I want to make a timeout from my screens but the LCDproc screens doesn't return anything to my client.

Thanks

haraldg, 2018-04-04:
This is clearer now, thanks!

I don't think it is possible to hijack other clientes (or the servers)
menus like that. I guess you would need to add a new configuration option
"allow_menu_hijack" to LCDd and LCDd.conf. In any case I also don't know
the menu_ids the server is using - I guess you will need to read to source
even for that.

And you definitely can't access the screens of other clients, because
screen ids are not unique among them: If your client tries to set the
backlight of a screen, the server *always* thinks it is refering to one
of it's own screens. If you want better backlight control of lcdproc
screens you will need to add the feature to lcdproc itself. (And preferably
send a patch/PR, if it is something generally useful.)

-- 
If you want to support my work:
see http://friends.ccbib.org/harald/supporting/
or donate via CLAM to xASPBtezLNqj4cUe8MT5nZjthRSEjrRQXN
or via peercoin to P98LRdhit3gZbHDBe7ta5jtXrMJUms4p7w

AndradeFelip, 2018-04-04:
thanks! 

that helps a lot. it is a lot of work, but I probably will do it.

Have a good day!

AndradeFelip, 2018-04-04:
Hi,

Is there a easy way to learn the source of LCDd without reading all of the files ? like a clean group of files that would work so that I could learn and add my stuff faster? I am lost on where to begin 

Thanks

haraldg, 2018-04-05:
If there is, I don't know about it (I started helping lcdproc only because
I sent in a patch and noticed it was essentially abandoned) - but LCDd source
isn't all that big (without the drivers) and the files all have quite
descriptive names. Shouldn't be too hard to get started I hope. For example
just looking at the names I'd guess that serverscreens.c is a good start
for you and then read other files for context when necessary.

AndradeFelip, 2018-04-05:
hi,
I have fount that this function

"void menuscreen_switch_item(MenuItem * new_menuitem)"

has no problem to call every menu possible. So LCDd source code menu_goto woks as a charm for what I want.
I was thinking maybe ignoring every Client call, with a parameter, and make all of them(Clients) NULL. 
In that way, we could set the LCDd menu to work with every client. Parameter with OFF default. 

On the other hand, your help on using serverscreens.c made possible for me to create my own menu inside LCDd and making the change in the menu not needed. But the setting for all Clients=NULL is still useful for Backlight and handling screens from other clients.

I am working on it, hope I was clear on what I am doing.
Have a good day.

haraldg, 2018-04-06:
Hi,

unfortunately it is not clear what you are doing - mostly I think I'm
missing the context of in which function do you want to set Client=NULL.

But I'm guessing you want to essentially merge the commands coming
from all clients into one. I don't think this would work well: screen ids
are not required to be gobally unique. E.g. multiple clients could have
a screen "s" - a lot of confusion would happen.

Also I don't see the use case of changing the backlight settings of
one client from an other. What actual problem are you trying to solve?
Wouldn't it be more natural just to improve the client itself?

I think if you really want to merge connections of multiple input
streams into a single client there could be two possible ways:
1) some kind of lcdproc proxy-server that would merge connections
   from multiple clients and forward them as one client connection to
   the real LCDd.
2) An amendment of the client protocol to allow attaching new connections
   to an existing client.

HTH,
Harald

-- 
If you want to support my work:
see http://friends.ccbib.org/harald/supporting/
or donate via CLAM to xASPBtezLNqj4cUe8MT5nZjthRSEjrRQXN
or via peercoin to P98LRdhit3gZbHDBe7ta5jtXrMJUms4p7w

haraldg, 2021-07-20:
I think this conversation is resolved. If not, feel free to reopen and provide additional input.

closed 2021-07-20 by haraldg