lcdproc/lcdproc #113: Segfault with using disk plugin

y2kbadbug, 2018-10-30:
Hello,

I currently am running lcdproc with the LIS driver and lcdproc seems to crash on me when loading the disk plugin.  I determined this from the debug output.
```
[root@daniel-desktop ~]# lcdproc -v
LCDproc 0.5.9

[root@daniel-desktop ~]# gdb lcdproc
GNU gdb 6.1.1 [FreeBSD]
Copyright 2004 Free Software Foundation, Inc.
GDB is free software, covered by the GNU General Public License, and you are
welcome to change it and/or distribute copies of it under certain conditions.
Type "show copying" to see the conditions.
There is absolutely no warranty for GDB.  Type "show warranty" for details.
This GDB was configured as "amd64-marcel-freebsd"...
(gdb) run -f
Starting program: /usr/local/bin/lcdproc -f

Program received signal SIGFPE, Arithmetic exception.
0x00000000004051e2 in disk_screen (rep=<value optimized out>, display=<value optimized out>, flags_ptr=<value optimized out>)
Current language:  auto; currently minimal
(gdb) bt
#0  0x00000000004051e2 in disk_screen (rep=<value optimized out>, display=<value optimized out>, flags_ptr=<value optimized out>)
#1  0x0000000000402d69 in update_screen (m=<value optimized out>, display=<value optimized out>) at mode.c:74
#2  0x0000000000402cb9 in main_loop () at _ctype.h:141
#3  0x0000000000402859 in main (argc=<value optimized out>, argv=<value optimized out>) at main.c:356
```
This is curently on FreeBSD 11.2 AMD64.

I can disable the disk plugin and the problem goes away, so I believe its isolated to that.

haraldg, 2018-10-30:
Thanks for the report.

It is a bit difficult to tell without full debug information, but I guess
this is caused by a division by zero and the only line I can find, where
this might happen is:

table[i].full = (huge) (lcd_cellwid * gauge_wid)
	* (huge) (mnt[i].blocks - mnt[i].bfree)
	/ (huge) mnt[i].blocks;

Do you per chance have a filesystem with zero blocks?

I guess we should guard against this case.

BTW: We should also get rid of #define huge long long int ...

y2kbadbug, 2018-11-01:
I found this in my "df"

```
/dev/fuse                           0B      0B      0B   100%    /usr/home/daniel/.gvfs
```
I am almost guessing the .gvfs partition might be causing it.

commit f2489a39 (2018-11-08): lcdproc: Protect against division by zero

Closes: #113
closed 2018-11-08 by haraldg