lcdproc/lcdproc #118: RFE: fold libmx5000 code into the mx5000 driver

jwrdegoede, 2019-04-17:
It looks like mx5000tools upstream is dead / inactive. The closest thing to an upstream I could find is an old github mirror and a pull-req there has so far gone unanswered:
https://github.com/coolacid/mx5000tools-revo/pull/1

I've been working on packaging mx5000tools for Fedora, so that the Fedora lcdproc package can be built with mx5000 support, but I think it is probably a better idea to just include the bits of libmx5000 which lcdproc needs directly into the lcdproc mx5000 driver and forget about libmx5000 as a standalone project.

It looks like the code which we need is about 600 lines if copied 1:1, probably less when modified ro better fit in lcdproc (e.g. directly embed necessary data into the driver struct, rather then malloc-ing it separately).

I'm willing to do the work for this and submit a pull-req with the necessary changes. The reason for filing this RFE issue is because I first wanted to know what you think about this, before spending time on it only to have the pull-req denied because you think it is a bad idea...

So what do you think about integrating the few bits of libmx5000 which we need directly into lcdproc?

haraldg, 2019-04-17:
Nothing you write above suggested to me that we should include libmx5000 in lcdproc. I think the proper thing is for you (or somebody else) to start maintaining mx5000tools upstream.

If I understand the information in the README correctly, then @coolacid never intended to maintain it so his lack of response isn't too surprising. But maybe he can still chime in here and share his thoughts. I also note, that mx5000tools seems actively maintained in gentoo. Thus such a decision should be coordinated with them as well.

Since quite clearly there hasn't been an active upstream in a long time, but there are a few active users showing up here an there, my recommendation is to conact all users you can find with reasonable effort and figure out among yourselves who is adopting this as new upstream.

jwrdegoede, 2019-04-17:
On 17-04-19 21:33, haraldg wrote:
> Nothing you write above suggested to me that we should include libmx5000 in lcdproc. I think the proper thing is for you (or somebody else) to start maintaining mx5000tools upstream.

AFAICT there are no other users of libmx5000 other then lcdproc and mx5000tools in and by itself is not all that useful.

So sure I can maintain mx5000tools as a separate upstream project, but that just does not feel like it is a very useful way to spend my time.

OTOH if we move the few 100 lines of code to inside the lcdproc mx5000 driver then all lcdproc builds can have mx5000 support out of the box.

Where as currently e.g. both Fedora and Debian contain lcdproc, but build it without mx5000 support.

haraldg, 2019-04-17:
I understand that this is tempting, but in the end it seems a zero sum
game: Either you maintain the code upstream or somebody maintains
it in lcdproc (which isn't all that well maintained itself). Naturally
I prefer the former, and the difference in effort doesn't look all that big.

And as noted previously: Gentoo is using this code. I won't agree to taking
this unless they are in support of the proposal too.

Also there are @coolacid and the people forking/starring his mirror
and there is mx5500-set, which claims to be based on the same code.

If all these people agree that they are just interested in mx5000tools
as dependency of lcdproc, then I probably wont block this. OTOH just
establishing this consensus seems more trouble then setting up a new
upstream repository somewhere.

jwrdegoede, 2019-04-17:
Ok, fair enough. I'll move forward with getting a mx5000tools package added to Fedora then.

closed 2019-04-17 by jwrdegoede