lcdproc/lcdproc #122: New Release?

y2kbadbug, 2019-05-06:
I just wanted to ask when a new release was planned?  I noticed there has been a lot of activity since the last release and some things seem more vital.

Also, if there is anything I can do to help with QA, I am open to it.

haraldg, 2019-05-06:
Right now we see unusually high activity. I guess I will only consider cutting a release once this has died down again. Also since we are going to bump the protocol version with the next release, I want to get a number of protocol related changes in before.

As far as QA is concerned: Right now nobody seems to actively test master on non-linux OSes. Doing so from time to time and at least when there is a release candidate, would help with portability a lot.

Of course any other help is welcome too. What driver are you using?

dod38fr, 2020-07-24:
I'd welcome a new release. 

On Debian, I've recently patched lcdproc to compile with gcc-10 even though this was fixed here in February.

All the best

dod38fr, 2023-11-18:
Hi
It may be time for a new release since no commit was done since 2021. I'm also interested in a release because usb support was fixed.

All the best

ethandicks, 2023-11-22:
I am working towards a new release in the next 30 days or so.  Target is definitely by the end of 2023.