lcdproc/lcdproc #131: Raspberry Pi 4B support

PaulWebster, 2019-07-03:
The Raspberry Pi 4B model has a different peripherals address.
Rather than hard-coding the value it would probably be better (and hopefully more future proof) to use the official RPi library to determine the base at run-time.
There is an issue with this though ... the RPi folks have not yet released an update to make it work but it is on their list to do.

See https://www.raspberrypi.org/documentation/hardware/raspberrypi/peripheral_addresses.md for the official way to do it and
https://www.raspberrypi.org/forums/viewtopic.php?f=33&t=244031&sid=9ae061e82d029dcf862ed051293d18cc#p1491303 for 
RPi forum post about it

PaulWebster, 2019-07-03:
However, it would add to the pre-reqs to get the RPi include file.

haraldg, 2019-07-03:
Hi Paul,

the RapsberryPI connection type is deprecated. The future proof way is to use the gpio connection type instead. Incidentally there is a open pull request #119 to improve the gpio connection type that needs testing. Since I guess you are using lcdproc on a RPi: Could you test this PR for us?

PaulWebster, 2019-07-24:
I don’t have access to RPi4 with nearby lcd until September ... but I saw someone asking questions about getting this working on RPi forum so I will highlight this over there.

tuxfamily, 2020-01-07:
@PaulWebster Thanks for the links, this was very usefull.
I lost a day because of this; each posts refer to the Pi 2/3 offset (0x3F000000), thanks to these links I was able to find the correct address for the Pi 4 and to build a driver with the hard coded gpio base address (0xfe200000).
Attached is my build (caution to reader, it's only for the Pi4!)
[hd44780.so.zip](https://github.com/lcdproc/lcdproc/files/4030091/hd44780.so.zip)

jduffas, 2020-05-21:
Bonjour, j'ai essayé votre driver pour Pi 4, et j'ai ce message d'erreur 

Driver [hd44780] init failed, return code -1
Could not load driver hd44780
There is no output driver
Critical error while initializing, abort.

avez-vous une idée du problème?

Merci.

Jean

jduffas, 2020-05-21:
Hello, I tried your driver pour Pi 4, and I got this error message : 

Driver [hd44780] init failed, return code -1
Could not load driver hd44780
There is no output driver
Critical error while initializing, abort.

do you have an idea?

Thank you.

Jean

tuxfamily, 2020-05-25:
@jduffas Sorry, not enought info to help you. But my driver is probably too old. I recompiled it with latest version of everything :) Unfortunatly, I'm not using it anymore (I switched to i2c), so I can't test it.
[hd44780.so.zip](https://github.com/lcdproc/lcdproc/files/4677425/hd44780.so.zip)

jduffas, 2020-05-25:
Hello,
I did it working with original driver,
I had to select « gpio » instead of raspberry (I think)
Thank you :-)

Jean

ethandicks, 2021-07-26:
I just tested #119 and left some comments.  After changing a couple of #ifdefs, I did get that to work.  I can definitely confirm that the raspberrypi connection does not find the right GPIO base for a Pi 4B (or a Pi 400)