lcdproc/lcdproc #140: Display crash and LCDd cant reuse it

username7291, 2019-10-25:
I have a HD44780 20x4.
First i attached it in 4 Bit Mode onto Raspberry GPIO. There was sometimes permanet Garbage on the Display. "service LCDd restart" dont help. Cryped stuff is now on the display, it change
with new Data. I need to unpower the Display.
Then cause i need Pins for SPI i connected the Display via I2C to Parallel in 4 Bit Mode.
The same thing. I look on the Display, no capicators on it, maybe the problem, added 100nF and 4µ7. Dont help. I messure unused Data Pins, internal Pullups found so this should be no Problem.
All Delay/Refresh options tryed.
Does lcd prog initalise the 4 Bit Mode right? I googled about and found "Your initialisation code to set the display in 4 bit mode is probably faulty and only works when the LCD has had a power on reset"  
 // Set 8 bit mode 3 times
 // If the LCD is in 8 bit mode it will be set again 3 times.
 // If the LCD is in 4 bit mode expecting the 2nd nibble the 
 // first write will be lost, the next two set 8 bit mode

And if LCD proc do so, do it too a Softreset 3 times on Startup?
Garbage may happent but i cant image why lcd proc is unable to reset the display.

haraldg, 2019-10-28:
I don't have a HD44780 myself, so I can't beu too helpful. However from the symptoms you describe, I'd exepect a problem with the physical connection like voltage incompatibility.

If your controller needs a stricter initialization sequence, we happily accept patches.

mskalski, 2019-10-28:
What is exactly model of your LCD? And what version of LCDd?

I had similar problems with Winstar OLED, garbage on lcdproc daemon restart, so I changed slightly LCD initialization according to original Hitachi datasheet and added more delays for Winstar OLED display.

See commits  bf717f1599ed24cb517ba2b808a2f259daa7130a and e8cfdd2feda1806788e9df7a9050a06ffdb35374, and PR #96

username7291, 2019-11-09:
It was first connected to raspberry, with in mind that raspberry cant handle 5V. Means the Display is configured as write only and has series resistors. Then it was connected via I2C PCF8574. This chip can work with 5V display and 3V I2C Bus of the rasperry(remember to disconnect the I2C pullups on ready boards from ebay). 

@mskalski
First i think i have the newest version cause all have the same version name. I compared it and seen
the new code has the missing Poweron Routines. So i installed the new Version from here(git clone).
It has the same issue, first garbadge or empty display, after restart garbadge signs(but the have somewhat todo with the stuff that should displayed).
The I2C should be slower than a direct connection, it is limited to 100khz and need transfer the Chip Adress and 8 Bits. I doesnt find a 16Bit LCD->I2C Board and too dont find support for this
in the Sourcecode(people tell they have less or no Problems with LCD in 8 Bit Mode).
The Display is a 20x4 "2004A" is on the Backside, it has 5 Chips. Its the most selled on ebay.