lcdproc/lcdproc #144: hd44780-usb4all hardware initialisation wrong way around?

scp38, 2020-01-05:
Dear all,

after running lcdproc for many years in hd44780-winamp mode I had to switch the mainboard and so had no parallel port any more. So I changed over to usb4-all since it offers also support for my 6 button keyboard.

After compiling (and yes, including the files for libusb-0.1) I noticed that the display only worked fine after the second start of the LCDd service.

I found in the hd44780-usb4all.c that the display initialisation is done BEFORE the USB4all-interface initialisation. So only after the second start of LCDd service, the display was correctly initialized.

IMHO there should be a change from 

	common_init(p, IF_4BIT);

	/* replace uPause with empty one after initialization */
	p->hd44780_functions->uPause = usb4all_HD44780_uPause;

	/* initialize usb-4-all controller */
	usb4all_init(p);

to 

	/* initialize usb-4-all controller */
	usb4all_init(p);

	common_init(p, IF_4BIT);

	/* replace uPause with empty one after initialization */
	p->hd44780_functions->uPause = usb4all_HD44780_uPause;

At least this code works for me now with the very first start of the LCDd service.

Regards,
Paul

haraldg, 2020-01-05:
@ThoMo I believe this is your driver. Can you comment on this?

thomo, 2020-01-06:
Never mentioned the issue before (at least I can't remember) and I also can't test it now (I do not use it anymore). Any way - after looking in the code the suggested change sounds reasonable to me.

closed 2020-01-07 by haraldg

scp38, 2020-01-07:
Wow! Thank you very much for your quick feedback. Very much appreciated!

commit 2f605770 (2020-01-07): server/drivers/hd44780-usb4all.c Improve initialisation

Closes: #144