lcdproc/lcdproc #160: New release?

ConiKost, 2020-08-18:
Are there any plans for a new release?

haraldg, 2021-07-20:
I hope to have things cleaned up enough for a release by the end of (northern hemisphere) summer.

Also this probably will be the last release done by me. If anybody considers picking up the work: Now is a good time to get involved.

ethandicks, 2021-07-21:
I am happy to pick this up going forward.

haraldg, 2021-07-23:
> I am happy to pick this up going forward.

Great! Then let's work on the upcoming release together!

ethandicks, 2021-07-28:
Happy to learn the ropes.

haraldg, 2021-09-27:
As far as I'm concerned, we are ready to start the release process. There are many things, that I would like to see fixed in the upcoming release, but it just won't happen from my side in any reasonable time frame. Better to go with what we have, I think.

@ethandicks would you like to create the release branch yourself? (At least this time I'm around for any questions. :-) )
A checklist/guide for preparing a release (candidate) is in:
https://github.com/lcdproc/lcdproc/blob/master/docs/release-checklist.md

ethandicks, 2021-09-27:
I will review the process and see about kicking it off

haraldg, 2022-01-21:
Any progress on this front? Do you need help with anything?

ConiKost, 2022-12-28:
Seems one year passed without any activity..

ethandicks, 2022-12-31:
I could probably use some handholding to get started.  Some personal
disasters a year ago (Dec and Jan) including no electricity for weeks
totally derailed my work.

Would love to get back on track.

-ethan

ConiKost, 2023-01-01:
Sorry to hear! Hope you will get well :-)
If you need some help, maybe I can help?

haraldg, 2023-01-03:
@ethandicks sorry to hear about your misfortune last year. I hope 2023 is better to you!

The way I did the releases is documented in https://github.com/lcdproc/lcdproc/blob/master/docs/release-checklist.md but that is not set in stone. Mostly it is just about updating all the version and date strings.

Usually I first announced a release candidate on the mailing list before making the release official. We don't have the mailing list anymore, so not sure, how you will do that. Maybe just drop the release candidates or maybe announce them to the known package maintainers to get some feedback. It shouldn't matter all that much anymore: I guess, a strict release procedure was more important when we were using CVS. Nowadays we can move change sets between branches quite easily.

@ConiKost I think more hands to help with lcdproc are always welcome. I don't think release management is particularly difficult, but you would need to get git write access. (I don't have the privileges to grant this to you, I believe.) I'm not sure, what the rules for write access are. You are quite well established on github but not so much with lcdproc. Maybe the admins would want to see some more PRs from you first?

Also testing the release tar ball on all platforms you care about, would be very helpful. (See the discussion about release candidates above.)

ConiKost, 2023-01-03:
> Maybe the admins would want to see some more PRs from you first?

That's fine. My intention wasn't to get write access to Repo but more to help by filling PRs :-) I will see, what I can do for PRs for the release-checklist.md.

ethandicks, 2023-11-22:
Reviewing issues here, I plan to kick off a new release in approximately 30 days.  This will give me time to prepare for regression testing the drivers I usually test (largely Matrix Orbital, Crystalfontz, and jw002) and to give time to review any pull requests that should be in this release.

strophy, 2025-09-02:
@ethandicks any plans or progress on this?