lcdproc/lcdproc #166: MtxOrb has trouble reading device info on some displays

bedaro, 2021-02-07:
I have a rather old Matrix Orbital LK202-24-USB that has long had a minor issue where LCDd cannot read the device type/serial number. This shows up in the logs every time LCDd starts:

```
MtxOrb: opened display on /dev/lcd0
MtxOrb: unable to read device type
MtxOrb: unable to read device firmware revision
MtxOrb: unable to read device serial number
Display detected: Matrix Orbital, Unknown model (0x00), Firmware Rev.: 0x00 0x00, Serial No: 0x00 0x00
```

This is also something I've seen reported elsewhere on the web, but not since 2012. This has been an issue for a long time, and I just reproduced it with the latest release 0.5.9.

I think the cause is the order in which commands are sent to the display to initialize it. I reproduced the problem with a simple python script (below) that sends all of the same commands in the same order as MtxOrb_init. With the backlight command commented out the display returns a byte for the module type command, but if that backlight command is sent the device type read hangs forever. I also noticed that sending the clear display command (254 88) makes the display respond to the device type query again. All this suggests that a simple reordering of the function calls at the end of MtxOrb_init would fix this problem, and I can submit a pull request, but it will of course need testing on other displays.

```
#!/usr/bin/env python3

import serial

device = '/dev/ttyUSB0'

ser = serial.Serial(device, 19200, 8)
# Hardware clear
ser.write(b'\xFEX')
# Line wrap on
ser.write(b'\xFEC')
# Autoscroll off
ser.write(b'\xFER')
# Cursor blink off
ser.write(b'\xFET')
# Default contrast
ser.write(b'\xFEP\x7A')
# Turn on backlight
# Enabling this will cause the display to fail to respond to the device
# type query that comes next!
#ser.write(b'\xFEB')

# Test: get device type
ser.write(b'\xFE7')
x = ser.read()
print(x)
```

bedaro, 2021-02-07:
Correction: while the phenomemon I found in my Python script is odd, that's not the issue in LCDproc. Testing with a trunk build found that increasing the timeout in a prior commit allowed the driver to read the model name, but the serial number and version timeouts are still too short. Using perf_counter() finds that it takes about 25 ms between writing the command and reading the response, so increasing the other two timeouts in get_info() fixes the problem with my display.

ethandicks, 2021-02-07:
On Sun, Feb 7, 2021 at 1:46 PM Ben Roberts <notifications@github.com> wrote:
> Correction: while the phenomemon I found in my Python script is odd, that's not the issue in LCDproc. Testing with a trunk build found that increasing the timeout in a prior commit allowed the [...]

I encountered something similar when adding the LK205-7T to the list
of known modules in 2019.  We haven't had a code release since then,
but  I increased the timeouts in get_info() to 40ms and 10ms (based on
observation with that module) and merged that change.  Please checkout
the current master branch and test.  If we need to extend the one
timeout to 25ms because of the LK202-24-USB, we can do that.

-ethan

ethandicks, 2021-02-07:
Re-reading your message, if by "trunk build" you mean the current master branch, yes, I get what you are saying now.  We can bump the two 10ms timeouts to 25ms and I don't expect any negative impact on other modules.  Please test with your LK202-24-USB and confirm that 25ms is enough (if you would be so kind as to attach the output from a successful run, that would be fantastic).

bedaro, 2021-02-07:
Yes, I meant the master branch (sometimes SVN language creeps in). I had tested with 40 ms, but a new test with 25 ms is also enough to do it. Here's the successful output:
```
MtxOrb: opened display on /dev/lcd0
Display detected: Matrix Orbital, Model: LK202-24-USB, Firmware Rev.: 0x21, Serial No: 0xfffffffe 0x58
```

commit 89b9c779 (2021-07-02): extended the wait time when reading firmware rev and serial number. Fixes #166
closed 2021-07-02 by ethandicks