lcdproc/lcdproc #174: Issued building hd44780

rogueops, 2021-05-09:
Using FreeBSD 12.2 and able to easily built new versions of the server and client apps and the drivers I has used before, but i just keep getting errors when trying to build hd44780.so, I tried:

./autogen.sh
./configure --enable-drivers=hd44780
# if i swap  hd44780 with one i have used before no issues, if I tried to build all it gets stuck on just hd44780
then ran make all

the errors point to issues with the files or unable to build the given file, what am I doing wrong?

kripton, 2021-05-09:
Would you mind posting the exact error messages? Otherwise, it's almost impossible to judge what might be wrong.
And it would be great to know which exact version or git commit you are trying to build

rogueops, 2021-05-10:
i did autogen.sh, the configure and enabled hd44780, then make and I get
this, i currently use crystalfontz so that builds, but my new box uses an
EZIO-100 so will need the HD44780 driver for when i get that box up

hd44780-bwct-usb.c:68:3: warning: implicit declaration of function
'usb_init' is invalid in C99 [-Wimplicit-function-declaration]
  usb_init();
  ^
hd44780-bwct-usb.c:69:3: warning: implicit declaration of function
'usb_find_busses' is invalid in C99 [-Wimplicit-function-declaration]
  usb_find_busses();
  ^
hd44780-bwct-usb.c:70:3: warning: implicit declaration of function
'usb_find_devices' is invalid in C99 [-Wimplicit-function-declaration]
  usb_find_devices();
  ^
hd44780-bwct-usb.c:72:6: error: no member named 'usbHandle' in 'struct
hd44780_private_data'
  p->usbHandle = NULL;
  ~  ^
hd44780-bwct-usb.c:73:6: error: no member named 'usbIndex' in 'struct
hd44780_private_data'
  p->usbIndex = 0;
  ~  ^
hd44780-bwct-usb.c:74:14: warning: implicit declaration of function
'usb_get_busses' is invalid in C99 [-Wimplicit-function-declaration]
  for (bus = usb_get_busses(); bus != NULL; bus = bus->next) {
             ^
hd44780-bwct-usb.c:74:12: warning: incompatible integer to pointer
conversion assigning to 'struct usb_bus *' from 'int' [-Wint-conversion]
  for (bus = usb_get_busses(); bus != NULL; bus = bus->next) {
           ^ ~~~~~~~~~~~~~~~~
hd44780-bwct-usb.c:74:54: error: incomplete definition of type 'struct
usb_bus'
  for (bus = usb_get_busses(); bus != NULL; bus = bus->next) {
                                                  ~~~^
hd44780-bwct-usb.c:45:10: note: forward declaration of 'struct usb_bus'
  struct usb_bus *bus;
         ^
hd44780-bwct-usb.c:77:19: error: incomplete definition of type 'struct
usb_bus'
    for (dev = bus->devices; dev != NULL; dev = dev->next) {
               ~~~^
hd44780-bwct-usb.c:45:10: note: forward declaration of 'struct usb_bus'
  struct usb_bus *bus;
         ^
hd44780-bwct-usb.c:77:52: error: incomplete definition of type 'struct
usb_device'
    for (dev = bus->devices; dev != NULL; dev = dev->next) {
                                                ~~~^
hd44780-bwct-usb.c:75:12: note: forward declaration of 'struct usb_device'
    struct usb_device *dev;
           ^
hd44780-bwct-usb.c:81:14: error: incomplete definition of type 'struct
usb_device'
      if (dev->descriptor.idVendor != BWCT_USB_VENDORID) {
          ~~~^
hd44780-bwct-usb.c:75:12: note: forward declaration of 'struct usb_device'
    struct usb_device *dev;
           ^
hd44780-bwct-usb.c:85:26: error: incomplete definition of type 'struct
usb_device'
      for (c = 0; c < dev->descriptor.bNumConfigurations; c++) {
                      ~~~^
hd44780-bwct-usb.c:75:12: note: forward declaration of 'struct usb_device'
    struct usb_device *dev;
           ^
hd44780-bwct-usb.c:87:17: error: no member named 'usbIndex' in 'struct
hd44780_private_data'
        for (p->usbIndex = 0; p->usbIndex < dev->config[c].bNumInterfaces;
[... 57 lines cut ...]
           ^
hd44780-bwct-usb.c:94:47: error: no member named 'usbIndex' in 'struct
hd44780_private_data'

 (dev->config[c].interface[p->usbIndex].altsetting[a].bInterfaceSubClass ==
0x01)) ||
                                           ~  ^
hd44780-bwct-usb.c:95:21: error: incomplete definition of type 'struct
usb_device'
                (dev->descriptor.idProduct == BWCT_USB_PRODUCTID)) {
                 ~~~^
hd44780-bwct-usb.c:75:12: note: forward declaration of 'struct usb_device'
    struct usb_device *dev;
           ^
hd44780-bwct-usb.c:98:18: error: no member named 'usbHandle' in 'struct
hd44780_private_data'
              p->usbHandle = usb_open(dev);
              ~  ^
hd44780-bwct-usb.c:98:30: warning: implicit declaration of function
'usb_open' is invalid in C99 [-Wimplicit-function-declaration]
              p->usbHandle = usb_open(dev);
                             ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
6 warnings and 20 errors generated.
*** Error code 1

rogueops, 2021-05-10:
version is whatever is currently up, i downloaded it at 2AM on the 9th MST

rogueops, 2021-05-10:
this looks like my issue but I am on freebsd and not debian

https://github.com/lcdproc/lcdproc/issues/57

was tired when i did this so probably missed the above ticket