lcdproc/lcdproc #175: Digipos WD-202 (PC-POS) Screen Issue

sxiii, 2021-05-29:
Greetings!

I've got working screen (I know as it lights up and goes through demo information about itself) Digipos WD-202 (on the chip sticker it also says PC-POS). I've put the screen into default CD5220 mode with DIP switches according to this documentation: http://www.goodsononline.com.au/download/User_manual/fps2002_um.pdf

When the screen starts I see that it says that it's in this mode. Screen itself is 20x2 while using 5x7 dots. I also know that screen should be on port /dev/ttyS0 because I've tested it with shortening the TX/RX pins to see echo and it worked.

This is my LCDd.conf file (only the important / meaningful part):
```bash
DriverPath=/home/sd/lcdproc/server/drivers/
Driver=serialPOS
Bind=127.0.0.1
Port=13666
User=nobody
[serialPOS]
Device=/dev/ttyS0
Size=20x2
Cellsize=5x7
Custom_chars=0
Type=CD5220
Speed=9600
```

But as you guesses the screen does not work. Additional information:
* EndeavourOS 1.4 (rolling release) (Arch based distro)
* Kernel 5.10.11-arch1-1 #1 SMP PREEMPT
* I'm running all commands with root
* That's how I start LCDd: `sudo killall LCDd && ./server/LCDd -c ./LCDd.conf` (or tried manually in two steps as well)
* That's how I start lcdproc: `sudo ./clients/lcdproc/lcdproc -s localhost -p 13666 -f C M L`
* Both previous commands are running from root directory of the project
* The project was just recently downloaded and compiled with all drivers according to instruction

So; 
1) Do you have any ideas what I could check to make the screen work?
2) My screen have the following modes: DSP800, ESC/pos, ADM 787, ADM 788, EMAX, UTC/P, UTC/S, CD5220. Should I maybe try something else than CD5220 which is more stable? Which mode and driver should I use in this case? (I've tried with EMAX mode and it didn't worked as well).

Thanks a lot!

shenghaoyang, 2021-05-29:
Hello!

If you get a chance, could you try the AEDEX protocol?

That should be the simplest.

sxiii, 2021-05-30:
Heyhey @shenghaoyang yes I've tried EMAX; AEDEX and Epson. None of them works sadly :( I'm starting to think maybe I hadn't connected all pins correctly... But RX and TX seems to be worked at the beginning as I've checked them with short-circuit method...