lcdproc/lcdproc #176: [Question] Matrix Orbital key protocol

vintagepc, 2021-06-13:
Forgive the Noob question - I had a rummage around in the source and some datasheets and this isn't completely obvious to me right now. 

How are key events received or polled from Matrix orbital-type displays? Is it a poll to the display itself or is it a composite USB device with a secondary channel for keystrokes?

To provide some context... I'm trying to determine whether it's possible to use https://www.adafruit.com/product/781 and alter the arduino source such that the outputs can be turned into a simple up/down/left/right keypad - and the first step to that is understanding how the protocol reports keys. The datasheet from Matrix isn't particularly helpful as they only mention enabling/disabling autopoll, not the command sent to report a press or the command to poll key state. :-/

ethandicks, 2021-06-14:
There is only one communication channel (RS-232 or USB) for Matrix Orbital displays.  Commands and responses go over the same channel.  The keypad isn't separate.

vintagepc, 2021-06-14:
Thanks for the reply. So looking at https://github.com/lcdproc/lcdproc/blob/master/server/drivers/MtxOrb.c around line 1500, I gather the key input is literally just an incoming character from the device node. Seems pretty straightforward to do what I'm hoping to achieve then!

closed 2021-06-29 by ethandicks

vintagepc, 2021-06-29:
Sorry, forgot to follow up and close this. I did confirm it works as expected if the Teensy code scans a keymatrix and just returns characters over serial.

ethandicks, 2021-06-29:
No worries.  I was checking other issues and saw this was resolved but unclosed.