lcdproc/lcdproc #177: Matrix Orbital Tri-Color Support

braewoods, 2021-06-29:
I noticed some displays are sold as having Tri-Color which defaults to White. Are there any plans to add support for this so the text color can be modified? The command sequence is 0xFE 0x82 and is called "Set Backlight Color". I can't seem to find any way to detect this so I surmise the only real option is to only issue the command if the user has set an option or variable for it.

ethandicks, 2021-07-20:
There is no feature support for colored backlights.  The current API only supports on, off, and blinking modes, and certain drivers support server-side settings for backlight brightness if the hardware allows it.  This would be a new feature that would be supported at the driver level to set a fixed color or would require expanding the API to allow clients to select different backlight colors.  

There are currently no plans to add this feature, but this issue can be a starting point.

braewoods, 2021-07-20:
I was only thinking of basic support to configure the color during driver initialization since it's not a very common feature.