lcdproc/lcdproc #178: Disk screen has buffer overflow and crashes LCDproc

vintagepc, 2021-07-07:
Observed today when I plug in and mount a USB drive on my computer, LCDProc will crash and refuse to re-launch. 

v0.5.7

```
 lcdproc -f -c /etc/lcdproc.conf 
*** buffer overflow detected ***: terminated
Aborted (core dumped)

```

Disabling the disk screen in /etc/lcdproc.conf (or unmounting the drive and restarting lcdproc) will resolve the issue. 
I have two drives that do this and several others that do not. The two that cause it are both USB3.0/32GB drives formatted as FAT32. 

Problem does not seem USB3 related as it also manifests if the same drive is plugged into a USB2 port. 

Bottom-most drive is the culprit. 
[image]

Display is a USB-serial connected 40x4 using the MtxOrb protocol.

haraldg, 2021-07-19:
0.5.7 is quite old. Can you try to reproduce the issue with 0.5.9 or even git master?

The other important info would be, which operating system you are using.

vintagepc, 2021-07-19:
OpenSuSE 15.3. 

Sorry, did not realize it was that out of date, usually OpenSuSE is quite good about keeping packages in sync with upstream on newer releases. (Especially since http://lcdproc.omnipotent.net/ only notes up to 0.5.6)

Hope to have some time to double check it in the next few days and will report back.

haraldg, 2021-07-19:
Yes, unfortunately nobody updated the website in a really long time.

Anyway: Since you are on linux, can you provide the content of /proc/mounts when one of the offending devices is mounted? This should contain everything needed to reproduce the issue.

vintagepc, 2021-07-19:
Will check after work tonight 👍

vintagepc, 2021-07-19:
Here's the output of `cat /proc/mounts`

```
sysfs /sys sysfs rw,nosuid,nodev,noexec,relatime 0 0
proc /proc proc rw,nosuid,nodev,noexec,relatime 0 0
devtmpfs /dev devtmpfs rw,nosuid,noexec,size=4096k,nr_inodes=65536,mode=755 0 0
securityfs /sys/kernel/security securityfs rw,nosuid,nodev,noexec,relatime 0 0
tmpfs /dev/shm tmpfs rw,nosuid,nodev 0 0
devpts /dev/pts devpts rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000 0 0
tmpfs /run tmpfs rw,nosuid,nodev,size=6564696k,nr_inodes=819200,mode=755 0 0
tmpfs /sys/fs/cgroup tmpfs ro,nosuid,nodev,noexec,size=4096k,nr_inodes=1024,mode=755 0 0
cgroup2 /sys/fs/cgroup/unified cgroup2 rw,nosuid,nodev,noexec,relatime,nsdelegate 0 0
cgroup /sys/fs/cgroup/systemd cgroup rw,nosuid,nodev,noexec,relatime,xattr,name=systemd 0 0
pstore /sys/fs/pstore pstore rw,nosuid,nodev,noexec,relatime 0 0
efivarfs /sys/firmware/efi/efivars efivarfs rw,nosuid,nodev,noexec,relatime 0 0
none /sys/fs/bpf bpf rw,nosuid,nodev,noexec,relatime,mode=700 0 0
cgroup /sys/fs/cgroup/perf_event cgroup rw,nosuid,nodev,noexec,relatime,perf_event 0 0
cgroup /sys/fs/cgroup/net_cls,net_prio cgroup rw,nosuid,nodev,noexec,relatime,net_cls,net_prio 0 0
cgroup /sys/fs/cgroup/blkio cgroup rw,nosuid,nodev,noexec,relatime,blkio 0 0
cgroup /sys/fs/cgroup/freezer cgroup rw,nosuid,nodev,noexec,relatime,freezer 0 0
cgroup /sys/fs/cgroup/memory cgroup rw,nosuid,nodev,noexec,relatime,memory 0 0
cgroup /sys/fs/cgroup/cpuset cgroup rw,nosuid,nodev,noexec,relatime,cpuset 0 0
cgroup /sys/fs/cgroup/rdma cgroup rw,nosuid,nodev,noexec,relatime,rdma 0 0
cgroup /sys/fs/cgroup/hugetlb cgroup rw,nosuid,nodev,noexec,relatime,hugetlb 0 0
cgroup /sys/fs/cgroup/cpu,cpuacct cgroup rw,nosuid,nodev,noexec,relatime,cpu,cpuacct 0 0
cgroup /sys/fs/cgroup/pids cgroup rw,nosuid,nodev,noexec,relatime,pids 0 0
cgroup /sys/fs/cgroup/devices cgroup rw,nosuid,nodev,noexec,relatime,devices 0 0
/dev/sda4 / ext4 rw,relatime 0 0
systemd-1 /proc/sys/fs/binfmt_misc autofs rw,relatime,fd=29,pgrp=1,timeout=0,minproto=5,maxproto=5,direct,pipe_ino=714 0 0
debugfs /sys/kernel/debug debugfs rw,nosuid,nodev,noexec,relatime 0 0
mqueue /dev/mqueue mqueue rw,nosuid,nodev,noexec,relatime 0 0
hugetlbfs /dev/hugepages hugetlbfs rw,relatime,pagesize=2M 0 0
tracefs /sys/kernel/tracing tracefs rw,nosuid,nodev,noexec,relatime 0 0
configfs /sys/kernel/config configfs rw,nosuid,nodev,noexec,relatime 0 0
fusectl /sys/fs/fuse/connections fusectl rw,nosuid,nodev,noexec,relatime 0 0
/dev/sda3 /media/WinSys fuseblk rw,nosuid,nodev,relatime,user_id=0,group_id=0,default_permissions,allow_other,blksize=4096 0 0
/dev/sda1 /boot/efi vfat rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,utf8,errors=remount-ro 0 0
/dev/md126p2 /media/WinData fuseblk rw,nosuid,nodev,relatime,user_id=0,group_id=0,default_permissions,allow_other,blksize=4096 0 0
/dev/md126p1 /home ext4 rw,relatime,stripe=32,data=ordered 0 0
tmpfs /run/user/1000 tmpfs rw,nosuid,nodev,relatime,size=3282344k,nr_inodes=820586,mode=700,uid=1000,gid=100 0 0
gvfsd-fuse /run/user/1000/gvfs fuse.gvfsd-fuse rw,nosuid,nodev,relatime,user_id=1000,group_id=100 0 0
tracefs /sys/kernel/debug/tracing tracefs rw,nosuid,nodev,noexec,relatime 0 0
binfmt_misc /proc/sys/fs/binfmt_misc binfmt_misc rw,nosuid,nodev,noexec,relatime 0 0
PrusaSlicer-2.3.2+linux-x64-GTK3-202107080721.AppImage /tmp/.mount_PrusaSSiFlVJ fuse.PrusaSlicer-2.3.2+linux-x64-GTK3-202107080721.AppImage ro,nosuid,nodev,relatime,user_id=1000,group_id=100 0 0
/dev/sdj1 /run/media/user/PRUSA_MINI vfat rw,nosuid,nodev,relatime,uid=1000,gid=100,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,showexec,utf8,flush,errors=remount-ro 0 0
```

as well as `df`:

vintagepc, 2021-07-19:
Side request... wouldn't mind being able to either blacklist or whitelist drives from being shown. I don't particularly care for the usage level of /boot/efi for example.

haraldg, 2021-07-19:
Well `clients/lcdproc/machine_Linux.c` already has some logic to drop uninteresting entries. How comfortable are you with C?
If you have an idea how to improve this, I'm open to pull requests.

vintagepc, 2021-07-19:
I'll look this coming weekend if I have some time :)

vintagepc, 2021-07-24:
I can confirm this issue is still present in 0.5.9+20200721. I'll wait for the PR to be merged :+1:

haraldg, 2021-07-24:
Well, the PR should be reworked as noted there. If nobody pick's it up, I'll do it before the next release.

vintagepc, 2021-07-25:
I'll have a poke at it today over my morning coffee, seems pretty straightforward and I'm in that area anyway to experiment with an "ignore" list for mountpoints.

PR #181 (2021-07-25): Fix buffer overflow in Disk screen with  wide displays and long mount paths.
Buffer overflow bugfix with rework to remove the now-unnecessary table/loop and an unused variable.

- Fixes #178 
- Supersedes  #158 
- Supersedes #180
closed 2021-07-25 by haraldg