lcdproc/lcdproc #185: Question - Winstar OLED ws0010 dim brightness

emanueleci, 2021-09-13:
Hi, i have an 20x4 Winstar OLED screen (HD44780 compatible, i2c wired with backpack), and it works perfect with latest DEV build.
I can also DIM screen using external backlight settings in LCDd.conf, so i can change from full bright to dim bright.
The problem is that now i need to implement a toggle to completely turn bright off and on, without the needs to clear screen. Is this possible?
I tryed to set OffBrightness to 0 in LCDd.conf but it doesn't seem to work.

Thank you 
Emanuele

ethandicks, 2021-09-13:
On Mon, Sep 13, 2021 at 3:07 PM emanueleci ***@***.***> wrote:
> Hi, i have an 20x4 Winstar OLED screen (HD44780 compatible, i2c wired with backpack), and it works perfect with latest DEV build.
> I can also DIM screen using external backlight settings in [...]

Looking at the source for hd44780.c, I see a couple of implementation
details that stand out. In particular, in hd44780_set_brightness(), it
doesn't change the display so much as change the internal brightness
value then marks the backlight state "invalid" which will get
corrected on a future display update.

If you mean "how can I "dim" to 0 without writing to the display at
all?", I don't think that's going to work.  Other parts of the code
skip sending chars if there are no visible changes, so without _some_
change to the contents of the display, it's likely that the
set_brightness is going to wait until there is a change.

Also, I see other places where it flips between WINST_PWRON and
WINST_PWROFF and doesn't use the value of off brightness directly, but
as a threshold for on vs off.

Between those two things, I don't think it's written to behave in the
manner you are describing.  I'm not quite sure of the behavior you are
looking for (time-out power down?  "wake-up" functionality? etc) but
any changes to the current implementation have to be written in a way
that doesn't break other drivers (only a few devices have fine-tuned
controllable brightness)

If you could describe in a little more detail how you want the
backlight to act (what triggers the change, especially), that would be
helpful.

emanueleci, 2021-09-13:
hi @ethandicks, thanks for your time.
I'm working on an audio player based on picorePlayer and LMS.
My player has 2 screens, 1 graphic oled for track covers and one Winstar OLED for track information etc.
All software is written in c++ and it uses LCDAPI to communiate with LCDProc.
I'm working now to a function that let users to turn display on and off via some toggle, so my goal is something like this:
Display full bright -> Display dimmed -> display off -> Display full bright -> etc.
But i didn't find any way to reach the "display off" goal without manually clear all the screen lines.

Emanuele

mskalski, 2021-09-13:
Hi @emanueleci 

I wrote support for Winstar OLED internal backlight handling some time ago, but I only wanted to map client-controllable backlight setting to high and low intensity instead of controlling LED backlight (connected to dedicated pin) on standard LCD screens.

On the other hand I made some changes to work with other displays (VFD ones), which have four-level brightness setting and I wanted to make controllable brightness for other types of displays I wasn't aware of. It is achieved with `Backlight` option, which is used to decide how to map backlight enabling and disabling:
- don't use at all (`none`)
- turn on/off some external pin (`external`)
- use internal command (like Winstar OLED has) for change display intensity (`internal`)
- the same as internal one, but allowing to use any commands, not hardcoded for display type (`internalCmds`).

So this is little hackish, but you can try to use options below in `LCDd.conf` to achieve completely off and fully on screen:
```
Backlight = internalCmds
BacklightCmdOn = 0x80
BacklightCmdOff = 0xC017
```

Command to turn off display is standard _Display Off_, and for maximum brightness is _Display On_ (0xC0) followed by Winstar's specific enabling internal power (0x17).

It should work, however I did't try this like that :-) I only made tests using Winstar internal ommands 0x13 and 0x17.

That way you should have display completely off when backlight is off and fully on, when backlight is on.

> ### All software is written in c++ and it uses LCDAPI to communiate with LCDProc.

What do you mean LCDAPI? Do you use [LCD commands from development guide](http://lcdproc.sourceforge.net/docs/lcdproc-0-5-5-dev.html#language-commands)?

- If yes, the simple solution is to set your screen's priority to value `hidden` to turn display fully off. Here is excerpt from LCDProc development guide, descriptions of client to server communication language:

  <div class="titlepage"><div><div><h3 class="title">Screens and widgets</h3></div></div></div><div class="variablelist"><dl><dt><span class="term">
	    <span class="command"><strong>screen_set
	      <code class="option"><em class="replaceable"><code>screen_id</code></em></code>
	      <code class="option"><em class="replaceable"><code>attributes</code></em>...</code>
	    </strong></span>
	  </span></dt><dd><p>
	      Sets attributes for the given screen. The following attributes
	      exist:
	      </p><div class="variablelist"><dl><dt><span class="term">
		    <code class="option">-name <em class="replaceable"><code>name</code></em></code>
		  </span></dt><dd><p>
		      Sets the screen's name as visible to a user.
		    </p></dd><dt><span class="term">
		    <code class="option">-wid <em class="replaceable"><code>int</code></em></code>
		    <code class="option">-hgt <em class="replaceable"><code>int</code></em></code>
		  </span></dt><dd><p>
		      Sets the size of the screen in characters. If unset, the full display size is assumed.
		    </p></dd><dt><span class="term">
		    <code class="option">-priority <em class="replaceable"><code>pri_class</code></em></code>
		  </span></dt><dd><p>
		      Sets the screen's priority.
		      The following priority classes exist:
		      </p><div class="variablelist"><dl><dt><span class="term"><code class="literal">hidden</code></span></dt><dd><p>
			      The screen will never be visible
			    </p></dd><dt><span class="term"><code class="literal">background</code></span></dt><dd><p>
			      The screen is only visible when no normal info screens exists
			    </p></dd><dt><span class="term"><code class="literal">info</code></span></dt><dd><p>
			      normal info screen, default priority
			    </p></dd><dt><span class="term"><code class="literal">foreground</code></span></dt><dd><p>
			      an active client
			    </p></dd><dt><span class="term"><code class="literal">alert</code></span></dt><dd><p>
			      The screen has an important message for the user.
			    </p></dd><dt><span class="term"><code class="literal">input</code></span></dt><dd><p>
			      The client is doing interactive input.
			    </p></dd><dt><span class="term"><em class="replaceable"><code>int</code></em></span></dt><dd><p>
			      a positive integer that maps to priority classes above
			      according to the mapping given in the table below.
			    </p><div class="informaltable">

  range | priority
  -- | --
[... 28 lines cut ...]
  
  So from client's perspective you could just hide screen when it should be fully turned off and use backlight on/off commands to get full and dimmed display using standard configuration in `LCDd.conf`:

  ```
  hello
  client_set -name music_player
  screen_add track_info
  screen_set track_info -name "Track Info" 

  # Hide screen
  screen_set track_info -priority hidden

  # Set dimmed display
  screen_set track_info -priority info -backlight off

  # Set full-brightness display
  screen_set track_info -priority info -backlight on
  ```

- If you don't use lcdproc client language, but would like to modify brightness by modifying `LCDd.conf` configuration file, then you could try to modify `Backlight` / `BacklightCmd*` options described above.

- If you just use modified version of LCDD daemon and just call [HD44780_set_brightness()](https://github.com/lcdproc/lcdproc/blob/master/server/drivers/hd44780.c#L1078) function, then this won't work for Winstar OLED display as it was not intended for it.

Greets
Michał

emanueleci, 2021-09-14:
Hi @mskalski, thank you.
I think something is not working as expected on my side. 
I start from the beginning to exclude other problems :)

Winstar OLED connected via i2C with PCF8574. Address is 0x27
Configuration i2c pin is:
i2c_line_RS=0x01
i2c_line_RW=0x02
i2c_line_EN=0x04
i2c_line_BL=0x08
i2c_line_D4=0x10
i2c_line_D5=0x20
i2c_line_D6=0x40
i2c_line_D7=0x80

model selected: 
Model = winstar_oled
Brightness=800
OffBrightness=300

Text is showed correctly, no garbage, no problem at all. 
Then if i try to change backlight: 

- Backlight=**external** and using lcdproc client language via CLI, as suggested by you for testing, backlight works (only full bright and DIM as expected)
- Backlight=**internal** and trying -backlight off and -backlight on via lcdproc CLI, nothing happens
- Backlight=**internalCmds** with BacklightCmdOn=0x80 and BacklightCmdOff=0xC017, same as above, nothing happens

It looks like **internal** commands is not working in my configuration via i2C, while external works.
Maybe internal and internalCmds is for serial wiring only?

Thanks, 
Emanuele

mskalski, 2021-09-22:
Hi, Emanuele

Could you verify settings are parsed as expected? I usually run `LCDd` daemon in foreground using one of following commands (on raspbian, please adapt this to your setup):

``` sh
$ LCDd -c LCDd.conf -f -s 0 -r 4
$ LCDd -c LCDd.conf -f -s 0 -r 5 2>&1 | grep -v '^screenlist_'
```
*(the second one shows also backlight setting commands provided by client - filtered out of many "screenlist_*" logs).*

I tested with WEH001602A Winstar OED display with I2C backpack and with settings below:
``` ini
## Server section with all kinds of settings for the LCDd server ##
[server]
DriverPath=/usr/local/lib/lcdproc/
driver = hd44780
Bind=0.0.0.0
Port=13666

#ReportLevel=4
#ReportToSyslog=yes
User=nobody
#Foreground=yes
#Backlight=open
Heartbeat=no
ServerScreen=no

Hello = Hello

## Hitachi HD44780 driver ##
[hd44780]
ConnectionType=i2c
Model=winstar_oled
Port=0x23
Device=/dev/i2c-1
Size=16x2

# Here is my required setup
i2c_line_RS=0x01
i2c_line_RW=0x02
i2c_line_EN=0x04
i2c_line_BL=0x08
i2c_line_D4=0x10
i2c_line_D5=0x20
i2c_line_D6=0x40
i2c_line_D7=0x80

# Use embedded commands to control brightness
#Backlight=internal

# Use provided below commands for brightness setting
Backlight=internalCmds
# Turn on display and make fully bright. But 0x0c would be enough
BacklightCmdOn=0x0c17
# Turn off display
BacklightCmdOff=0x08
# These are standard commands used when Backlight = internal
#BacklightCmdOn=0x17
#BacklightCmdOff=0x13

# EOF
```

Everything works as I expected: 
- `Backlight = internal` makes display fully bright or dimmed when client enables or disables backlight respectively
- `Backlight = internalCmds` invokes given commands also when client controls backlight

Except commands I provided - **they were wrong**, [according to spec](https://www.crystalfontz.com/controllers/Winstar%20Display/WS0010) it should be **`BacklightCmdOn=0x0C17`** and **`BacklightCmdOff=0x08`**. That way enabling backlight makes display fully bright and switching backlight off turns off also entire display.

And make sure `Backlight` option is provided only once on `[hd44780]` section, otherwise it is treated as list.

Greets
Michał

emanueleci, 2021-09-23:
Hi Michal,
this is my results:

- **External:** backlight works (only full bright and DIM as expected). LCDd daemon in foreground doesn't provide any usefull info about commands sent
- **Internal**: Backlight stay ON, OFF command doesn't work at all
  backlight: internal
  hd44780: setting BL off using winstar_oled internal cmd: 13
  hd44780: setting BL on using winstar_oled internal cmd: 17
- **InternalCmds**: ON and OFF **Works** with your new commands suggested
  HD44780: backlight: internalCmds
  HD44780: backlight config commands: on: c17, off: 08

So **internal** OFF is not working (maybe because it uses 13 instead of 08), but i can use internalCmds and reach my goal.
May i ask you where did you find to use 0x08 and 0x0C17? I really don't understand how to find this info in the WS0010 spec you linked. Are we saying that it could be possible to use an intermediate value to choose any level of brightness? How to calculate the value?

Thank you very mutch
Emanuele

mskalski, 2021-09-24:
Hi

>Reading in depth I realized that you are probably referring to "D: DISPLAY ON / OFF BIT
When D is set to "1", the display is turned ON. When D is set to "0", the display is turned OFF and the
display data is stored in the DDRAM. The display data can be instantly displayed by setting D to "1". "

Yes, this is how I imagined to achieve fully off screen.

**Internal** only works for Winstar WS0010 OLED controller like described in spec, which has brightness control (2 levels) using special commands 0x17 and 0x13. Here are some examples of these: [20x2 OLED display](https://www.crystalfontz.com/product/cfal2002ay-character-oled-module) and [here](https://kamami.pl/en/alphanumeric-oled-displays/578777-weh002002alpp5n00008-longlife-oled-128x64-display-module.html) or [16x2 OLED display](https://kamami.pl/en/alphanumeric-oled-displays/178755-oled-weh001602alpp5n00001.html).

According to your description it looks like your display is regular LCD one with black surface and dots highlighted by external LED (like for example [here](https://www.crystalfontz.com/product/cfah1602bymijt-text-display-module-16x2)), possibly this is the reason only `external` backlight option works. For external highlight there is no way to change programmatically brightness intensity without some external circuitry.

Greets, Michał

emanueleci, 2021-09-24:
Hi Michal no, i have a winstar WEH002004AWPP5N00005, a 20x4 Oled with WS0010 as reported by spec.
I made a lot of attempts, included moving JV0 and JV connection on the backside (you can read some info here [](https://picaxeforum.co.uk/threads/winstar-oled-brightness-control-version-2-now-using-lcd-contrast-pin-axe133.31030/)

I understood that capability involved in backlight are (from spec):

- D: DISPLAY ON/OFF BIT: When D is set to "1", the display is turned ON. When D is set to "0", the display is turned OFF and the
display data is stored in the DDRAM. The display data can be instantly displayed by setting D to "1

This is possible by setting:
BacklightCmdOn=0x0c
BacklightCmdOff=0x08
and it **WORKS** using internalCmds, with full OFF and full ON

- PWR: ENABLE/DISABLE INTERNAL POWER
This bit is used to turn ON or turn OFF the internal power.
When PWR = 1, the internal power is turned ON.
When PWR = 0, the internal power is turned OFF.

This is possible by setting:
BacklightCmdOn=0x17
BacklightCmdOff=0x13
and it **DOES NOT WORKS** using internalCmds for me.

Maybe we have a different pcb and firmware revision and my display has not that feature (internal power, not well documented in spec).

Trying to investigate further.
Emanuele