lcdproc/lcdproc #187: SpacePilot Pro

KenwoodFox, 2021-12-04:
I was doing a little sluthing around on irc, exploring how my space pilot pro used to work with the logitech screen-stuff from windows way back, a user mentioned that lcdproc had come from similar logitech components, and I became currious if perhaps the same prototocal would work with my SpacePilot, that would be sick! From what i know logitech made all the components, it would be really neat to use the screen for something.

Id like to know where to get started testing, or maybe anybody has ideas about the feasibility, this rocks!

[image]

haraldg, 2021-12-16:
Maybe they have been referring to the lcdproc g15 driver. However we don't implement the protocol ourselves. We just use the implementation in libg15.

wlbaker, 2022-07-14:
I was recently able to reverse engineer the USB protocol for this device.  It is a 240x320 display that uses 16-bits/pixel for rgb.  4-bits/r, 4-bits/g 4-bits/b.  A simple uint_16[240][320] is the starting point for user space.  There is a (512 byte?) header also attached to the BULK usb payload, but it appears only the first 16 bytes have any meaning and I've copied them straight from the USB trace.  Hope to contribute the code to a library but that is a work in progress.  Literally reverse engineered a couple of days ago, but it's a very simple interface.  Also hoping to see better overall 3DConnexion device support.

haraldg, 2022-07-15:
Awesome! If you are on linux, maybe this could become a framebuffer or fbtft or "whatever is the thing these days" driver. Then this could be used from almost any application / library.

wlbaker, 2022-07-16:
As soon as I have something I will post it in a git repository.  Never take
a programmer's estimate as fact, but I'm working on it now and will post it
long before it is production quality, probably next weekend.  I'll release
something that simply drives the LCD.  I have working code, but it's not
pretty.  To get to this point I started hacking the g15tools project.  It
was a good starting point for me, but that code is less than production
quality itself.  Refactoring now, cleaning, and updating to libusb-1.0.  I
saw another one of your posts and it reminded me of a saying: "Life is too
short to learn autoconfig".

wlbaker, 2022-07-16:
Outside the scope of lcdproc, wouldn't a proper linux kernel driver for the
3dconnexion devices have advantages over the awkward user space solution
provided by spacenavd and associated utilities?

KenwoodFox, 2022-07-18:
That could be nice but, i've kinda always liked spacenavd.

Id really love to be able to use that screen, if its outside the scope of LCD proc, maybe we should try seeing about implementing it into a kernel driver? Time to learn how to develop for the kernel.

wlbaker, 2022-07-23:
Here is the project to demonstrate drawing text and graphics on the
SpacePilot Pro LCD using libusb.  Standard C.  All userspace.  Only useful
for programmers at this point, but it does build an executable that
actually demonstrates the functionality.  Should only be used as a starting
point to the LCD management.  But not a bad start!!

I looked closely at the Logitech LCD Manager and have started thinking
about a similar program for Linux.  It's interesting how LCD Manager
customizes the keys based on the selected application.  That's nice for
apps that don't understand how to handle the raw input from the device
keys.  It also gives some use for the LCD display.  Looking further ahead
at the "Enterprise" device, the LCD is too central to be overlooked.

The general subject of handling programmable keys is not something that
I've seen well handled in Linux.  Although I am saying "programmable keys",
I really mean keys outside the standard keyboard mapping that are sometimes
used to invoke custom behavior -- almost none of these devices have actual
programmable keys.  There are some mechanisms, like IBUS or DBUS that might
provide a standard technical underpinning for that functionality -- as well
as other keyboard managers for entering foreign languages.  I'm just
starting to experiment in the area and learn what is the current state of
the art.  Input is very welcome!  It looks to me like there is no
one-size-fits-all solution.  A really complete solution would seem to
involve kernel drivers, raw key input, a library to enable programs willing
to adopt it, as well as the top-level LCD Manager type application.  That's
a big bite to handle.

Long term, spacenavd will need some serious work to remain viable in a
Wayland world.  It provides raw events through the /dev/input/event
devices, but it really focuses on X keyboard events.

wlbaker, 2022-07-23:
I guess I should actually post the link:
https://github.com/wlbaker/spacepilotpro-lcd

KenwoodFox, 2022-07-26:
This is really great, thanks for publishing the source code!

Ill maybe open an issue or pr on your new repo to continue discussion but, i think little steps might be a good direction, you mentioned the huge scope in your post above..

Maybe getting the screen to show up as an extra display would be.. a good start~ that would make me personally very happy, i kinda doubt we'd get much adoption beyond that but who knows! I can say for sure it would be great to detach a window from blender, freecad or one of the other programs that uses spacenavd, and have a little mini-window filling the screen on the device

Ill compile and test with my model :3

closed 2022-07-26 by KenwoodFox