lcdproc/lcdproc #204: Logitech G510 support/howto

Lurgainn, 2023-08-30:
Hello everyone,
I saw the issue #124 and I saw the commits to permit g15 driver to direct access to hidraw, but I can't understand if these changes are compatible with G510 keyboard.
I'm using Xubuntu 23.04 with 6.2 kernel, so I'm able to use all the macro/memory/lcd keys of my keyboard using the standard uinput (see [uinput-macropad](https://github.com/sebastiansam55/uinput-macropad)), but I can't find a way to get access to the display.  
I guess that the G510's display are identical to that of G15, but in /dev/hidraw* I don't find anything related to the G510's display. And as told in #124, the usb output endpoint of the keyboard is in the same usb interface with input from some (media?) keys.
Thanks for any help.

This is the output of lsusb of the keyboard:
```
Bus 003 Device 087: ID 046d:c22d Logitech, Inc. G510 Gaming Keyboard
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x046d Logitech, Inc.
  idProduct          0xc22d G510 Gaming Keyboard
  bcdDevice            1.65
  iManufacturer           1 Logitech
  iProduct                2 G510 Gaming Keyboard
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0042
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      65
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               4
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     166
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
Device Status:     0x0000
  (Bus Powered)
```

Lurgainn, 2023-09-02:
Sorry, I found that Xububtu package was compiled without g15 support.
Downloaded from here and all is OK.

closed 2023-09-02 by Lurgainn

mskalski, 2023-09-02:
Well, Ubuntu (checked on 18.04 and probably also newer versions) include `g15` driver in `lcdproc-extra-drivers` package:
```
$ dpkg -L lcdproc-extra-drivers
/.
/usr
/usr/lib
/usr/lib/x86_64-linux-gnu
/usr/lib/x86_64-linux-gnu/lcdproc
/usr/lib/x86_64-linux-gnu/lcdproc/g15.so
/usr/lib/x86_64-linux-gnu/lcdproc/hd44780.so
/usr/lib/x86_64-linux-gnu/lcdproc/lirc.so
/usr/lib/x86_64-linux-gnu/lcdproc/picolcd.so
/usr/lib/x86_64-linux-gnu/lcdproc/xosd.so
/usr/share
/usr/share/doc
/usr/share/doc/lcdproc-extra-drivers
/usr/share/doc/lcdproc-extra-drivers/copyright
/usr/share/doc/lcdproc-extra-drivers/NEWS.Debian.gz
/usr/share/doc/lcdproc-extra-drivers/changelog.Debian.gz
```