lcdproc/lcdproc #205: unable to build lcdproc

5h4rk1337, 2023-09-29:
Downloaded master today.
run
```
sh ./autogen.sh
./configure --enable-drivers=all
```
"sh ./autogen.sh" runs fine.
"./configure --enable-drivers=all" error

```
tobi@pop-os:~/Downloads/lcdproc-master$ ./configure --enable-drivers=all
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking whether to enable debugging... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of gcc... gcc3
./configure: line 4799: syntax error near unexpected token `newline'
./configure: line 4799: `    '''
```

I run Pop 22.04 jammy
Thanks for your support <3

ethandicks, 2023-11-24:
Which Linux distro and version?  What version of gcc?

ethandicks, 2023-11-24:
I don't run Pop on my Raspberry Pi.  This will take some investigation, but to speed things up, please post the results of 'dpkg -l'

5h4rk1337, 2023-11-25:
Well, thanks for your effort. I tried it again today and it worked without any problems.

closed 2023-11-25 by 5h4rk1337