lcdproc/lcdproc #208: G15 driver

5h4rk1337, 2023-11-25:
Hi there,
how can I make lcdproc work with my Logitech G15? I would love to see if my G15 would work with lcdproc. My goal is it, to have a system monitor on the LCD. The G15Daemon gives me just a clock.

OS: Pop 22.04 jammy
Kernel: x86_64 Linux 6.5.6-76060506-generic

When I run ./configure --enable-drivers=all it says:
`checking for g15daemon_client.h... no
configure: WARNING: libg15daemon_client.h not found, the g15 driver will lack g15daemon support
checking for libg15render.h... no
configure: WARNING: The g15driver needs libg15render.h`

Found the Source Code of libg15render somewhere in the internet and uploaded it to GitHub.
https://github.com/5h4rk1337/libg15render/tree/main

Thank you for your support <3

5h4rk1337, 2024-10-23:
Finally got time to test it. Unfortunately it failed.

Building the libg15 from g15tools worked fine after i installed libusb-dev.

In `configure.in` I used pkg-config instead of pkgconf. Seems to work well. Did this because I run pop_OS! (debian).

I think my problem is to compile lcdproc correctly.
I got the source from https://github.com/lcdproc/lcdproc
I run `LDFLAGS="-L/usr/local/lib" ./configure --enable-drivers=g15 --prefix=/usr/local`
The output says:
```
configure: checking which drivers to compile...
checking for g15daemon_client.h... no
configure: WARNING: libg15daemon_client.h not found, the g15 driver will lack g15daemon support
checking for libg15render.h... yes
checking for g15r_initCanvas in -lg15render... yes
---------------------------------------
LCDd will be compiled with the drivers:
    -  g15
---------------------------------------
```
I compiled it anyways.
I configured the server.
But, like expected, LCDproc failed to start.
```
$ LD_LIBRARY_PATH="/usr/local/lib:/usr/local/lib/lcdproc" /usr/local/sbin/LCDd -c /usr/local/etc/LCDd.conf -f
LCDd 0.5dev, LCDproc Protocol 0.4
Copyright (C) 1998-2017 William Ferrell, Selene Scriven
                        and many other contributors

This program is free software; you can redistribute it and/or
modify it under the terms of the GNU General Public License
as published by the Free Software Foundation; either version 2
of the License, or (at your option) any later version.

g15: Sorry, cannot find a G15 keyboard
Driver [g15] init failed, return code -1
Could not load driver g15
There is no output driver
Critical error while initializing, abort.
```

Any idea why LCDproc is unable to find `g15daemon_client.h` and `libg15daemon_client.h`?

Thanks a lot!

ancoron, 2024-10-23:
@5h4rk1337 You don't need the G15 daemon and it's just a warning for a reason during configuration of LCDproc.

Your real issue is:
```
...
g15: Sorry, cannot find a G15 keyboard
...
```

This means that the libg15 itself is not able to find (or access) your keyboard.

This may be caused by some device permission issues. E.g. for my G510, I have these udev rules (in `/etc/udev/rules.d/logitech-g510.rules`):
```
ACTION=="remove", GOTO="logitech_end"
ACTION=="unbind", GOTO="logitech_end"

ACTION=="add", KERNEL=="hidraw*", SUBSYSTEM=="hidraw", \
    ATTRS{manufacturer}=="Logitech", ATTRS{product}=="G510 Gaming Keyboard", \
    TAG+="uaccess", GROUP="wheel", MODE="0660"

ACTION=="add", SUBSYSTEM=="leds", \
    ATTRS{manufacturer}=="Logitech", ATTRS{product}=="G510 Gaming Keyboard", \
    RUN+="/bin/chgrp -R wheel '/sys%p'", RUN+="/bin/chmod -R g=u '/sys%p'"
ACTION=="change", SUBSYSTEM=="leds", \
    ATTRS{manufacturer}=="Logitech", ATTRS{product}=="G510 Gaming Keyboard", \
    ENV{TRIGGER}!="none", \
    RUN+="/bin/chgrp -R wheel /sys%p", RUN+="/bin/chmod -R g=u /sys%p"

# just to see that these rules are actually called...
KERNEL=="g15::kbd_backlight", SUBSYSTEM=="leds", ATTR{brightness}="200", ATTR{color}="#ff44cc"
KERNEL=="g15::power_on_backlight_val", SUBSYSTEM=="leds", ATTR{brightness}="200", ATTR{color}="#ff44cc"

LABEL="logitech_end"
```

The actual group (`wheel` in my example) might be something else on your distro and must be a group that your user is attached to.

5h4rk1337, 2024-10-26:
The display is working now. Thank you very much.

Added my user to the group plugdev.

/etc/udev/rules.d$ cat 99-logitech-g15.rules 
```
ACTION=="remove", GOTO="logitech_end"
ACTION=="unbind", GOTO="logitech_end"

# Regel für Logitech G15 Gaming Keyboard (Hauptgerät)
KERNEL=="hidraw*", SUBSYSTEM=="hidraw", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c227", MODE="0666"

# Regel für das G15 Display oder zusätzliche Komponenten
KERNEL=="hidraw*", SUBSYSTEM=="hidraw", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c223", MODE="0666"

ACTION=="add", KERNEL=="hidraw*", SUBSYSTEM=="hidraw", \
    ATTRS{manufacturer}=="Logitech", ATTRS{product}=="G15 Refresh Keyboard", \
    TAG+="uaccess", GROUP="plugdev", MODE="0660"

ACTION=="add", SUBSYSTEM=="leds", \
    ATTRS{manufacturer}=="Logitech", ATTRS{product}=="G15 Refresh Keyboard", \
    RUN+="/bin/chgrp -R plugdev '/sys%p'", RUN+="/bin/chmod -R g=u '/sys%p'"
ACTION=="change", SUBSYSTEM=="leds", \
    ATTRS{manufacturer}=="Logitech", ATTRS{product}=="G15 Refresh Keyboard", \
    ENV{TRIGGER}!="none", \
    RUN+="/bin/chgrp -R plugdev /sys%p", RUN+="/bin/chmod -R g=u /sys%p"

# just to see that these rules are actually called...
KERNEL=="g15::kbd_backlight", SUBSYSTEM=="leds", ATTR{brightness}="2"
KERNEL=="g15::lcd_backlight", SUBSYSTEM=="leds", ATTR{brightness}="2"

LABEL="logitech_end"
```

/etc/systemd/system$ cat LCDd.service 
```
[Unit]
Description=LCDd Service for lcdproc
After=network.target

[Service]
Type=simple
Environment="LD_LIBRARY_PATH=/usr/local/lib:/usr/local/lib/lcdproc"
ExecStart=/usr/local/sbin/LCDd -c /usr/local/etc/LCDd.conf -f
Restart=on-failure

[Install]
WantedBy=multi-user.target
```

/etc/systemd/system$ cat LCDd.service
```
[Unit]
Description=LCDd Service for lcdproc
After=network.target

[Service]
Type=simple
Environment="LD_LIBRARY_PATH=/usr/local/lib:/usr/local/lib/lcdproc"
ExecStart=/usr/local/sbin/LCDd -c /usr/local/etc/LCDd.conf -f
Restart=on-failure

[Install]
WantedBy=multi-user.target
tobi@pop-os:/etc/systemd/system$ cat lcdproc.service 
[Unit]
Description=lcdproc Client
Requires=LCDd.service
After=LCDd.service

[Service]
ExecStart=/usr/local/bin/lcdproc
Restart=on-failure
RestartSec=5s

[Install]
WantedBy=default.target
```

Unfortunately the buttons for the display do not work. Looks to me like the hid-lg-g15 driver is not working correctly.

lsmod | grep lg
```
hid_lg_g15             20480  0
hid                   180224  5 hid_logitech,usbhid,hid_generic,hid_lg_g15
```

sudo evtest /dev/input/by-id/usb-046d_G15_GamePanel_LCD-
event-if00
```
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x46d product 0xc227 version 0x111
Input device name: "Logitech Gaming Keyboard Gaming Keys"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 656 (?)
    Event code 657 (?)
    Event code 658 (?)
    Event code 659 (?)
    Event code 660 (?)
    Event code 661 (?)
    Event code 688 (?)
    Event code 691 (?)
    Event code 692 (?)
    Event code 693 (?)
    Event code 696 (?)
    Event code 697 (?)
    Event code 698 (?)
    Event code 699 (?)
    Event code 700 (?)
Properties:
Testing ... (interrupt to exit)
Event: time 1729982694.179431, type 1 (EV_KEY), code 696 (?), value 1
Event: time 1729982694.179431, -------------- SYN_REPORT ------------
Event: time 1729982694.331455, type 1 (EV_KEY), code 696 (?), value 0
Event: time 1729982694.331455, -------------- SYN_REPORT ------------
Event: time 1729982694.519405, type 1 (EV_KEY), code 697 (?), value 1
Event: time 1729982694.519405, -------------- SYN_REPORT ------------
Event: time 1729982694.650404, type 1 (EV_KEY), code 697 (?), value 0
Event: time 1729982694.650404, -------------- SYN_REPORT ------------
Event: time 1729982695.862417, type 1 (EV_KEY), code 698 (?), value 1
Event: time 1729982695.862417, -------------- SYN_REPORT ------------
Event: time 1729982696.018423, type 1 (EV_KEY), code 698 (?), value 0
Event: time 1729982696.018423, -------------- SYN_REPORT ------------
Event: time 1729982696.182401, type 1 (EV_KEY), code 699 (?), value 1
Event: time 1729982696.182401, -------------- SYN_REPORT ------------
Event: time 1729982696.329452, type 1 (EV_KEY), code 699 (?), value 0
Event: time 1729982696.329452, -------------- SYN_REPORT ------------
Event: time 1729982696.641399, type 1 (EV_KEY), code 700 (?), value 1
Event: time 1729982696.641399, -------------- SYN_REPORT ------------
Event: time 1729982696.727399, type 1 (EV_KEY), code 700 (?), value 0
Event: time 1729982696.727399, -------------- SYN_REPORT ------------

```

`(EV_KEY), code 700 (?), value 0` should be something like `KEY_KBD_LCD_MENU1`
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/hid/hid-lg-g15.c

LCDd.conf
```
[server]
ToggleRotateKey=Enter
PrevScreenKey=Left
NextScreenKey=Right
```

It's a bit annoying when you can't switch between screens with the touch of a button.

ancoron, 2024-10-27:
@5h4rk1337 Yes, I also had that issue.

Since I wanted some other customizations in my setup with the multi-color G510, I ended up with going a different route:
1. LCDproc server + client (clock only)
2. custom local server in JavaScript (using [lcdprocjs](https://github.com/jlagden/lcdprocjs)) to add and control LCDproc screens
3. [uinput-macropad](https://github.com/sebastiansam55/uinput-macropad) to actually make use of the LCD keys
4. shell wrapper to push commands from macropad to the custom local server

1. - 3. are all running under the SystemD user session.

I wish there was a more modern way to do the same, since many projects in this area are not really well maintained anymore, but that's what ended up working for me.

The custom stuff I've created is not published anywhere, since it is too much tailored to my setup.

5h4rk1337, 2024-11-03:
I asked Hans, the maintainer of the G15 Linux driver "hid_lg_g15", if he has any idea why the keys are not working properly.

Here is his answer:
```
It looks like everything is working as it should. The ? in evtest
simply means that you are using an older evtest version.

Maybe your lcdproc version is older too ?

The latest lcdproc code does support the keys ootb.

Note that you need something like this in LCDd.conf
for the keys to work to control the menu in lcdproc:

[server]
Driver=g15
Driver=[linux_input]

[linux_input]
Device="Logitech Gaming Keyboard Gaming Keys"
# Keymap for the G15's 5 LCD-menu buttons
key=0x2b8,Escape
key=0x2b9,Left
key=0x2ba,Up
key=0x2bb,Down
key=0x2bc,Enter

Regards,

Hans
```

My version of lcdproc is 0.5dev.
I tried to compile version 0.5.9, but it always fails.

```
sh autogen.sh
Running aclocal ...
Running autoheader...
Running automake  ...
Running autoconf ...
configure.ac:72: warning: The macro `AC_PROG_CC_STDC' is obsolete.
configure.ac:72: You should run autoupdate.
./lib/autoconf/c.m4:1666: AC_PROG_CC_STDC is expanded from...
configure.ac:72: the top level
configure.ac:81: warning: The macro `AC_LANG_C' is obsolete.
configure.ac:81: You should run autoupdate.
./lib/autoconf/c.m4:72: AC_LANG_C is expanded from...
lib/m4sugar/m4sh.m4:692: _AS_IF_ELSE is expanded from...
lib/m4sugar/m4sh.m4:699: AS_IF is expanded from...
./lib/autoconf/general.m4:2249: AC_CACHE_VAL is expanded from...
./lib/autoconf/general.m4:2270: AC_CACHE_CHECK is expanded from...
acinclude.m4:1084: AX_CFLAGS_GCC_OPTION_NEW is expanded from...
acinclude.m4:1155: AX_CFLAGS_GCC_OPTION is expanded from...
configure.ac:81: the top level
configure.ac:81: warning: The macro `AC_TRY_COMPILE' is obsolete.
configure.ac:81: You should run autoupdate.
./lib/autoconf/general.m4:2847: AC_TRY_COMPILE is expanded from...
lib/m4sugar/m4sh.m4:692: _AS_IF_ELSE is expanded from...
lib/m4sugar/m4sh.m4:699: AS_IF is expanded from...
./lib/autoconf/general.m4:2249: AC_CACHE_VAL is expanded from...
./lib/autoconf/general.m4:2270: AC_CACHE_CHECK is expanded from...
acinclude.m4:1084: AX_CFLAGS_GCC_OPTION_NEW is expanded from...
acinclude.m4:1155: AX_CFLAGS_GCC_OPTION is expanded from...
configure.ac:81: the top level
configure.ac:82: warning: The macro `AC_LANG_C' is obsolete.
configure.ac:82: You should run autoupdate.
./lib/autoconf/c.m4:72: AC_LANG_C is expanded from...
lib/m4sugar/m4sh.m4:692: _AS_IF_ELSE is expanded from...
lib/m4sugar/m4sh.m4:699: AS_IF is expanded from...
[... 364 lines cut ...]
gcc -DHAVE_CONFIG_H -I. -I../..  -I../.. -I../../shared -DSYSCONFDIR=\"/usr/local/etc\" -DPIDFILEDIR=\"/var/run\"   -Wall  -O3 -Wno-unused-function -ftrampolines -MT machine_OpenBSD.o -MD -MP -MF .deps/machine_OpenBSD.Tpo -c -o machine_OpenBSD.o machine_OpenBSD.c
mv -f .deps/machine_OpenBSD.Tpo .deps/machine_OpenBSD.Po
gcc -DHAVE_CONFIG_H -I. -I../..  -I../.. -I../../shared -DSYSCONFDIR=\"/usr/local/etc\" -DPIDFILEDIR=\"/var/run\"   -Wall  -O3 -Wno-unused-function -ftrampolines -MT machine_FreeBSD.o -MD -MP -MF .deps/machine_FreeBSD.Tpo -c -o machine_FreeBSD.o machine_FreeBSD.c
mv -f .deps/machine_FreeBSD.Tpo .deps/machine_FreeBSD.Po
gcc -DHAVE_CONFIG_H -I. -I../..  -I../.. -I../../shared -DSYSCONFDIR=\"/usr/local/etc\" -DPIDFILEDIR=\"/var/run\"   -Wall  -O3 -Wno-unused-function -ftrampolines -MT machine_NetBSD.o -MD -MP -MF .deps/machine_NetBSD.Tpo -c -o machine_NetBSD.o machine_NetBSD.c
mv -f .deps/machine_NetBSD.Tpo .deps/machine_NetBSD.Po
gcc -DHAVE_CONFIG_H -I. -I../..  -I../.. -I../../shared -DSYSCONFDIR=\"/usr/local/etc\" -DPIDFILEDIR=\"/var/run\"   -Wall  -O3 -Wno-unused-function -ftrampolines -MT machine_Darwin.o -MD -MP -MF .deps/machine_Darwin.Tpo -c -o machine_Darwin.o machine_Darwin.c
mv -f .deps/machine_Darwin.Tpo .deps/machine_Darwin.Po
gcc -DHAVE_CONFIG_H -I. -I../..  -I../.. -I../../shared -DSYSCONFDIR=\"/usr/local/etc\" -DPIDFILEDIR=\"/var/run\"   -Wall  -O3 -Wno-unused-function -ftrampolines -MT machine_SunOS.o -MD -MP -MF .deps/machine_SunOS.Tpo -c -o machine_SunOS.o machine_SunOS.c
mv -f .deps/machine_SunOS.Tpo .deps/machine_SunOS.Po
gcc -DHAVE_CONFIG_H -I. -I../..  -I../.. -I../../shared -DSYSCONFDIR=\"/usr/local/etc\" -DPIDFILEDIR=\"/var/run\"   -Wall  -O3 -Wno-unused-function -ftrampolines -MT util.o -MD -MP -MF .deps/util.Tpo -c -o util.o util.c
mv -f .deps/util.Tpo .deps/util.Po
gcc -DHAVE_CONFIG_H -I. -I../..  -I../.. -I../../shared -DSYSCONFDIR=\"/usr/local/etc\" -DPIDFILEDIR=\"/var/run\"   -Wall  -O3 -Wno-unused-function -ftrampolines -MT iface.o -MD -MP -MF .deps/iface.Tpo -c -o iface.o iface.c
mv -f .deps/iface.Tpo .deps/iface.Po
gcc  -Wall  -O3 -Wno-unused-function -ftrampolines  -L/usr/local/lib -o lcdproc main.o mode.o batt.o chrono.o cpu.o cpu_smp.o disk.o load.o mem.o eyebox.o machine_Linux.o machine_OpenBSD.o machine_FreeBSD.o machine_NetBSD.o machine_Darwin.o machine_SunOS.o util.o iface.o ../../shared/libLCDstuff.a -ldl 
/usr/bin/ld: iface.o:(.bss+0x0): multiple definition of `iface'; main.o:(.bss+0x60): first defined here
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:436: lcdproc] Fehler 1
make[3]: Verzeichnis „/media/user/1TB-SSD/home/user/Downloads/g15/lcdproc-0.5.9/clients/lcdproc“ wird verlassen
make[2]: *** [Makefile:369: all-recursive] Fehler 1
make[2]: Verzeichnis „/media/user/1TB-SSD/home/user/Downloads/g15/lcdproc-0.5.9/clients“ wird verlassen
make[1]: *** [Makefile:467: all-recursive] Fehler 1
make[1]: Verzeichnis „/media/user/1TB-SSD/home/user/Downloads/g15/lcdproc-0.5.9“ wird verlassen
make: *** [Makefile:387: all] Fehler 2
```

5h4rk1337, 2024-11-05:
It is finally working. The missing piece was the linux_input driver.
You have to compile lcdproc like this:
```
sh autogen.sh
LDFLAGS="-L/usr/local/lib" ./configure --enable-drivers=g15,linux_input --prefix=/usr/local
make
sudo make install
```
Before you reinstall it, make sure to back up your config files!
And edit your LCDd.conf like Hans sad in his Mail.

closed 2024-11-05 by 5h4rk1337

ancoron, 2024-11-07:
@5h4rk1337 Wonderful! With that I can also adjust my setup and make it a bit simpler, hopefully.