lcdproc/lcdproc #214: hd44780-gpiod fails to compile with libgpiod-2.1.1 and gcc-14

heitbaum, 2024-05-04:
With gcc13 the compile worked - but will have issues with the libraries - as these functions are not available in libgpiod2.

```
Making all in drivers
make[3]: Entering directory '/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-13.0-devel/build/lcdd-0e2ce9b9c46c47363436f9ee730f7c71bf455f0f/.x86_64-libreelec-linux-gnu/server/drivers'
/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-13.0-devel/toolchain/bin/x86_64-libreelec-linux-gnu-gcc -DHAVE_CONFIG_H -I. -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-13.0-devel/build/lcdd-0e2ce9b9c46c47363436f9ee730f7c71bf455f0f/server/drivers -I../..  -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-13.0-devel/build/lcdd-0e2ce9b9c46c47363436f9ee730f7c71bf455f0f  -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-13.0-devel/toolchain/x86_64-libreelec-linux-gnu/sysroot/usr/include -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-13.0-devel/toolchain/x86_64-libreelec-linux-gnu/sysroot/usr/include/libusb-1.0 -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-13.0-devel/toolchain/x86_64-libreelec-linux-gnu/sysroot/usr/include/libftdi1 -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-13.0-devel/toolchain/x86_64-libreelec-linux-gnu/sysroot/usr/include/libusb-1.0 -fPIC -Wall -march=x86-64-v3 -Wall -pipe  -O2 -fomit-frame-pointer -DNDEBUG -O3 -Wno-unused-function -ftrampolines -MT hd44780-hd44780-gpiod.o -MD -MP -MF .deps/hd44780-hd44780-gpiod.Tpo -c -o hd44780-hd44780-gpiod.o `test -f 'hd44780-gpiod.c' || echo '/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-13.0-devel/build/lcdd-0e2ce9b9c46c47363436f9ee730f7c71bf455f0f/server/drivers/'`hd44780-gpiod.c
../../../server/drivers/hd44780-gpiod.c: In function 'init_gpio_pin':
../../../server/drivers/hd44780-gpiod.c:73:16: error: implicit declaration of function 'gpiod_chip_get_line'; did you mean 'gpiod_chip_get_info'? [-Wimplicit-function-declaration]
   73 |         *pin = gpiod_chip_get_line(pins->chip, number);
      |                ^~~~~~~~~~~~~~~~~~~
      |                gpiod_chip_get_info
../../../server/drivers/hd44780-gpiod.c:73:14: error: assignment to 'struct gpiod_line *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
   73 |         *pin = gpiod_chip_get_line(pins->chip, number);
      |              ^
../../../server/drivers/hd44780-gpiod.c:80:13: error: implicit declaration of function 'gpiod_line_request_output'; did you mean 'gpiod_line_request_get_fd'? [-Wimplicit-function-declaration]
   80 |         if (gpiod_line_request_output(*pin, "LCDd", 0) < 0) {
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~
      |             gpiod_line_request_get_fd
../../../server/drivers/hd44780-gpiod.c: In function 'send_nibble':
../../../server/drivers/hd44780-gpiod.c:104:9: error: implicit declaration of function 'gpiod_line_set_value'; did you mean 'gpiod_line_request_set_value'? [-Wimplicit-function-declaration]
  104 |         gpiod_line_set_value(pins->d7, ch & 0x08);
      |         ^~~~~~~~~~~~~~~~~~~~
      |         gpiod_line_request_set_value
../../../server/drivers/hd44780-gpiod.c: In function 'hd_init_gpiod':
../../../server/drivers/hd44780-gpiod.c:150:22: error: implicit declaration of function 'gpiod_chip_open_lookup'; did you mean 'gpiod_chip_open'? [-Wimplicit-function-declaration]
  150 |         pins->chip = gpiod_chip_open_lookup(chip);
      |                      ^~~~~~~~~~~~~~~~~~~~~~
      |                      gpiod_chip_open
../../../server/drivers/hd44780-gpiod.c:150:20: error: assignment to 'struct gpiod_chip *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
  150 |         pins->chip = gpiod_chip_open_lookup(chip);
      |                    ^
```

heitbaum, 2024-05-05:
> Isn't that kind of expected as libgpiod and libgpiod2 are quite different? I believe to support libgpiod2 somebody would need to write a hd44780-gpiod2 connection type.

From my look that is what it looks like. In the autoconf it does check for -lgpiod but it doesn’t check the version.

```
--- a/acinclude.m4      2024-05-04 13:24:19.121109911 +0000
+++ b/acinclude.m4      2024-05-04 13:35:28.082754524 +0000
@@ -245,13 +245,6 @@
                        ],[
                                AC_MSG_WARN([Could not find libugpio, not building hd44780-ugpio driver])
                        ])
-                       AC_CHECK_LIB(gpiod, main,[
-                               HD44780_DRIVERS="$HD44780_DRIVERS hd44780-hd44780-gpiod.o"
-                               LIBGPIOD="-lgpiod"
-                               AC_DEFINE(HAVE_GPIOD, [1], [Define to 1 if you have libgpiod])
-                       ],[
-                               AC_MSG_WARN([Could not find libgpiod, not building hd44780-gpiod driver])
-                       ])
                        if test "$ac_cv_port_have_lpt" = yes ; then
                                HD44780_DRIVERS="$HD44780_DRIVERS hd44780-hd44780-4bit.o hd44780-hd44780-ext8
bit.o hd44780-hd44780-winamp.o hd44780-hd44780-serialLpt.o hd44780-hd44780-lcm162.o"
                        fi
```

haraldg, 2024-05-06:
How are distributions going to handle this? Will there be two versions installed in parallel (like we had with libusb) or will they do a hard migration?

Is there any recommendation from libgpiod maintainers how to handle transition or some overview about the differences?

heitbaum, 2024-05-06:
I had a quick check and fedora went to v2 is 39 - https://packages.fedoraproject.org/pkgs/libgpiod/libgpiod/
We moved https://github.com/LibreELEC/LibreELEC.tv to v2 with LE12.

manio, 2024-12-07:
CC: @heitbaum: #219

commit 71877ee0 (2024-12-13): hd44780/gpiod: rewrite to support libgpiod2

Closes #214, #217.
closed 2024-12-13 by haraldg