lcdproc/lcdproc #215: support for mainline linux /dev/lcd kernel module/driver?

penfold42, 2024-05-24:
I've extended the simple "text" driver to also handle the native linux /dev/lcd driver.

Q1) is there interest in merging this?

Q2) if so, should i split it out as a discrete driver?
The changes are minimal - https://github.com/penfold42/lcdproc/tree/devlcdchar
I strip off the ascii border and send a couple of escape codes for init, clear screen and backlight control

haraldg, 2024-05-25:
I think this is worthwhile. And yes IMO, it should be it's own driver. Also it would need documentation (userguide, man page, example config file) to be complete.

Is there some documentation for the /dev/lcd driver API/ABI? It would be helpful to mention in some comment as this probably isn't common knowledge and google wasn't too helpful. But mostly I'm just curious.

penfold42, 2024-05-26:
I think I've covered all the bits required: #216